john30/ebusd #468: HTTP not responding

jordipalet, 2021-11-06:
I'm using the ebus adapter 3 with Ethernet port, and despite I get very frequent errors such as:
2021-11-06 20:20:36.118 [bus notice] device invalid
2021-11-06 20:20:36.746 [bus error] unable to open x.x.x.x:9999: ERR: generic I/O error

(I'm using EBUSD_OPTS="--scanconfig --httpport=8889 --log=all:notice -d enh:x.x.x.x:9999")

I can do a scan/find and it seems to correctly report:

scan id = no data stored
scan.05  = Vaillant;VR921;2303;5703
scan.05 id = 21;21;24;0020261008;0938;026911;N7
scan.08  = Vaillant;HMU00;0607;5103
scan.08 id = 21;21;35;0010031696;0001;005337;N4
scan.15  = Vaillant;CTLS2;0509;1304
scan.15 id = 21;21;28;0020260971;0953;007739;N4
scan.76  = Vaillant;VWZIO;0202;0103
scan.76 id = 21;21;10;0010031650;0082;005260;N9

However, trying http://10.10.10.89:8889/data doesn't provide any result (browser/curl just waiting)

I'm missing something ?

By the way I tried also using udp instead of enh, etc. same results. It also seems to me that sometimes powering off/on the ebus adapter 3, the log shows from time to time a lot of extra data and less errors, but then the errors come back again, not sure if it is related, or there is a way to "sync" the adapter with the ebus, etc.

Tks!

john30, 2021-11-06:
can you ping the adapter when ebusd reports it as "unable to open" from the same host?
provide some ebusd logging please
what does the LEDs on the adapter look like in this situation?

jordipalet, 2021-11-06:
Yes, ping is always working.

ebusd --version
ebusd 21.2.v21.2

(downloaded today from ebusd-21.2_armhf-buster.deb)

LEDs:
Orange on (I think is the one described as yellow, for the power indication)
Green blinking (eBus receive)
Blue off
Red off

If I unplug/plug the power adapter that provides power to the USB connector, I can see the Blue on, then off and also sometimes the red blinking.

In the Ethernet adapter I have the orange and green on, and a red one on as well.

Also when I do find/scan, it looks like the 
broadcast outsidetemp = 12.125
broadcast signoflife = no data stored
broadcast vdatetime = 20:27:03;06.11.2021
are the same since a few hours ago, not sure if that's the expected behaviour ...

jordipalet, 2021-11-14:
So ... any hint on what else I can do/check?

john30, 2021-11-16:
please try with a build from the latest sources

jordipalet, 2021-11-16:
Yep, it seems to work, I've installed ebusd-21.3_armv7-stretch.deb and seems to work. I will do some testing with the domoticz integration to see if I can visualize some data.

By the way, it will be good that when a new version is installed, /etc/default/ebusd is respected.

jordipalet, 2021-11-16:
It is working fine now, tks!