john30/ebusd #469: CSV zuweisen

m-ask, 2021-11-06:
Hallo!

Ich habe eine Anfängerfrage, aber ich komme leider in meinem Projekt keinen Millimeter nicht weiter.
Bei mir handelt es sich um eine Vaillant geotherm Wärmepumpe bei der ich einerseits die Daten auslesen möchte und dann auch einzelene Parameter verstellen möchte.
Mein Problem ist, dass die Steuerung anscheinend seine Konfiguration nur beim Hochfahren sendet. Somit kann die Konfiguration nicht zugewiesen werden.
Meine Frage, kennt wer das Problem und hat eine Lösung bzw. gibt es eine Möglichkeit eine csv fix zuzuweisen ohne automtische Konfiguration?

Danke
m-ask

m4tze-smart, 2021-11-15:
Hi, ich habe zwar einen anderen WP Hersteller, aber habe bei mir die CSV Files direkt unter /etc/ebusd/ gespeichert und mit --configpath=/etc/ebusd/ geladen. 
Vielleicht hilft es wenn Du die CSV's für Deine WP local zuweist.

john30, 2021-11-16:
das log sieht eher so aus, als könnte der Adapter gar nicht auf den Bus schreiben. Damit wirst Du dann nicht viel Freude haben. Welcher Adapter ist das? Falls mit Poti: nachjustieren!

m-ask, 2021-11-16:
hallo john30
ich nutze den esera eBus Koppler mit USB Anschluss, eigentlich sollte der schreiben können (sagt auf jeden Fall der Support)
die Einstellung mit dem Poti hab ich schon gemacht und prinzipiell kommen die Daten auch sauber rein nur eben das Schreiben will nicht so recht
hat jemand Erfahrung mit dem esera Koppler bzw. welche andere Möglichkeit gibt es (ich hab zwar ebusd einen Adapter reserviert aber das vmtl. noch dauern)
Danke
Christian
@m4tze-smart: local hab ich auch schon probiert funktioniert auch nicht

mwildbolz, 2021-11-18:
> hallo john30 ich nutze den esera eBus Koppler mit USB Anschluss, eigentlich sollte der schreiben können (sagt auf jeden Fall der Support) die Einstellung mit dem Poti hab ich schon gemacht und [...]

Hallo m-ask!
Ich verwende ebenfalls den Esera EBUS-Koppler via USB mit einer Vaillant flexoTherm und habe keine Probleme mit dem Schreiben auf den Bus.
Ich hatte allerdings einmal das Problem, das mein EBUS-Koppler defekt war (eine Diode war durchgebrannt) und dann konnte ich nur mehr lesen! Problem in diesem Zusammenhang war, dass der Koppler längere Zeit am Bus hing, ohne am USB angeschlossen zu sein. Da sind diese Koppler ein bisschen anfällig (lt. Aussage v. Esera).

Vielleicht hilft dir das weiter.

Lg

m-ask, 2021-11-18:
Hallo mwildbolz!
Danke für die Info. Das hört sich sehr ähnlich zu meinem Problem an. Hast du das Problem selbst repariert oder einen neuen bestellt?
Danke
m-ask

mwildbolz, 2021-11-19:
> 
> 
> Hallo mwildbolz! Danke für die Info. Das hört sich sehr ähnlich zu meinem Problem an. Hast du das Problem selbst repariert oder einen neuen bestellt? Danke m-ask

Nachdem mein EBUS-Koppler zu diesem Zeitpunkt nur ein paar Wochen alt war, hatte ich ihn reklamiert und einen Austausch bekommen.
Notfalls müsste aber (meiner Meinung nach) eine Reparatur auch selbst möglich sein. Man müsste nur herausfinden, was das genau für ein Bauteil ist und es entsprechend umlöten (war soweit ich mich erinnern kann ein bedrahtetes Bauteil, vermutlich eine Diode).

m-ask, 2021-11-19:
Hallo mwildbolz!
Danke ich schicke das Teil jetzt auch ein und geb euch dann Rückmeldung ob das Problem dadurch behoben wurde.
lg
m-ask

m-ask, 2022-01-03:
Hallo!
Ich hab den eBUS-Koppler zuück bekomme. Leider funktioniert es immer noch nicht.
Keine Ahnung was ich noch machen soll.
vg
m-ask

2022-01-03 21:00:49.851 [main error] scan config 15: ERR: wrong symbol received
2022-01-03 21:00:52.138 [main error] scan config c0: ERR: wrong symbol received
2022-01-03 21:01:05.881 [main error] scan config 05: ERR: arbitration lost
2022-01-03 21:01:08.088 [main error] scan config 15: ERR: wrong symbol received
2022-01-03 21:01:10.615 [main error] scan config c0: ERR: wrong symbol received
2022-01-03 21:01:23.929 [main error] scan config 05: ERR: arbitration lost
2022-01-03 21:01:27.639 [main error] scan config 15: ERR: arbitration lost
2022-01-03 21:01:30.933 [main error] scan config 80: ERR: arbitration lost
2022-01-03 21:01:34.162 [main error] scan config c0: ERR: arbitration lost
2022-01-03 21:01:48.002 [main error] scan config 05: ERR: wrong symbol received
2022-01-03 21:01:51.710 [main error] scan config 15: ERR: arbitration lost
2022-01-03 21:01:54.998 [main error] scan config 80: ERR: arbitration lost
2022-01-03 21:01:57.263 [main error] scan config c0: ERR: wrong symbol received
2022-01-03 21:02:10.497 [main error] scan config 05: ERR: arbitration lost
2022-01-03 21:02:13.026 [main error] scan config 15: ERR: wrong symbol received
2022-01-03 21:02:17.178 [main error] scan config 80: ERR: arbitration lost

john30, 2022-01-09:
please post some raw=bytes logging so we can check when exactly the communcation fails

m-ask, 2022-01-10:
anbei das Log File
[ebusd.log](https://github.com/john30/ebusd/files/7841601/ebusd.log)

john30, 2022-01-26:
looks like the adapter can't write to the bus.

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30