john30/ebusd #470: D1 Mini wifi works standalone, not on Adaptor 3.0

Nerdvana2, 2021-11-07:
I'm configuring a new Adaptor 3 with the supplied D1 mini, which I have flashed with the latest ebusd-esp firmware.

The D1 consistently connects to wifi when it is not plugged in to the Adaptor 3 board:
`Welcome to eBUS adapter 3, build 20211030
ebusd device string: enh:192.168.0.206:9999
WiFi signal: 62% (-69dBm)
Press return within 5 seconds to change the configuration...
Entering eBUS mode.
`However, plugging the D1 in to the Adapator 3 appears to completely kill wifi:
`Welcome to eBUS adapter 3, build 20211030
ebusd device string: enh:(IP unset):9999
WiFi signal: 0% (-100dBm)
Press return within 5 seconds to change the configuration...
Entering eBUS mode.
`
The jumpers on J12 match the documentation (ie. pins 4&5 connected)

The Adaptor 3 is not yet connected to the ebus. But I don't expect that to have any impact on wifi signal strength.

ebusd behaves exactly as expected, given the wifi issue: connects when the D1 is standalone, not when it is on the Adaptor board.

Any and all suggestions welcome !!!
Thanks in advance
Sean

john30, 2021-11-07:
I'd give a try with another power supply as the Wemos is rather sensitive to the supply.
Did you plug the power to the Wemos or to the adapter?

Nerdvana2, 2021-11-07:
That was it !!   Moved to a 2.1 A USB supply and everything burst in to life :)

Thanks for the very quick response, John..... and thanks for a very elegant, flexible and comprehensive software/hardware combo !!
Cheers, Sean

closed 2021-11-07 by Nerdvana2