john30/ebusd #473: Divider/value leading to 0.0 outputs

ardjanvb, 2021-11-15:
I am new to ebusd, so might be something really stupid I am overlooking. If so, please laugh at me and point me in the right direction after.

Whenever there is a divider defined in the config .csv file, ebusctl is reporting back 0.0 for the specified IDs. When I remove the divider, ebusctl is reporting back the value my unit reads without division.

For example: Aussenlufttemperatur = 7.4 on the screen of unit (in degrees Celsius)

csv divider = 10, ebusctl read Aussenlufttemperatur = 0.0
csv divider = empty, ebusctl read Aussenlufttemperatur = 74

I'm of course trying the get the formatted value to be able to send that to MQTT. What am I missing here?

I am using below csv. 

`# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment,field2,part (m/s),datatypes/templates ,divider/values,unit,comment,field3,part (m/s),datatypes/templates,divider/values,unit,comment,field4,part (m/s),datatypes/templates,divider/values,unit,comment,field5,part (m/s),datatypes/templates,divider/values,unit,comment
### BRINK RENOVENT EXCELLENT 300 & 400 (plus)
##
## This ebus config may work for Ubbink, VisionAIR, WOLF CWL series and some other systems that are identical
## You may need to modify the 7c address in the next 2 lines for your brand/model. Or just uncomment the necessary lines.

## Renovent Excellent 400 (plus) series
*r,kwl,,,,7c,4050,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*w,kwl,,,,7c,4050,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,

## Renovent Excellent 300 series
## *r,kwl,,,,3c,4050,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
## *w,kwl,,,,3c,4050,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,

#### AKTUELLE KONFIGDATEN ####
r,,LuftmengeStufe0,,,,4050,21,aktuell,,UIR,,m³/h,,minimum,,UIR,,m³/h,,maximum,,UIR,,m³/h,,Schrittgroesse,,UIR,,m³/h,,Werkseinstellung,,UIR,,m³/h
r,,LuftmengeStufe1,,,,4050,01,aktuell,,UIR,,m³/h,,minimum,,UIR,,m³/h,,maximum,,UIR,,m³/h,,Schrittgroesse,,UIR,,m³/h,,Werkseinstellung,,UIR,,m³/h
r,,LuftmengeStufe2,,,,4050,02,aktuell,,UIR,,m³/h,,minimum,,UIR,,m³/h,,maximum,,UIR,,m³/h,,Schrittgroesse,,UIR,,m³/h,,Werkseinstellung,,UIR,,m³/h
r,,LuftmengeStufe3,,,,4050,03,aktuell,,UIR,,m³/h,,minimum,,UIR,,m³/h,,maximum,,UIR,,m³/h,,Schrittgroesse,,UIR,,m³/h,,Werkseinstellung,,UIR,,m³/h
r,,BypassTemperatur,,,,4050,04,aktuell,,SIR,10,°C,,minimum,,SIR,10,°C,,maximum,,SIR,10,°C,,Schrittgroesse,,SIR,10,°C,,Werkseinstellung,,SIR,10,°C
r,,BypassHysterese,,,,4050,30,aktuell,,SIR,10,°C,,minimum,,SIR,10,°C,,maximum,,SIR,10,°C,,Schrittgroesse,,SIR,10,°C,,Werkseinstellung,,SIR,10,°C
r,,Bypassbetrieb,,,,4050,1b,aktuell,,UIR,0=auto;1=geschlossen;2=geoeffnet,,,minimum,,UIR,,,,maximum,,UIR,,,,Schrittgroesse,,UIR,,,,Werkseinstellung,,UIR
r,,ZentralheizungWRG,,,,4050,07,aktuell,s,UIR,0=aus;1=ein,,,minimum,,UIR,,,,maximum,,UIR,,,,Schrittgroesse,,UIR,,,,Werkseinstellung,,UIR
r,,DruckungleichgewichtZulaessig,,,,4050,08,aktuell,s,UIR,0=nicht zulaessig;1=zulaessig,,,minimum,,UIR,,,,maximum,,UIR,,,,Schrittgroesse,,UIR,,,,Werkseinstellung,,UIR
r,,FestesDruckungleichgewicht,,,,4050,09,aktuell,,SIR,,m³/h,,minimum,,SIR,,m³/h,,maximum,,SIR,,m³/h,,Schrittgroesse,,SIR,,m³/h,,Werkseinstellung,,SIR,,m³/h
r,,RHSensorVorhanden,,,,4050,32,aktuell,,UIR,,,,minimum,,UIR,,,,maximum,,UIR,,,,Schrittgroesse,,UIR,,,,Werkseinstellung,,UIR,,
r,,RHSensorEmpfindlichkeit,,,,4050,33,aktuell,,SIR,,,,minimum,,SIR,,,,maximum,,SIR,,,,Schrittgroesse,,SIR,,,,Werkseinstellung,,SIR
r,,BeleuchtungDisplay,,,,4050,1d,aktuell,,UIR,,%,,minimum,,UIR,,%,,maximum,,UIR,,%,,Schrittgroesse,,UIR,,%,,Werkseinstellung,,UIR,,%
r,,VorheizRegister_inst,,,,4050,31,aktuell,,UIR,1=ja;0=nein

#### FILTER UND LAUFZEITEN ####
r,,TageMitFilter,,,,4050,22,,,UIR,,bei 90 Tagen kommt Filterwarnung 
r,,LuftmengeMitFilter,,,,4050,23,,,UIR,,tm³ 
r,,LuftmengeFilterSchwellwert,,,,4050,1c,,,UIR,,tm³ 
r,,BetriebsstundenTotal,,,,4050,24,,,UIR,-5,h,,,,UIR,,,,,,UIR,,,,,,UIR,,,,,,UIR
r,,LuftmengeTotal,,,,4050,25,,,UIR,-1000,m³,,,,UIR,,,,,,UIR,,,,,,UIR,,,,,,UIR

#### KONFIGURATION AENDERN ####
w,,LuftmengeStufe0,408001,,,4080,21,,,UIR,,m³/h
w,,LuftmengeStufe1,408001,,,4080,01,,,UIR,,m³/h
w,,LuftmengeStufe2,408001,,,4080,02,,,UIR,,m³/h
w,,LuftmengeStufe3,408001,,,4080,03,,,UIR,,m³/h
w,,BypassTemperatur,408001,,,4080,04,,,UIR,10,°C
w,,EWTT-,408001,,,4080,19,,,UIR,10,°C
w,,EWTT+,408001,,,4080,1A,,,UIR,10,°C
w,,BypassHysterese,408003,,,4080,30,,,UIR,10,°C
w,,Bypassbetrieb,408003,,,4080,1b,,,UIR,0=auto;1=geschlossen;2=geoeffnet
w,,VorheizRegister_an,408003,,,4080,31,,,UIR,0=nein;1=ja
w,,ZentralheizungWRG,408007,,,4080,07,,,UIR,0=aus;1=ein
w,,DruckungleichgewichtZulaessig,408008,,,4080,08,,,UIR,0=nicht zulaessig;1=zulaessig
w,,FestesDruckungleichgewicht,408009,,,4080,09,,,SIR,,m³/h
w,,Ventilatorbetrieb,,,,40a1,,,,ULR,0x0=Feuchteschutz;0x00010001=Reduziert;0x00020002=Normal;0x00030003=Intensiv,,,,,IGN:1

#### SENSORDATEN ####
r,,Feuchte,40220120,,,4022,20,,,UIR,10,% ,RelativeHumidity
r,,FeuchtigkeitsSteigerung,40220121,,,4022,21,,,UIR,0=Error;1=Not Initialized;2=Sensor Not Active;3=PowerUp Delay;4=Normal RH;5=Boost Rising;6=Boost Stable;7=Boost Decending,,HumidityBoostState
r,,SoftwareVersion,40220100,,,4022,00,,,STR:*,,,SoftwareVersion
[... 36 lines cut ...]
r,,PerilexPosition,40220105,,,4022,05,,,UIR,,,PerilexPosition
r,,Contact1Position,4022011B,,,4022,1b,,,UIR,,,Contact1Position
r,,Contact2Position,4022011C,,,4022,1c,,,UIR,,,Contact2Position
r,,EWTStatus,4022011F,,,4022,1f,,,UIR,0=Enabled;1=Disabled,,EWTStatus
r,,EWTT-,40500119,,,4050,19,aktuell,,SIR,10,°C,,minimum,,SIR,10,°C,,maximum,,SIR,10,°C,,Schrittgroesse,,SIR,10,°C,,Werkseinstellung,,SIR,10,°C
r,,EWTT+,4050011A,,,4050,1A,aktuell,,SIR,10,°C,,minimum,,SIR,10,°C,,maximum,,SIR,10,°C,,Schrittgroesse,,SIR,10,°C,,Werkseinstellung,,SIR,10,°C
r,,OptionTemperature,4022011A,,,4022,1a,,,UIR,,,OptionTemperature

#### UNDEFINIERT ####
#r,,undef_01,undef,,,,0a,,,HEX:10
#r,,undef_02,undef,,,,0b,,,HEX:10
#r,,undef_03,undef,,,,0c,,,HEX:10
#r,,undef_04,undef,,,,0d,,,HEX:10
#r,,undef_05,undef,,,,0e,,,HEX:10
#r,,undef_06,undef,,,,0f,,,HEX:10
#r,,undef_07,undef,,,,10,,,HEX:10
#r,,undef_08,undef,,,,11,,,HEX:10
#r,,undef_09,undef,,,,12,,,HEX:10
#r,,undef_10,undef,,,,13,,,HEX:10
#r,,undef_11,undef,,,,14,,,HEX:10
#r,,undef_12,undef,,,,15,,,HEX:10
#r,,undef_13,undef,,,,16,,,HEX:10
#r,,undef_14,undef,,,,17,,,HEX:10
#r,,undef_15,undef,,,,18,,,HEX:10
`

john30, 2021-11-16:
which ebusd version are you using?

ardjanvb, 2021-11-16:
I'm using 21.2 which seems to be the newest version available for armhf.

Any chances of a 21.3 version?

john30, 2021-11-16:
21.3 was published short time ago. anyway I'd prefer if you could try with a build from the latest source

ardjanvb, 2021-11-16:
It seems like I've got the armhf 21.2 version on my Pi Zero W now. Which
architecture version should I install, since there is no armhf version
available of 21.3? I've tried to update via terminal, yet it responded with
an in release for ebusd.

Sorry for the dumb questions, still exploring the wonderful world of Linux.

john30, 2021-11-20:
just use the .deb package of the [release](https://github.com/john30/ebusd/releases/tag/v21.3)

john30, 2022-09-17:
closed due to inactivity

closed 2022-09-17 by john30