john30/ebusd #474: MQTT - Anfängerfrage

gibsonrocker, 2021-11-23:
EBus-Adapter: Version 3 aus dem FHEM-Forum. 

Ich habe folgende Installation: Debian 11, Installation über das Debian-Repository. 

Nun wollte ich mit MQTT anfangen (das ist alles Neuland für mich). Ich habe im log-file von ebusD folgenden Fehler: 
2021-11-22 14:40:31.481 [mqtt error] invalid mosquitto version 2 instead of 1
2021-11-22 14:40:31.482 [main error] error registering data handlers

Hier https://github.com/john30/ebusd/issues/443 sehe ich, dass Du demnächst auch ein Update für das Debian-Repository machen willst (https://github.com/john30/ebusd/releases/tag/v21.3). 

Soweit scheint der Fehler „gelöst“ zu sein, wenn ich alles richtig verstanden habe. Ich warte einfach auf das Update. 
Jetzt habe ich aber noch eine blöde Frage (als ebus-Adapter und MQTT Neuling): 

Ich habe den MQTT Broker auf einer eigenen virtuellen Maschine mit Debian. Muss ich jetzt auf der Debian-Maschine auf der ebusD läuft noch irgendwas installieren, wenn ich die Version von ebusD mit MQTT Unterstützung installiert habe? Braucht es da noch was um von der ebusD-Maschine auf den Broker zu publishen? Oder ist das in der eBusD Installation alles drin?

john30, 2021-11-25:
installiere doch einfach das [release file](https://github.com/john30/ebusd/releases/tag/v21.3).
das ist alles schon drin, die Kommunikation geht dann einfach über TCP

gibsonrocker, 2021-11-26:
Ja, vielen Dank. Das habe ich jetzt gemacht. Dank Deiner Anleitung kann das auch ein nicht Linux-Gott. Vielen Dank! 

Wenn aber jetzt ein Update kommt dann muss ich das alles noch einmal machen und dann die Config neu machen, oder? Übers Repository hätte ja die Möglichkeit mit apt-get update ohne Verlust von Einstellungen, oder?

john30, 2021-11-28:
apt-get update macht auch nichts anderes, als das neue Paket runterzuladen und zu installieren

john30, 2021-12-18:
das [repo](https://github.com/john30/ebusd-debian/) ist inzwischen aktualisiert

closed 2021-12-18 by john30