john30/ebusd #475: Problem Ebusd in Dockercontainer on Raspberry Pi 4

peterolaf, 2021-11-23:
I want to use a Ebusd docker container to connect a raspberry Pi 4B to a Brink Renovent 300 unit. I did try several configs , but I do not see any useful output and many errors. What do I wrong.... Please look into my debug log.

Thanks a lot for your help.

Peter

[Debuglog-ebusd-docker.txt](https://github.com/john30/ebusd/files/7590328/Debuglog-ebusd-docker.txt)

peterolaf, 2021-11-23:
Sorry: wrong format logfile, new one txt format.....
[Debuglog-ebusd-docker.txt](https://github.com/john30/ebusd/files/7590476/Debuglog-ebusd-docker.txt)

john30, 2021-11-25:
you need to provide the right CSV files in the config path which seems not to be the case. I don't know if there are CSVs for brink devices already,

peterolaf, 2021-11-26:
Thanks for your answer. I am using a Brink CSV file for the Renovent 300 type from GitHub 

[
](https://github.com/dstrigl/ebusd-config-brink-renovent-excellent-300#ebusd-for-brink-renovent-excellent-300)

The Ebusd can read this configfile with no errors :

2021-11-23 18:17:43.797 [main notice] ebusd 21.3.v21.3 performing configuration check...
2021-11-23 18:17:43.802 [main notice] found messages: 77 (0 conditional on 0 conditions, 0 poll, 0 update)

So I think the problem is something in the communication.

because I see in the debug log  a lot of communication errors:

2021-11-23 14:30:00.248 [main debug] <<< done
2021-11-23 14:30:01.651 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2021-11-23 14:30:01.698 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-11-23 14:30:02.694 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-11-23 14:30:02.767 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-11-23 14:30:04.730 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-11-23 14:30:04.850 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-11-23 14:30:04.978 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-11-23 14:30:05.098 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-11-23 14:30:05.171 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2021-11-23 18:04:41.931 [main debug] performing regular tasks
2021-11-23 18:04:41.931 [main debug] >>> read -f Softwareversion
2021-11-23 18:04:41.932 [bus info] send message: 313c40220100
2021-11-23 18:04:41.932 [network debug] [00002] wait for result
2021-11-23 18:04:41.952 [bus debug] start request 31
2021-11-23 18:04:41.952 [bus debug] arbitration start with 31
2021-11-23 18:04:42.002 [bus debug] arbitration lost
2021-11-23 18:04:42.002 [bus debug] ERR: arbitration lost during ready, retry
2021-11-23 18:04:42.050 [bus debug] ERR: SYN received during receive command, switching to ready
2021-11-23 18:04:42.098 [bus debug] start request 31
2021-11-23 18:04:42.098 [bus debug] arbitration start with 31
2021-11-23 18:04:42.147 [bus debug] arbitration lost
2021-11-23 18:04:42.147 [bus debug] ERR: arbitration lost during ready, retry
2021-11-23 18:04:42.194 [bus debug] ERR: SYN received during receive command, switching to ready
2021-11-23 18:04:42.242 [bus debug] start request 31
2021-11-23 18:04:43.596 [bus debug] start request 31
2021-11-23 18:04:43.596 [bus debug] arbitration start with 31
2021-11-23 18:04:43.645 [bus debug] arbitration lost
2021-11-23 18:04:43.645 [bus debug] notify request: ERR: arbitration lost
2021-11-23 18:04:43.646 [bus error] send to 3c: ERR: arbitration lost
2021-11-23 18:04:43.646 [bus error] send message part 0: ERR: arbitration lost
2021-11-23 18:04:43.646 [main debug] <<< ERR: arbitration lost

Could it caused by a syntactic right config-file but a different BrinkDevice? 
Is it possible to readout the hex values  from the Brink device directly from the USB port to check which type Brinkdevice is connected?

thanks a lot.....

john30, 2021-11-28:
did you bind the USB device correctly in the docker run command? check for the rights on the device within the container.
and maybe you want to add some "--latency=50" or so to the start params of ebusd

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30