john30/ebusd #476: Where i can buy adapter ebus to connect with Vaillant

pepsonEL, 2021-11-24:
Hi. Where i can buy adapter 3 ebus to connect it with my Vaillant ?

pepsonEL, 2021-11-24:
thanks.... but on page https://adapter.ebusd.eu/index.en.html  it is no available.

fhackenberger, 2021-11-24:
You can only reserve one from the next batch, but I got the information (email) that they won't be available before next year, so went with the commercial one from ESERA in the mean time (also linked on the hardware page).

pepsonEL, 2021-11-24:
First i reserved universal adapter and i get some days ago email to pay for it, but i change conception to but only with WIFI and write to developer but no respond.

john30, 2021-11-25:
I [replied already](https://github.com/john30/ebusd-esp/issues/54#issuecomment-974683949) days ago to your first comment, maybe you should read it?
please stop posting further duplicates of the same question in different forums/repos, this is really annoying and wasting my time.

closed 2021-12-18 by john30