john30/ebusd #479: Sprache aus der CSV Datei richtig übermitteln

gibsonrocker, 2021-11-28:
Hi,

Ich habe eine Frage die ich per Google nicht rausfinde. 

EbusD läuft. MQTT läuft. Json wird übermittelt. Alles fast gut. 

Leider sind alle Werte in Englisch. In der CSV-Datei zum Beispiel auch geladen wird (Logfile sagt):
2021-11-28 14:54:41.564 [main notice] read scan config file vaillant/15.700.csv for ID "70000", SW0209, HW4103
stehen im "Comment" hinten den englischen Bezeichnungen die deutschen Kommentare bzw. Übersetzungen. Kann ich jetzt den ebusD irgendwie dazu bekommen die Werte auf Deutsch zu übermitteln? 

Hab sowas für die Config gefunden (--configlang=DE). Das ändert aber nichts (Service hab ich nach der Änderung neu gestartet). 

Was mache ich falsch?

john30, 2021-12-12:
Im Moment gibt es auf dem webservice nur de. ebusd unterstützt bereits das multilinguale Format, aber ebusd-configuration muss erst in dieses gebracht werden. Erst dann klappt auch die Einstellung mit configlang.

john30, 2022-11-13:
duplicate of #344

closed 2022-11-13 by john30