john30/ebusd #480: Err:invalid position on csv decode

bmoesdijk, 2021-11-29:
Hello 

Can anybody help me with understanding the following error I receive in my logging.
2021-11-29 13:19:22.378 [update info] received MM cmd: 10030800088025008080c00080
2021-11-29 13:19:22.379 [update error] unable to parse write Allure AC Bedrijfsdata2 from 10030800088025 00 80 80c00080 / : ERR: invalid position
I'm looking for the outside temperature which should be located in the 8-9 byte -> 0080

I have a .csv file with the following entry which should match and decode the message according to https://ebus-wiki.org/lib/exe/fetch.php/ebus/spec_prot_7_v1_6_3_d.pdf (§3.4.1) Or am I mistaking?

w,,Bedrijfsdata2,Bedrd2,,,0800,,Field1,,D2B,,,,Field2,,D2B,,,,Field3,,D1B,,,,Field4,,UCH,0x00=WTW_aktiv;0x01=Verwarmingscircuit_activ,,,Field5,,D2B,,,,Field6,,skip,,,,Flied7,,skip,,,

john30, 2021-12-18:
if you set the ZZ to 03 (for the master), then it should work. the default part for the fields is determined from the target. if the target is a master, then the fields are assigned to the master part of the message by default. you could also set the part to "m" for master for every field.

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30