john30/ebusd #481: How to emulate controller

luzik, 2021-12-04:
Is it possible to emulate temperature controller like Calormatic 450?

I want to report lowest temperature in home rooms.

What does Calormatic do ? Is it actively report:
- room temperature
- temperature set
- heat curve
- schedule
Or just respond to EcoTEC as a slave?

Or maybe Calormatic does calculate FlowTempDesired
Does anybody know formula to calculate FlowTempDesired ?

k41c, 2022-02-12:
Hi luzik,

on my boiler I need to set the boiler temperature to max. So the controller will send commands to stop the boiler heating when not needed. The controler get the feedback of the heatingwater temperature and will controll it by calculation, in my case, by outside temperature and heatcurve. If you have a room temperature then this will overdrive the previusly calculated value if its to warm or cold. Allso dont forget the start and stop ramps if you are using a night reduced temperature e.g. 18°C.
At the moment I can't tell if its master-master or master-slave comunication. But I will report as far as I'm ready with my adapter. My plan is to simulate a roomtemperature. And switch to different rooms as reference, so day its livingroom and night sleeping room.

PureDJ, 2022-02-27:
I have a ColorMatic 360 and I have successfully taken over control, actually in both ways.
I made my own controller with PID, outside temp (with a heating curve), away mode, schedule with early start (room is at decired temp when the schedule is set, not start heating when the schedule start), delayed lower temp when the TV is on etc.

I can set the control function into 2 modes, tell the ColorMatic what temp I want and let it handle the controls (in my case more gas consumption, because it only know 2 modes, full heat or no heat) or let my own PID controller do the work and read out the actual temp.

I'm not sure what you want to do.
But telling the ColorMatic the desired temp you can do with writing ComfTempDesired
If you want full control you can set OperatingMode to 0, but be carefull! you need to write SetMode(override) at least every second otherwise the heater will switch to on/off thermostat mode and Valiant is telling users to have a wire loop over the contact input it will put your heater in full on!
 
 Hope this helps.

ostauss, 2022-10-24:
Dear @PureDJ, 
I'm really interested in the details of your topic “create own controller”. 
Currently, I “control the controller (f37)” but I want to “replace” this simple controller with an advanced (outside, heating curve, etc). 
Maybe you can share your “learnings”? 
(i also use node-red...)
regards Oliver

netweaver1970, 2022-10-24:
@PureDJ I'm also very interested in your own implementation. Right now I have a Migo (Bulex/Vaillant rebranded Netatmo smart thermostat), which has internet based outside temp supply, heating curve setup, self learning PID parameters, pre-heating etc.. I'm controlling the Migo via HomeAssisant, using the Bulex Cloud service. And I have Zigbee TRVs on my radiators. As such I'm not unhappy with the Migo factory+ HA + Smart TRVs solution but I'm always interested in seeing how other people solved it, in case it's better. My eBus adapter + HA integration is currently more a monitoring and (if needed/ deemed better/valuable) a 
way to take control of the system.

PureDJ, 2022-10-24:
Funny thing my HA crashed, so I'm setting up everything at the moment, it's been a while, but what I have accomplished is a complete PID controller, outside temp I could not read out, but I used a weather forcast site to get the actual outside temp.
I'm going to rewrite some controllers, I found out that my automatic temp lowering not always worked correctly (I use my wifi router to check if one of our families phone is connected to the wifi, if not, it lowers the setpoint 1 degree, if it has been lowered and one phone is connected, it raises the setpoint temp by 1 degree (celcius)
I can share the node red setup if you want, but let me correct it first, it's a bit messy now

john30, 2022-10-30:
I was forced to do something similar a couple of years ago as my heat pump controller crashed and I was not able to get it back to life. did a lot of replaying recorded messages from before its death but I could not convince the heat pump to do something useful. really awful. so, just wanted to mention that it seems to be one of the most complicated and challenging tasks and I more or less gave up (and ended in buying a new controller)