john30/ebusd #482: Suspicious debug log entries

majowi5, 2021-12-07:
Hi John,
During optimizing my ebusd setup I logged the following debug messages( see below):
Suspicious is the missing of an error, if you log just “info” and not “debug”. This situation lasts in this example for about 15 s. I am rather sure of missing traffic during this time period.
My log setup is bus: debug, all others:info
I logged quite a lot of these occurrences. Do you have any idea how this could occur?

```
2021-12-07 00:19:54.606 [update info] received MS cmd: 1008b512020000 / 00
2021-12-07 00:19:54.607 [update notice] received update-write bai StatusCirPump QQ=10: off
2021-12-07 00:19:56.151 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:19:56.899 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:19:57.647 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:19:58.395 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:19:59.146 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:19:59.894 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:20:00.640 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:20:01.388 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:20:02.138 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:20:02.888 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:20:03.636 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:20:04.386 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:20:05.137 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:20:05.883 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:20:06.635 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:20:07.385 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:20:08.131 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:20:08.879 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-07 00:20:09.651 [update info] received MS cmd: 1008b510090000006effff05ff00 / 0101
2021-12-07 00:20:09.652 [update notice] received update-write bai SetMode QQ=10: auto;0.0;55.0;-;1;0;1;0;0;0
```

john30, 2021-12-12:
do you see any successful traffic with the master data part?
it is quite normal that masters send messages to slaves that don't answer these. Especially for Vaillant devices.

majowi5, 2021-12-12:
Before and after that bulk of errors traffic is normal with almost no errors. See also the example above. There is successfull master data traffic just after the bulk of errors.
I will switch on "lograwdata" to get closer to the root cause.

majowi5, 2021-12-26:
These log entries are caused by the VRC470. VRC470 polls periodically nonexistant devices. (ZZ=35,50,75,0a,ec,ed) Message is
QQZZ070400 (=Identification). As these devices do not answer, ebusd logs _ERR: read timeout during receive command ACK, switching to skip._

closed 2021-12-26 by majowi5