john30/ebusd #483: 21.3 Problems

Revilo91, 2021-12-19:
Hello, 

I have problems after update from 21.2 to 21.3 

ebusd couldn't "find" the other participants of the ebus (in 21.2 there was able found) 

config:

> EBUSD_OPTS="--scanconfig=full -d /dev/ttyAMA0 --latency=50 --httpport=8889 --htmlpath=/var/ebusd/html --loglevel=debug"

ebusctl i

> version: ebusd 21.3.v21.3
> update check: OK
> device: /dev/ttyAMA0
> signal: acquired
> symbol rate: 23
> max symbol rate: 95
> reconnects: 0
> masters: 1
> messages: 11
> conditional: 0
> poll: 0
> update: 4
> address 31: master #8, ebusd
> address 36: slave #8, ebusd

logfile: 
> ....
> 2021-12-19 18:44:27.242 [main notice] ebusd 21.3.v21.3 started with full scan on device /dev/ttyAMA0
> 2021-12-19 18:44:27.243 [main info] loading configuration files from http://ebusd.eu/config/
> 2021-12-19 18:44:27.280 [main info] reading templates /
> 2021-12-19 18:44:27.319 [main info] read templates in /
> 2021-12-19 18:44:27.319 [main info] reading file memory.csv
> 2021-12-19 18:44:27.357 [main info] successfully read file memory.csv
> 2021-12-19 18:44:27.357 [main info] reading file broadcast.csv
> 2021-12-19 18:44:27.396 [main info] successfully read file broadcast.csv
> 2021-12-19 18:44:27.396 [main info] read config files
> 2021-12-19 18:44:27.397 [bus notice] bus started with own address 31/36
> 2021-12-19 18:44:27.398 [main info] registering data handlers
> 2021-12-19 18:44:27.398 [main info] registered data handlers
> 2021-12-19 18:44:27.469 [bus notice] signal acquired
> 2021-12-19 18:44:32.170 [network info] [00001] HTTP connection opened xxxxxxxxxxx
> 2021-12-19 18:44:32.171 [network debug] [00001] wait for result
> 2021-12-19 18:44:32.171 [main debug] >>> GET /data
> 2021-12-19 18:44:32.172 [main debug] <<< HTTP/1.0 200 OK
> Content-Type: application/json;charset=utf-8
> Content-Length: 628
> Server: ebusd/21 ...
> 2021-12-19 18:44:32.172 [network info] [00001] connection closed
> 2021-12-19 18:44:33.172 [network debug] dead connection removed - 0
> 2021-12-19 18:44:37.172 [main debug] performing regular tasks
> 2021-12-19 18:44:37.172 [main notice] starting initial full scan
> 2021-12-19 18:44:37.172 [bus info] scan 02 cmd: 3102070400
> 2021-12-19 18:44:47.173 [main debug] performing regular tasks
> 2021-12-19 18:44:57.173 [main debug] performing regular tasks
> ....

Hopefully, this can help you.

Thanks in advance

Revilo91, 2021-12-21:
Strange... 

I own the Adapter3 and i changed only the connector from green to blue (see yellow marked area) --> this fixed my problem
[image]

closed 2021-12-21 by Revilo91

john30, 2021-12-23:
you didn't tell ebusd that it is a device working in enhanced mode. using "-d enh:/dev/ttyAMA0" instead of "-d /dev/ttyAMA0" would have solved it

Revilo91, 2021-12-24:
Thanks for this hint. Now it works better. Thank you very much!