john30/ebusd #484: ebusd | no switch on mixer cycle | wolf CGB-2 (AM), BM2 wall mounted, DWTM

eh-gh, 2021-12-22:
I've sent a support request to fhem, I got no answer yet
https://forum.fhem.de/index.php?topic=123866.0
maybe here:
my heating system as follows
CGB-2 (AM), BM2 wall mounted, DWTM as mixer
configuration: 1 mixer + 1 hot water
the ebusd configuration comes from a user alpha1974

My problem is that I can switch on the mixer cylce vie ebusctl.
I do not grab any valuable msg from the ebus stream in case I switch on/off manually the heating (over mixer1) at BM-2

I can switch e.g. on the production of hot water via:
ebusctl hex 3550230900752701005D010010
the strange is that I can not see this message either when I use grab during switching on via ebusctl 

what do I wrong? tnx in advance

currently:
ebusctl i output
version: ebusd 21.3.v21.3-114-g43d12b7
update check: revision v21.3 available, broadcast.csv: different version available
device: /dev/ttyebus
access: *
signal: acquired
symbol rate: 58
max symbol rate: 205
min arbitration micros: 5
max arbitration micros: 17
min symbol latency: 4
max symbol latency: 4
reconnects: 0
masters: 5
messages: 38
conditional: 0
poll: 0
update: 6
address 03: master #11
address 08: slave #11, scanned "MF=Wolf;ID= !;SW=8000;HW=0000", loaded "wolf/08.csv"
address 30: master #3
address 31: master #8, ebusd
address 35: slave #3, scanned "MF=Wolf;ID=;SW=0204;HW=0100", loaded "wolf/35.csv"
address 36: slave #8, ebusd
address 51: slave, scanned "MF=Kromschroeder;ID=  ;SW=0171;HW=0150"
address 70: master #4
address 75: slave #4, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=8080"
address f1: master #10
address f6: slave #10, scanned "MF=Kromschroeder;ID=  ;SW=0171;HW=0150"

john30, 2021-12-23:
you'd need to provide the CSVs and some ebusd logging

eh-gh, 2021-12-24:
Hi John,
time of the actions above ~ 2021-12-22 20:00-22:00 CET, therefore I cut the log
[ebusd_evidence.tar.gz](https://github.com/john30/ebusd/files/7772805/ebusd_evidence.tar.gz)

thanks for helping
EH

eh-gh, 2021-12-25:
Hi John,
maybe the followings help more with detailed debug, i did not change anything in csv files only more debug for daemon is set
test scenario:
- heating, boiler in standby
- start ebusd
- switch hot water creation manually on BM2 (mode: continuous) /no hex msg with pattern 5023 found in logs/
- switch hot water creation  to standby with "ebusctl hex 3550230900752700005D010010", the msg is shown in logs

my goal is to switch on/off the mixer at least with hex commands, but similar to hot water, I do not see any related hex message in logs in case I change the status of mixer via BM-2

thanks again for your time in advance
EH
[ebusd.log](https://github.com/john30/ebusd/files/7775753/ebusd.log)
[ebusd_raw.log](https://github.com/john30/ebusd/files/7775754/ebusd_raw.log)

config file for ebusd service:
```
/usr/bin/ebusd --scanconfig -d /dev/ttyebus -p 8888 -l /var/log/ebusd.log --enablehex --enabledefine -c /home/pi/alpha1974 --accesslevel=* --latency=20 --mqttport=1883 --mqttjson --mqtthost=127.0.0.1 --mqtttopic=ebusd/%circuit/%name --lograwdatafile /var/log/ebusd_raw.log --lograwdata --logareas=all --loglevel debug
```

john30, 2022-01-26:
can't open your files

john30, 2022-03-19:
if you want to check for incomplete/bad messages, then you'd have to do raw logging with bytes and check the log exactly at the time you expect the change to happen.
without a concrete hint to the time, I can't help

eh-gh, 2022-04-16:
please find the logs attached:
[2022-04-15.zip](https://github.com/john30/ebusd/files/8499904/2022-04-15.zip)

scenario
- start ebusd
- switch on hot water creation
- switch off hot water creation

- switch on heating creation
- switch off heating creation
- stop ebusd

all actions are done manually on BM-2 controller

q is still: I do not find any information what happens when I switch on/off heating circle on BM-2

hint:
I can switch on/off hot water program via ebus:
```
# Brauchwasserprogramm BM-2 setzten works
set Brauchwasserprogramm_Standby cmd {"hex 3050230900752700005D010010"}
set Brauchwasserprogramm_Auto    cmd {"hex 3050230900752701005D010010"}
set Brauchwasserprogramm_Allways cmd {"hex 3050230900752702005D010010"}
```

thanks for you time

john30, 2022-06-19:
if you can't find any related message logged by ebusd at the time you do the actions on the BM-2 controller, then it might just not send any message on the bus. or it sends an incomplete/invalid message. you can check that more easily by using the `--lograwdata` flag so that ebusd will log all received data between SYN symbols to the ebusd.log file. Then also note the exact time when you initiate an action on the controller (making sure your time is accurate) so that you can check the log afterwards.

eh-gh, 2022-08-06:
sorry for the late answer, i have limited physical access to the system

please find the logs

- switch to continuous heating at BM2: 2022-08-06 19:02:30
- switch of (standby): 2022-08-06 19:03:30

can you tell me which part should describe the actions above?

what I found e.g.

[ebusd.log](https://github.com/john30/ebusd/files/9275484/ebusd.log)
[ebusd_info.txt](https://github.com/john30/ebusd/files/9275485/ebusd_info.txt)
[ebusd_raw.log](https://github.com/john30/ebusd/files/9275486/ebusd_raw.log)

john30, 2022-09-17:
hm, 71 (76) is not listed in the info, so it probably does not answer the ID query. 
but I'm not sure that this is the right message and since so little messages are decoded, it is really hard to tell. it would require a deep analysis of the messages seen to gather the meaning.
however, quite close to the time you did the switching, there are two messages that might be related:
```
2022-08-06 19:02:30.591 [update notice] received write feuerung betrd QQ=30: aus;Verbraucheraus;5.00;-;-;10.0;8
2022-08-06 19:03:34.635 [update notice] received write feuerung betrd QQ=30: Heizen;Verbraucheraus;34.00;-;-;10.0;8
```
so maybe that one is just badly defined?

eh-gh, 2022-09-25:
hi @john30 
thanks for the update,
Ive just changed the DWTM mixer module to a newer one MM2, i thought this could help

I updated the ebusd to the latest and i used the configuration from the ebusd-configuration repo.

the info is:
`root@raspberrypi:/home/pi# ebusctl i
version: ebusd 22.4.v22.4-13-gf9773b1
device: /dev/ttyebus
access: *
signal: acquired
symbol rate: 44
max symbol rate: 123
min arbitration micros: 8
max arbitration micros: 94
min symbol latency: 2
max symbol latency: 14
reconnects: 0
masters: 5
messages: 66
conditional: 0
poll: 0
update: 6
address 03: master #11
address 08: slave #11, scanned "MF=Wolf;ID= !;SW=8000;HW=0000", loaded "wolf/08.csv"
address 30: master #3
address 31: master #8, ebusd
address 35: slave #3, scanned "MF=Wolf;ID=;SW=0204;HW=0100", loaded "wolf/35.csv"
address 36: slave #8, ebusd, scanning
address 51: slave, scanned "MF=Kromschroeder;ID=  ;SW=0100;HW=-"
address 70: master #4
address 75: slave #4, scanned "MF=Kromschroeder;ID=  ;SW=0100;HW=-"
address f1: master #10`

If i change the state of the mixer cycle, it still can't be found in the raw log.

As I understood each element should have 2parts:1 master + 1 slave
In my config the component 51 has no master, the component f1 has no slave pair.

Theory: as I could see from other post, the Nr. 51 would be the mixerNr1, so there should be a master with Nr. 46 and if I could listen to, I would see the status change messages to the mixer.

- Could it be right?
- If yes, what kind of option I still have to find the right msgs?
- Do you think with an adapter3 I could have better chances?

tnx again
eh

`/usr/bin/ebusd -d /dev/ttyebus -p 8888 -l /var/log/ebusd.log --enablehex --enabledefine --accesslevel=* --latency=50 --mqttport=1883 --mqttjson --mqtthost=127.0.0.1 --mqtttopic=ebusd/%circuit/%name --lograwdatafile /var/log/ebusd_raw.log --lograwdata --logareas=all --loglevel debug --lograwdatasize=100 --dumpfile=/tmp/ebusd_dump.bin --dump --dumpflush --configpath=/home/pi/repos/ebusd-configuration/ebusd-2.x.x/de --scanconfig=full`

john30, 2022-10-03:
> As I understood each element should have 2parts:1 master + 1 slave In my config the component Nr. 51 has no master, the component Nr. f1 has no slave pair.

no, only 25 addresses are master and have an associated slave address, the rest are slave-only addresses, i.e. can't send anything to the bus on their own

> Theory: as I could see from other posts, the Nr. 51 would be the slave mixerNr1, so there should be a master with Nr. 46 and if I could listen to, I would see the status change messages to the mixer.

no, 51 is slave-only

>     * Do you think with an adapter3 I could have better chances?

since the missing messages are the problem, a better adapter would only yield less errors in communication, but that's not you problem here. so I'd say: no

eh-gh, 2022-10-04:
Thanks @john30 
clear, most probably I have to buy a Wolf Home (Link|Pro) Adapter
i close the issue

closed 2022-10-16 by john30