john30/ebusd #486: cant install addon

Kentrp, 2021-12-24:
Hi When i install addon from repository in HA i get error "Failed to install add-on
The command '/bin/ash -o pipefail -c apk add --no-cache build-base argp-standalone git mosquitto-dev cmake' returned a non-zero code: 5" can someone help to resolve this?

john30, 2021-12-24:
can't tell anything about the HA addon, please contact the author.
for ebusd, I can't reproduce. please pull the latest image using `docker pull john30/ebusd`

Kentrp, 2021-12-24:
kent@raspberrypi:~ $ docker pull john30/ebusd
Using default tag: latest
latest: Pulling from john30/ebusd
Digest: sha256:0e5232148c08bbc073dda25ccbc3436c0871fd01a9446865dfa20ab006f3981f
Status: Image is up to date for john30/ebusd:latest
docker.io/john30/ebusd:latest

john30, 2021-12-24:
your time is invalid, please check the system time

Kentrp, 2021-12-24:
kent@raspberrypi:~ $ date
п'ятниця, 24 грудня 2021 14:58:54 +0200

How to update date in docker thank you

Kentrp, 2021-12-24:
Now i have other problem the container starts and stopped with "Stopped for a few seconds with exit code 22"

john30, 2021-12-26:
great!

closed 2021-12-26 by john30