john30/ebusd #487: Vaillant VRC 720 with oder ID

snooze83, 2021-12-24:
Hello Friends.
I get a new VRC 720 from Vaillant. My problem is, the ID from the panel is not 72000. 
My panel id is:
"MF=Vaillant;ID=CTLV2;SW=0514;HW=1104"
Iam sure it works with the 15.700.csv, but how can i "say" ebusd that it must take the 15.700.csv?

Greetings Andreas

snooze83, 2021-12-25:
Hello Friends,
i solved the problem. I have created a hyperlink to 15.700.csv named 15.ctlv2.csv.

Merry Christmas

closed 2021-12-25 by snooze83

klaus77777777, 2023-04-01:
Hello snooze83,

thx for you hint after I read your entry I got it and I was able to solve my issue.
many thx

regards
Klaus

mainmind83, 2023-05-15:
> Hello snooze83,
> 
> thx for you hint after I read your entry I got it and I was able to solve my issue. many thx
> 
> regards Klaus

Try to copy file 10.700.csv and rename to15.ctlv2.csv

stingone, 2023-06-19:
> 15.ctlv2.csv

just a check. my id is Vaillant;CTLV2;0514;1104. i copied the 15.720 and renamed it to 15.ctlv2.csv however its not doing anything and not creating a new device under MQTT with the add-on

snooze83, 2023-06-20:
Hello stingone,
dont rename it,  make a hyperlink to 15.700.csv named 15.ctlv2.csv.

Greetings snooze

fifalars, 2023-10-13:
He means a symbolic link: ln -s on Linux

majuss, 2024-01-21:
@snooze83 how did you get your panel ID?

klaus77777777, 2024-01-21:
Hello,

don’t know if I got your question

I did a
ebusctl info and got
…
address 26: slave, scanned „MF=Vaillant;ID=VR_71;SW=0203;HW=5904", loaded "vaillant/26.vr_71.csv“
…

and than I named the file beginning with the address 26 and than the id vr_71 and vr_71 was detected by ebusctl, result of the name is 26.vr_71.csv
good luck

regards

Klaus

john30, 2024-02-11:
it is ongoing: https://github.com/john30/ebusd-configuration/pull/266

TeXiCiTy, 2024-03-27:
Current sensoComfort thermostat is at version ID=CTLV**3** already...

akloud, 2024-12-15:
I created a symbolic link for the CTLV3 pointing to the CTLV2. This worked nicely. 

Still, I have no clue about the general workflow how to track down undecoded values and how to add them to the configuration afterwards.

@john30, would that not be a good addition to the Wiki? Maybe it could encourage more people to dig into the configuration at this level.

john30, 2025-01-10:
ctlv3 is in [the ](https://ebus.github.io/next/)