john30/ebusd #488: Adapter 3 + wemos d1 mini = Vaillant VSC206 problems read hardware and configuration

pepsonEL, 2021-12-25:
Hi. I buy from you Adapter 3 with Wemod D1 Mini. 
My Hardware is Vaillant EcoCompact VSC206, Multimatic VRC700/6, module VR71, remote control VR91 and Vaillant Gateway VR920.
I flashed wemos by latest FW Build: 20211205
Adapter works on IP 192.168.4.39 port 9999
Then on Synology i install Docker and then download latest container ebusD 21.3 and run it by command:
`-f --scanconfig -d enh:192.168.4.39:9999 --latency\=8000 --mqtthost\=192.168.4.6 --mqttport\=1883 --mqttuser\=MY_LOGIN --mqttpass\=MY_PASSWORD`

Then in MQTT Explorer i see sometimes scan BAI, 700, VR71 , VR91 VR920 but sometime i dont see example 700... or VR920. What i do wrong ? Please help me....
Screen below show and logs from event from docker.....
[image]
[image]
[image]
[image]

LOgs event from docker ebusd
https://pastebin.com/2fiv3sgi

Please help me .....

pepsonEL, 2021-12-25:
[image]

Christoph1111, 2021-12-25:
Did you edit the 15.700.csv to your needs?
Without polling, there is no output.

pepsonEL, 2021-12-25:
No i dont edit any CSV. I use option --scanconfig and config should be automatically use... ?

**Without polling, there is no output.**

What you mean ?

Christoph1111, 2021-12-25:
When you "ebusctl i", you see the files which are loaded for decoding. In my case it's 15.700.csv. Lines in this file start with eg.
`r;w,,PrFuelSumHwcThisMonth,ThisMonthsFConsumptionHwc,,,,5100,,,energy4,,,`.
If you want this parameter (PrFuelSumHwcThisMonth) polled, you have to add a priority number from 1 to 9 after the leading **r**. After reloading `ebusctl reload` this parameter is polled and you can watch the output in the log, in the http page or in MQTT browser depending on your `--pollinterval=xx` in EBUSD_OPTS.

For testing (before scrambling your config), just put a `ebusctl read -f PrFuelSumHwcThisMonth` to control that ebusd has read the appropriate config file.

pepsonEL, 2021-12-25:
But if i only run MQTT Explorer and sometime show more info about VRC700 but sometime not show ? Why ? It is run with automatically scan config....

Christoph1111, 2021-12-25:
Don't know your hardware and without Console it's hard to debug.
Try to change the 15.700.csv. If there is no change, maybe your hw is the problem.

pepsonEL, 2021-12-25:
But how change 15.700.csv in docker ?

Christoph1111, 2021-12-25:
Don't know docker, I have classical linux. Without console information, I can't help you.
When ebusd cannot identify your master because of eg. triggering problems on the wires, the csv is not loaded and you have no output. Sometimes it works, sometimes not.

pepsonEL, 2021-12-25:
Problem on wires ? But all hardware from Vaillant works ok. No problem.... But what problem with wires ?

Christoph1111, 2021-12-25:
The ebus wires to your D1?

pepsonEL, 2021-12-25:
But what you ask?

pepsonEL, 2021-12-25:
My Adapter is connected to PLUG in my boiler VSC206.... by wire 2x0,75mm2

john30, 2021-12-26:
since the synology is not the fastest device, I'd recommend to increase the latency value to 40 ms, i.e. use "--latency\=40" instead of "--latency\=8000" (the factor 1000 is not needed any more)

john30, 2022-09-18:
closed due to inactivity

closed 2022-09-18 by john30