john30/ebusd #490: Need help for connection with okofen TEM

nicoalas38, 2021-12-28:
Hi all, I have currently an issue with a TEM 2945 C3 K Master. 
i am using the adapter 3.0 with a RPI . 
I have configured the TEM as master and allowing 1 slave. 
i can see that that the ebusd is working , but can’t progress any further. May the system is not compatible with my configuration.

ebusctl info :
pi@raspberrypi:~ $ ebusctl info
version: ebusd 21.3.v21.3-130-g13221e2
update check: revision v21.3 available
device: /dev/ttyAMA0, enhanced
signal: acquired
symbol rate: 27
max symbol rate: 61
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd

Nothing else and the log give me the following : (var/log/ebusd.log)

2021-12-28 12:17:30.040 [bus error] signal lost
2021-12-28 12:17:31.271 [bus notice] signal acquired
2021-12-28 12:17:31.319 [bus notice] ...<001800060c0000001e000000000006031e1803061e0c180c18001e06001e00...
2021-12-28 12:17:33.053 [bus notice] ...<001800060c0000001e000000180c000006001e0018030000001e001e000018...
2021-12-28 12:17:33.903 [bus notice] ...<00060c0000001e000000060c001e1e060018000006001e00001800180c0600...
2021-12-28 12:17:35.029 [bus error] signal lost
2021-12-28 12:17:36.426 [bus notice] signal acquired
2021-12-28 12:17:36.436 [bus notice] ...<0000001e0000000000000000000000000000001e00001800060c0000001e00...
2021-12-28 12:17:39.087 [bus error] signal lost
2021-12-28 12:17:39.896 [bus notice] signal acquired
2021-12-28 12:17:39.896 [bus notice] ...<001e0c0018001e00001800180c06000000001e001e0c0000001e0000180006...
2021-12-28 12:17:40.740 [bus notice] ...<0c0000001e000000000006031e18031e1e0c180c1800061e001e0000180006...
2021-12-28 12:17:41.594 [bus notice] ...<0c0000001e000000180c000006001e0018030000001e001e00001800060c00...
2021-12-28 12:17:42.537 [bus notice] ...<00001e000000060c001e1e060018000006001e00001800180c06000000001e...

I can’t figure out what wrong, I don’t see any identification from the TEM. 

Any help will be greatly appreciated.

Thanks 

nicolas

john30, 2022-01-02:
does the device really use an eBUS interface? or is this something else? The logged traffic does not look like eBUS protocol at all

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30