john30/ebusd #492: Unstable connection

moskovskiy82, 2021-12-30:
I have switched ebus adapter 3 from wemos to USB. Now i'm experienincg strange issues. It connects sometimes and sometimes doesn't. 
And looking at the raw data - seems quite weird. Doesn't look like a valid signal. Checked the cables already. Reset the boiler (Protherm) but still the same

ebusd --lograwdata --configpath=http://ebusd.eu/config/ --scanconfig=full -f -d /dev/ttyUSB0
2021-12-30 09:26:42.988 [main notice] ebusd 21.3.v21.3 started with full scan on device /dev/ttyUSB0
2021-12-30 09:26:43.462 [bus notice] bus started with own address 31/36
2021-12-30 09:26:43.517 [bus notice] signal acquired
2021-12-30 09:26:44.865 [bus notice] <fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...
2021-12-30 09:26:46.214 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...
2021-12-30 09:26:47.885 [bus notice] ...<fefefefefefefffc7ffffefefefefefefefefefefefefefefefefefefefefe...
2021-12-30 09:26:49.227 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...
2021-12-30 09:26:50.575 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...
2021-12-30 09:26:51.924 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...
2021-12-30 09:26:53.272 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...
2021-12-30 09:26:53.463 [main notice] starting initial full scan
2021-12-30 09:26:54.625 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...
2021-12-30 09:26:55.973 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...
2021-12-30 09:26:57.319 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...
2021-12-30 09:26:58.666 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...
2021-12-30 09:27:00.017 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...
2021-12-30 09:27:01.362 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...
2021-12-30 09:27:02.715 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...
2021-12-30 09:27:04.061 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...

and then after 4 minutes
2021-12-30 09:31:21.997 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...
2021-12-30 09:31:24.036 [bus error] signal lost

john30, 2021-12-31:
this looks a lot like you have another process using the same ttyUSB0 device. candidates for such processes are fhem and other stuff. you can check with lsof if another process is using the device

moskovskiy82, 2022-01-01:
> this looks a lot like you have another process using the same ttyUSB0 device. candidates for such processes are fhem and other stuff. you can check with lsof if another process is using the device

It's a fresh raspberry install. Nothing except ebusd is installed and nothing is quering ttyUSB0. Have checked that.

john30, 2022-01-02:
did you jumper the adapter to non-enhanced mode? If not, you need to use `-d enh:/dev/ttyUSB0` instead of `-d /dev/ttyUSB0`

john30, 2022-09-18:
closed due to inactivity

closed 2022-09-18 by john30