john30/ebusd #493: RPI 1 B : New install

kimi1983, 2021-12-30:
Hello,

i have tried several times to install the latest version of ebusd 21.3 on my rpi b.
I am using an old buster distribution since otherwise i am getting errors on compiling ttyebus

When i try to start the ebusd service i get the following 👍 
ebusd.service : Failed with result 'signal'

Can you please help me. I did the installation already several times using my old documentation i did in the days from version 3.1

best regards,
Kim

john30, 2022-01-26:
have a look at the log file in /var/log/ebusd.log as well as the system log. start ebusd in foreground using the appropriate commandline options

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30