john30/ebusd #496: Auto polling with r1; combined with scanconfig

rmeekers, 2022-01-04:
I'm using ebusd as an Add-on in Home Assistant. It's configured with scanconfig and checkconfig. The CSV files are pulled automatically from https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.1.x.

So far so good.
But since I'm using MQTT climate integration in Home Assistant, I would need to change the r; in the config files to r1; so it will be read out automatically and pushed on MQTT.

Is there any way to do that without pulling the config files locally (which doesn't work in the Home Assistant Addon setup)?

netweaver1970, 2022-01-04:
You can share your HA ebusd relevant setup? 

Did you make your own repo to add it in HA?

I'm very interested, I want to kill one Raspberry Pi if possible...

rmeekers, 2022-01-04:
> You can share your HA ebusd relevant setup?

I'm using this addon: https://github.com/LukasGrebe/ha-addons

doubleO8, 2022-01-05:
Hi, as stated in https://github.com/john30/ebusd/wiki/3.3.-MQTT-client ...
> In addition to that, appending a question mark and a poll priority digit allows adjusting the poll priority for read messages. For example, using mosquitto_pub -m '?3' -t 'ebusd/uih/YieldSum/get' would set the uih/YieldSum message to be polled with priority 3 and is equivalent to issuing ebusctl -c uih -p 3 YieldSum.

... it is possible to enable a customised polling behaviour for the key/value pairs you are interested in (this might also be an approach for satisfying #494).

According to my experience so far a restart of the ebusd service will lose these custom polling settings thus I use a cron job for triggering ebusd to read key/value pairs in addition to this approach.

rmeekers, 2022-01-05:
> In addition to that, appending a question mark and a poll priority digit allows adjusting the poll priority for read messages. For example, using mosquitto_pub -m '?3' -t 'ebusd/uih/YieldSum/get' [...]

As I understand that part, the extra parameter does not trigger automatic polling, but simply defines the polling priority. And thus not what I'm looking for. Or do I misunderstand this?

doubleO8, 2022-01-05:
If I interpret https://github.com/john30/ebusd/wiki/4.1.-Message-definition correctly ...

> When appending a digit 1-9 to the normal "read" type, the master-slave read message will be polled in regular intervals. The appended digit is the polling priority, i.e. messages with priority 1 are polled in each poll cycle, priority 2 messages only in every second poll cycle and so on.

... this should be equivalent. Anyway you can try to pub/sub as described above and see if the values will be updated in Home Assistant as expected.

rmeekers, 2022-01-06:
this does not the behaviour I'm looking for. Because it means that I have to trigger the polling from within Home Assistant with an automation. I really need the polling to be initiated from within ebusd.

john30, 2022-01-09:
you don't have to trigger the polling. the poll prio is exactly for doing that. the only thing you need to do is to tell ebusd which messages it is supposed to poll. you can do that with one of these methods:
* use your own csvs
* update the prio using the tcp port or ebusctl after an ebusd restart
* update the prio using mqtt after ebusd reports [scan complete](https://github.com/john30/ebusd/wiki/3.3.-MQTT-client#introduction)

rmeekers, 2022-01-09:
@john30 thanks! I've updated the MQTT page with this info to make it more clear.

closed 2022-01-09 by rmeekers

john30, 2022-01-26:
which page is that?

rmeekers, 2022-01-26:
> which page is that?

https://github.com/john30/ebusd/wiki/3.3.-MQTT-client/_compare/63d1e8c2c1e2e5457bf6e5a38cd738aec1e4b00f...5e8ac6a6700c81b2c8b0ead979a656c6dea02cb4