john30/ebusd #497: Issues after clean install

rhellenbrand, 2022-01-05:
Noticed after a clean install (buster) from mine raspberry pi3 Ebusd (21.3) didn’t work.

I couldn’t find the ebusd.log file to see wat’s wrong and did add a ebusd.log file with read and write permissions for all users.
Ebusd still didn’t work and could see in the log file that ebusd.pid file was missing.

After adding this file with R/W permissions and “0” in it ebusd did work like intended.
But if I now stop ebusd the ebusd.pid file is deleted and I can’t start ebusd anymore (have to add a new ebusd.pid file).

Is this a known issue or do I have to change the permissions of the folder to fix it?

Second issue is the poll of the solar yieldthisyear.
Mine solar system (VPM 20/2 S) does responds with /00 on it, and this creates only unwanted traffic on the bus.
How can I disable this poll?

rhellenbrand, 2022-01-06:
Found where the pid file is created.

/src/ebusd/main.cpp

659 // create pid file and try to lock it
660  pidFile = fopen(opt.pidFile, "w+");
661
662   umask(S_IWGRP | S_IRWXO);  // set permissions of newly created files to 750

Normal “Users” don’t have read and write permission!!!
Is locking of the R/W permissions for “user” needed?
John, can you change this?

rhellenbrand, 2022-01-08:
Poll issue is solved.

Ebusd has found following devices:
06.pms.csv
08.hmu.csv
0a.pmw.hwc.csv
15.700.csv
26.vr_71.csv
52.vr_70.csv
ec.sol.sc.csv

In ec.sol.sc.csv the file yield3f40.inc is included.

There is the YieldThisYear poll defined.
r9,,YieldThisYear,Ertrag dieses Jahr,,,,"3f00:2;3f08:2;3f10:2;3f18:2;3f20:2;3f28:2;3f30:2;3f38:2;3f40:2;3f48:2;3f50:2;3f58:2",,,energy;energy;energy;energy;energy;energy;energy;energy;energy;energy;energy;energy;,,,

Mine system solar system isn’t controlled by a VR 68, the VPM 20 S is controlled by the VR700.
This is why there isn’t a YieldThisYear response.

VPM20 is only transmitting the total yield in regular base to the VR700.

I only still have the permission problem of the ebusd.pid file!

john30, 2022-01-09:
then this can be closed I guess

closed 2022-01-09 by john30

rhellenbrand, 2022-01-09:
Hi John,

Not completely.
After clean install there in none ebusd.log file (installed by running the latest deb file)
Had to manual create the ebus.log file to get ebusd running.

rhellenbrand, 2022-01-09:
Hi John,

You did close the clean install issue, but do have still an issue.

After clean install there in none ebusd.log file (installed by running the latest deb file)
Had to manual create the ebus.log file to get ebusd running.

Maybe you can add creating the ebusd.log file if it not exist in the deb file.

Thanks,

Ron

Closed #497<https://github.com/john30/ebusd/issues/497>.

-

john30, 2022-01-09:
do you start is as root? or what do you do exactly?

rhellenbrand, 2022-01-09:
See if I logoff it automatic starts as PI

john30, 2022-01-09:
you need to start is as root

rhellenbrand, 2022-01-09:
Have to figure out how to root boot the system, have only PI option in this (default raspberry PI) install.