john30/ebusd #499: PID file location

jordipalet, 2022-01-08:
I've realized that when I reboot my system, I got several errors like:

systemd[1]: /lib/systemd/system/ebusd.service:9: PIDFile= references path below legacy directory /var/run/, updating /var/run/ebusd.pid → /run/ebusd.pid; please update the unit file accordingly.

before ebusd starts.

If I restart the system manually, it doesn't happen.

I've tried adding --pidfile=/run/ebusd.pid into the ebus.conf file, but the problem is the same, so it looks like the raspbian is starting the service "initially" ignoring the config file?

rhellenbrand, 2022-01-08:
Think I have the same problem.

Pid file is created and deleted in ebusd main see:
/src/ebusd/main.cpp
659 // create pid file and try to lock it
660 pidFile = fopen(opt.pidFile, "w+");
661
662 umask(S_IWGRP | S_IRWXO); // set permissions of newly created files to 750

If you are logged in as "user" you don’t have permissions to read and write the pid file and this creates the issue.
The permission (in line 662) 750 has to be changed to 755 or 777.

jordipalet, 2022-01-08:
This is happening even if I'm not "logged-in", I mean if I do a "sudo reboot" and not log in immediately. So I'm not sure it is related to the user permissions.

rhellenbrand, 2022-01-08:
To test if it is a permission issue you could try following.

Stop the ebus service
By closing the service the PID file is deleted.

Manual create your /run/ebusd.pid file with 777 permission.
Think it must contain a string like 0\0
Than reboot.

If the file exist ebusd will write the PID number in it.
If you don’t get the error during start-up and the service starts, it is a permission issue.

john30, 2022-01-09:
I can't reproduce this on a RPi buster with 21.3. maybe this is related to the umask? you might want to check [this](https://wiki.debian.org/Debate/umask).

jordipalet, 2022-01-09:
> To test if it is a permission issue you could try following.
>
> Stop the ebus service By closing the service the PID file is deleted.
>
> Manual create your /run/ebusd.pid file with 777 permission. Think [...]

Same problem even if I've created the pid file in both directories (/run and /var/run). Only because I've monit service to restart the services it comes up after a few seconds. The interesting thing is that the PID in both dirs has the same value. I'm not sure to understand why, if I'm using --pidfile=/run/ebusd.pid

john30, 2022-01-09:
/run and /var/run are usually symlinked to each other.
do you start ebusd as user root?

jordipalet, 2022-01-09:
Yes, I was thinking they are symlinked, but I was surprised by the error message "references path below legacy directory /var/run/, updating /var/run/ebusd.pid → /run/ebusd.pid", may be the error message is outdated.

When the rpi is rebooted, it starts with the service, I believe that means user root ... when it is restarted by monit, it start as root.

# eBus
  check process eBus matching "ebusd"
        start program = "/bin/systemctl start ebusd.service"
        stop program = "/bin/systemctl stop ebusd.service"
        if cpu > 1% for 5 cycles then restart
        if totalmem > 5 MB for 5 cycles then restart

ps -ef | grep ebusd
root      2010     1  0 11:26 ?        00:00:01 /usr/bin/ebusd --scanconfig --httpport=8889 --log=all error -d enh:10.10.10.23:9999

rhellenbrand, 2022-01-09:
Would it not be easier to change the permissions in main.cpp to avoid issues with clean PI installs?

660 pidFile = fopen(opt.pidFile, "w+");
661
662 umask(S_IWGRP | S_IRWXO); // set permissions of newly created files to 750
Could the last line not be changed in 777?

Just created the ebusd.pid file in the /var/run/ folder with 
sudo nano ebusd.pid

Changed permissions like in main.ccp
sudo chmod 750 ebusd.pid
Ebusd doesn’t start

sudo chmod 755 ebusd.pid
Ebusd doesn’t start

sudo chmod 777 ebusd.pid
Ebusd does start!

So PI like I’m logged in now has only “other” permissions.

rhellenbrand, 2022-01-09:
Ok 

Managed to login as root, but under root the desktop taskbar is missing!!! 
With ctrl+alt+T i could open the terminal and ebusd runs fine.

Think I have to build ebusd myself and change permission in line 662 to 777 so I can run ebusd as “PI” user.
To have none taskbar on the desktop is not real practical!

The missing taskbar under root seems to be intended and not easy to fix.
Or where can I download an image of the OS without this feature?

Ron

john30, 2022-01-09:
guys, I really don't know what you're doing here.
you need to make sure that the system starts ebusd as user root or at least with a user having enough privileges. there is no need to have a root UI or anything like that.
ebusd is a daemon and as such it is not supposed to be started in daemon mode as a non-privileged user. maybe you didn't do the installation correctly. check the output of `ls -la /usr/bin/ebusd` and post the content of the start script.

rhellenbrand, 2022-01-09:
Hi John,

Problem is that default Pi OS if you are logged in as “root” doesn’t have a taskbar on the desktop.
And this is not real practical, you must do everything from the terminal.
if the permissions of the ebusd.pid file was 777 instead of 750 you could login as PI user with working desktop and ebusd would work.

For a final standalone application ebusd works fine, but if you have to do a lot of debugging on ebus messages it is not real practical.

Other thing I noticed is if you install by deb file the file ebusd.log in folder /var/log/ is not created, had to manual create this file to get ebusd working.

Ron

jordipalet, 2022-01-09:
I always ssh as user "pi", not root, but ebusd is installed by default, so I guess it runs as root?

pi@iot:~ $ ls -la /usr/bin/ebusd
-rwxr-xr-x 1 root root 339056 Nov 10 07:05 /usr/bin/ebusd

default start script:

pi@iot:~ $ cat /etc/init.d/ebusd 
#!/bin/bash
#
### BEGIN INIT INFO
# Provides:        ebusd
# Required-Start:  $network $remote_fs $syslog
# Required-Stop:   $network $remote_fs $syslog
# Default-Start:   2 3 4 5
# Default-Stop:    0 1 6
# Short-Description: controls ebusd, the daemon for communication with eBUS heating systems.
### END INIT INFO

DAEMON=/usr/bin/ebusd
PIDFILE_PREFIX=/var/run/ebusd
PIDFILE_SUFFIX=.pid

if test -f /lib/lsb/init-functions; then
	. /lib/lsb/init-functions
else
	. /etc/init.d/functions

	log_daemon_msg()  { logger "$@"; }
	log_end_msg()     { [ $1 -eq 0 ] && RES=OK; logger ${RES:=FAIL}; }
	log_failure_msg() { logger "FAIL $@"; }
	log_warning_msg() { logger "WARN $@"; }
	status_of_proc () { status $3; }

fi

[ -r /etc/default/ebusd ] && . /etc/default/ebusd

if [ ! -x $DAEMON ]; then
	log_failure_msg "$DAEMON is not available or not executable."
	exit 5
fi

ALL_OPTS=${!EBUSD_OPTS*}
if [ ${#ALL_OPTS[@]} -lt 1 ]; then
  ALL_OPTS[0]=EBUSD_OPTS
fi
instance="EBUSD_OPTS$2"
instance=${!instance}

start_instance () {
	local opts suffix
	opts=$1
	shift
	suffix="${opts#EBUSD_OPTS}"
	log_daemon_msg "Starting ebusd${suffix}" ebusd
	pidfile=$PIDFILE_PREFIX${suffix}$PIDFILE_SUFFIX
	start-stop-daemon --start --quiet --oknodo --pidfile $pidfile --exec $DAEMON -- --pidfile $pidfile ${!opts}
	log_end_msg $?
}

stop_instance () {
	local opts suffix
	opts=$1
	shift
	suffix="${opts#EBUSD_OPTS}"
	log_daemon_msg "Stopping ebusd${suffix}" ebusd
	pidfile=$PIDFILE_PREFIX${suffix}$PIDFILE_SUFFIX
	start-stop-daemon --stop --quiet --oknodo --pidfile $pidfile
	log_end_msg $?
	rm -f $PIDFILE_PREFIX${suffix}$PIDFILE_SUFFIX
}

case $1 in
	start)
		if [ -z "$2" ]; then
			for opts in $ALL_OPTS; do
				start_instance $opts
			done
		elif [ -z "$instance" ]; then
			log_failure_msg "ebusd$2 is not configured"
		else
			start_instance EBUSD_OPTS$2
		fi
  		;;
	stop)
		if [ -z "$2" ]; then
			for opts in $ALL_OPTS; do
				stop_instance $opts
			done
		else
			if [ -z "$instance" ]; then
				log_warning_msg "ebusd$2 is not configured"
			fi
			stop_instance EBUSD_OPTS$2
		fi
  		;;
	restart|force-reload)
		ARGS=($@)
		$0 stop ${ARGS[@]:1} && sleep 2 && $0 start ${ARGS[@]:1}
  		;;
	status)
		if [ -z "$2" ]; then
			for opts in $ALL_OPTS; do
				suffix=${opts#EBUSD_OPTS}
				pidfile=$PIDFILE_PREFIX$suffix$PIDFILE_SUFFIX
				status_of_proc -p $pidfile $DAEMON ebusd$suffix
			done
		else
			if [ -z "$instance" ]; then
				log_warning_msg "ebusd$2 is not configured"
			fi
			pidfile=$PIDFILE_PREFIX$2$PIDFILE_SUFFIX
			status_of_proc -p $pidfile $DAEMON ebusd$2
		fi
		;;
	*)
		echo "Usage: $0 {start|stop|restart|force-reload|status} [instance]"
		exit 2
		;;
esac

eh-gh, 2022-01-11:
hi changing  the file /etc/systemd/system/ebusd.service manually

from 
PIDFile=/var/run/ebusd.pid

to
PIDFile=/run/ebusd.pid

eliminates the warning in messages 

https://github.com/john30/ebusd/pull/502 created

rhellenbrand, 2022-01-11:
I’m totally unknown with Linux but you already noticed that.

Do change the permissions in the terminal with "sudo su"
Then I can start ebusd manual, and it works fine.

In the past I stopped ebusd with “service ebusd stop” and manual started it again to reload the csv files after modification.
This seems not to work anymore, have to reboot to load the new csv files.

Looked in “/etc/systemd/system/” but I don’t see the file "ebisd.service" does this causes this behaviour?

Where can I find the ebusd.service file?

Did installed ebusd by deb file!

eh-gh, 2022-01-12:
@rhellenbrand to be honest i do not see any connection between your problem and the original one.

can you please open a new ticket and try to summarize, what is not working for you? Have you tried to reinstall the ebusd package? Can you include the output into the new ticket?
tnx