john30/ebusd #500: Software to edit CSV files?

moskovskiy82, 2022-01-09:
Which software can be used to edit CSV files? Sounded simple at first but excel, google docs and others i've tried remove quotation marks.
For example bai.308523.csv
*ws,#service,,,,,"B509","0E",,,,,,

After importing and exporting becomes
*ws,#service,,,,,B509,0E,,,,,,

eh-gh, 2022-01-11:
Can you try excel text import with the following config:

[image]

bdariusb, 2022-01-17:
I use [this](https://www.editcsvonline.com/). Just need a browser

john30, 2022-01-26:
I always use LIbreoffice, works fine

djwmarcx, 2023-01-27:
I personally use this plugin for Visual Studio Code:

https://marketplace.visualstudio.com/items?itemName=janisdd.vscode-edit-csv

When opening the files, you'll need to setup some things in order to read/write the file properly.

1. Open the csv as usual in vscode
2. At the top-right corner, click on "Edit CSV"
3. File will be readed incorrectly. At the top of the visual editor, click on "Read Options" to show the options panel
5. Configure the editor with the following settings

[image]

6. Then click on "Reset data and apply read options".

Once the options are configured, you can close the panel

The only downside is that you need to change the read options every time you open a new file.