john30/ebusd #501: Logrotaterrer since V21.3

Nico184, 2022-01-09:
Since I updated to version 21.3 an ubuntu 18.04.6 LTS logrotate doesn't work. The logfile /var/log/ebusd.log is present and grows bigger.
If I start logrotate -d /var/log/ebusd.log manually I get an error massage: 
error: /var/log/ebusd.log:1 lines must begin with a keyword or a filename (possibly in double quotes)
It seems that the ebusd.log is in wrong file format? All other logfiles are treated correctly.
Till version 21.1 it works without problems.
Maybe anyone can give me a hind?
Thanks a lot

eh-gh, 2022-01-09:
Hi
I think you should give the config file as parameter:
logrotate -d /etc/logrotate.d/ebusd

Nico184, 2022-01-09:
HIi,
thanks.. my fault - I startes manually with wrong file but if I do it like you said I get this massage:
error: Ignoring /etc/logrotate.d/ebusd because of bad file mode - must be 0644 or 0444.
Any hind?

eh-gh, 2022-01-09:
please check 
ls -l /etc/logrotate.d/ebusd , if the ownership is not root/root change with 

sudo chown root.root /etc/logrotate.d/ebusd
and try again

@john30, i think if the pkg is installed with sudo dpkg -i ... (not as root), the file /etc/logrotate.d/ebusd has no root.root ownership but the user, who runs the installation , which would be required for logrotate

Nico184, 2022-01-09:
did so,
ownership was `-rwxr-xr-x 1 root root 150 Nov  9 22:52 /etc/logrotate.d/ebusd
`
same behaviour...

eh-gh, 2022-01-09:
somehow you've got different permissions as mine:

please run additionally:
sudo chmod 0644 /etc/logrotate.d/ebusd

Nico184, 2022-01-23:
Thanks for help, I changed permissions and now it works since one week.
And many thanks for providing and supporting ebusd to john and the other guys of this community

closed 2022-01-23 by Nico184