john30/ebusd #504: Colormatic 370 and VR81 dump required for emulation

k41c, 2022-01-16:
Hi John30,

great project! On my setup my boiler is at the other side of our house. So there is no posibility to route cables to the loving room. My plan is to put a tempsensor in the livingroom and emulate the VR81 at the boiler, to get the feedback temp and optimal control the system and switch in the evening to the sleeping room as reference. 

So my question: 
Can anyone supply my some dumpfiles with a colomatic 370 and an VR81?
1. At start up.
2. during day 
3. during night
So I would be able to reverse eng. And emulate the VR81. Self taking supplying back my results ;)

Many thanks in advaned!

Mitti10, 2022-02-22:
Hi k41c
here is 'grab result' made in the evening of winter day for my setup that includes Ecotec, VR430, V61, V81.
[ebusd-grabresult.txt](https://github.com/john30/ebusd/files/8117047/ebusd-grabresult.txt).

PureDJ, 2022-02-27:
I have made a thermostat emulation with success, I replied to topic #481 how to do this.
I have another setup, but because you can use the ColorMatic I have on other boilers, I think the way it works is the same.

k41c, 2022-03-04:
Hi Mitti10,

many thanks for the Log! It's strage that there is aother unknown slave 26 wich addresed with VR81 data. It seems there is some watch dog or simular. Are you may able to log some more data. May be if possible while start up and leave running for 10min. or so? Thnaks in advanced. Regards

k41c, 2022-03-04:
Hi PureDJ,

thanks for your input. Purely there is no simular command for the Colormatic 430. I found there is raw room temp. But its marked as read only. Anyhow I will try to override and see whats happen. Hopefully get some time this weekend or next.... Regrads

PureDJ, 2022-03-04:
Well, what I did is I made actually 2 systems I can switch between.
One is a complete thermostat in Node-red, with PID etc. and one as it was before.
The Colormatic is sending SetMode to the boiler every second, but I can command the Colormatic to stop doing that and my own PID controller can take over, but I made beside that also that HA and Node-Red can send the setpoint to the Colormatic so when it's not in shutdown it can control the boiler.

To let the colormatic stop sending SetMode I send the command OperatingMode 0 to the the colormatic (OperatingMode set to 1 let's the colormatic take control again)

With ComfTempDesired I can set the desired setpoint, but you can also use it to read the setpoint set on the Colormatic

I hope that will help a bit.

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30