john30/ebusd #506: EBUS Adapter 3

MichaelErmer, 2022-01-20:
After some initial issues, until I figured out that I have to use `enh:` for the device address, my adapter has been working very well.

However I still have a lot of timeouts, I think that might be related to an issue I had in the beginning and is still in my setup, the adapter only seems to work if I set `--latency=1`, it does not respond with any other or if I remove the argument.

Is there a known issue or is there anything I can do to improve the performance?

I'm using the "official" V3 USB Adapter.

john30, 2022-01-26:
please provide ebusd logs with lograwdata=bytes for both cases including all the start parameters used

john30, 2022-09-17:
closed due to inactivity

closed 2022-09-17 by john30