john30/ebusd #508: How to initialize polling of several messages in Docker Compose

freerkvenhuizen, 2022-01-25:
Hi,

I am working on setting up the following chain: ebusd -> mosquitto -> telegraf -> influxdb using docker-compose.
Currently the entire chain seems to be working as expected, but there is 1 thing I still have to do manually which I would like to automate somehow.

After initializing ebusd I have to manually use Mosquitto to enable polling of the parameters I'm interested in using (for example) the following command: _docker container exec mosquitto mosquitto_pub -t 'ebusd/700/z1RoomTemp/get'  -m '?3'_
As far as I understand, this command tells ebusd to enable polling for this message, which indeed works; I nicely see the data flowing into the database at a regular interval. 

I would however like to add this initialization somewhere in my code such that ebusd automatically knows which messages to poll. How to do this?

freerkvenhuizen, 2022-01-25:
Content of the docker-compose yaml file:

version: "3.5"

services:

  mosquitto:
    container_name: mosquitto
    image: eclipse-mosquitto:latest
    user: "0"
    ports:
      - "1883:1883"
      - "9001:9001"
    volumes:
      - ./mqtt/data:/mosquitto/data
      - ./mqtt/log:/mosquitto/log
      - ./mqtt/config:/mosquitto/config

    restart: always

  influxdb:
    container_name: influxdb
    image: influxdb:latest
    ports:
      - "8086:8086"
    volumes:
      - ./influxdb:/var/lib/influxdb2
      - ./influxdb:/etc/influxdb2
    environment:
      - DOCKER_INFLUXDB_INIT_MODE=setup
      - DOCKER_INFLUXDB_INIT_USERNAME=telegraf
      - DOCKER_INFLUXDB_INIT_PASSWORD=P@ssw0rd
      - DOCKER_INFLUXDB_INIT_ORG=telegraf
      - DOCKER_INFLUXDB_INIT_BUCKET=telegraf
      - DOCKER_INFLUXDB_INIT_RETENTION=1w
      - DOCKER_INFLUXDB_INIT_ADMIN_TOKEN=9859DAA6-3B3F-48FE-A981-AE9D31FBB334
    restart: always

  telegraf:
    container_name: telegraf
    image: telegraf:latest
    ports:
      - "8092:8092"
      - "8094:8094"
      - "8125:8125"
    volumes:
      - ./telegraf/telegraf.conf:/etc/telegraf/telegraf.conf:ro
    restart: always

  ebusd:
      container_name: ebusd
      image: john30/ebusd
      restart: unless-stopped
      tty: true
      devices:
        #- '/dev/serial/by-id/usb-Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller_62928ffa24deea119e8ccf149a583cc7-if00-port0:/dev/ttyUSB0'
        - '/dev/ttyUSB0:/dev/ttyUSB0'
      ports:
        - 8888:8888
      #configuration volume if you need to customize it (also add --configpath=/config to command param)
      volumes:
        - ./ebusd/log:/var/log
      depends_on:
        - mosquitto
      command: -f --scanconfig --mqtthost=mosquitto --mqttport=1883 --mqttjson --accesslevel="*"
      privileged: true

mwildbolz, 2022-01-26:
I would suggest that you clone the ebusd-configuration repository (https://github.com/john30/ebusd-configuration) to a local directory and change the message type from "r" to "r1-r9" (according to the poll priority needed) for all values you want to poll regularly (refer to [message definition](https://github.com/john30/ebusd/wiki/4.1.-Message-definition)). The parameter `--pollinterval=SEC` should also be set!

So you don't need to do that manually via MQTT.

freerkvenhuizen, 2022-01-26:
Awesome, that sounds exactly what I'm looking for! Thanks!

john30, 2022-02-20:
the MQTT integration settings are capable of adjusting poll priority of messages matching filter criteria, see [here](https://github.com/john30/ebusd/blob/master/contrib/etc/ebusd/mqtt-integration.cfg) and [here](https://github.com/john30/ebusd/wiki/MQTT-integration)

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30