john30/ebusd #510: problem to connect two boilers

Kentrp, 2022-01-27:
Hi I don't known how to connect two boilers (gas Protherm Lynx and electric Protherm Ray) without eBus controller.
I can connect them separately and all work ok. Can your help me, thanks

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
> ebusd --foreground --httpport=8889 --mqtthost=192.168.1.2 --mqttport=1883 --mqttuser=//// --mqttpass= /// --scanconfig --mqttlog --configpath=/config/ebusd-configuration/ebusd-2.1.x/en --latency=60 --accesslevel=* --device=/dev/ttyUSB0 --log=all:debug
2022-01-27 14:32:10.891 [main notice] ebusd 21.3.v3.15.0-4372-gd2c04b9343 started with auto scan on device /dev/ttyUSB0
2022-01-27 14:32:10.891 [main info] loading configuration files from /config/ebusd-configuration/ebusd-2.1.x/en
2022-01-27 14:32:10.892 [main info] reading templates /
2022-01-27 14:32:10.893 [main info] read templates in /
2022-01-27 14:32:10.893 [main info] reading file .__templates.csv
2022-01-27 14:32:10.894 [main error] error reading config files from /config/ebusd-configuration/ebusd-2.1.x/en: ERR: end of input reached, last error: .__templates.csv:1: ERR: end of input reached, missing message type/name/pbsb
2022-01-27 14:32:10.900 [bus notice] bus started with own address 31/36
2022-01-27 14:32:10.900 [main info] registering data handlers
2022-01-27 14:32:10.900 [mqtt info] mosquitto version 2.0.14 (compiled with 2.0.14)
2022-01-27 14:32:10.901 [mqtt debug] log Client ebusd_21.3_374 sending CONNECT
2022-01-27 14:32:10.901 [mqtt debug] connection requested
2022-01-27 14:32:10.901 [main info] registered data handlers
2022-01-27 14:32:10.905 [mqtt debug] log Client ebusd_21.3_374 received CONNACK (0)
2022-01-27 14:32:10.905 [mqtt notice] connection established
2022-01-27 14:32:10.905 [mqtt debug] publish ebusd/global/version ebusd 21.3.v3.15.0-4372-gd2c04b9343
2022-01-27 14:32:10.905 [mqtt debug] log Client ebusd_21.3_374 sending PUBLISH (d0, q0, r1, m1, 'ebusd/global/version', ... (35 bytes))
2022-01-27 14:32:10.905 [mqtt debug] publish ebusd/global/running true
2022-01-27 14:32:10.905 [mqtt debug] log Client ebusd_21.3_374 sending PUBLISH (d0, q0, r1, m2, 'ebusd/global/running', ... (4 bytes))
2022-01-27 14:32:10.906 [mqtt debug] log Client ebusd_21.3_374 sending SUBSCRIBE (Mid: 3, Topic: ebusd/#, QoS: 0, Options: 0x00)
2022-01-27 14:32:10.919 [mqtt debug] log Client ebusd_21.3_374 received SUBACK
2022-01-27 14:32:10.920 [mqtt debug] log Client ebusd_21.3_374 received PUBLISH (d0, q0, r1, m0, 'ebusd/global/version', ... (35 bytes))
2022-01-27 14:32:10.921 [mqtt debug] log Client ebusd_21.3_374 received PUBLISH (d0, q0, r1, m0, 'ebusd/global/running', ... (4 bytes))
2022-01-27 14:32:10.921 [mqtt debug] log Client ebusd_21.3_374 received PUBLISH (d0, q0, r1, m0, 'ebusd/global/scan', ... (8 bytes))
2022-01-27 14:32:10.922 [mqtt debug] log Client ebusd_21.3_374 received PUBLISH (d0, q0, r1, m0, 'ebusd/global/updatecheck', ... (328 bytes))
2022-01-27 14:32:10.922 [mqtt debug] log Client ebusd_21.3_374 received PUBLISH (d0, q0, r1, m0, 'ebusd/global/signal', ... (4 bytes))
2022-01-27 14:32:10.922 [mqtt debug] log Client ebusd_21.3_374 received PUBLISH (d0, q0, r1, m0, 'ebusd/global/uptime', ... (2 bytes))
2022-01-27 14:32:10.923 [mqtt debug] log Client ebusd_21.3_374 received PUBLISH (d0, q0, r1, m0, 'ebusd/scan.08/id', ... (34 bytes))
2022-01-27 14:32:10.923 [mqtt debug] log Client ebusd_21.3_374 received PUBLISH (d0, q0, r1, m0, 'ebusd/scan.08/', ... (24 bytes))
2022-01-27 14:32:11.040 [bus debug] ERR: SYN received during no signal, switching to ready
2022-01-27 14:32:11.040 [bus notice] signal acquired
2022-01-27 14:32:15.938 [bus notice] new master 03, master count 2
2022-01-27 14:32:16.025 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2022-01-27 14:32:20.902 [main debug] performing regular tasks
2022-01-27 14:32:20.903 [bus info] scan 08 cmd: 3108070400
2022-01-27 14:32:20.912 [bus debug] start request 31
2022-01-27 14:32:20.912 [bus debug] arbitration start with 31
2022-01-27 14:32:26.940 [mqtt debug] publish ebusd/global/uptime 16
2022-01-27 14:32:26.940 [mqtt debug] log Client ebusd_21.3_374 sending PUBLISH (d0, q0, r0, m4, 'ebusd/global/uptime', ... (2 bytes))
2022-01-27 14:32:26.941 [mqtt debug] publish ebusd/global/signal true
2022-01-27 14:32:26.941 [mqtt debug] log Client ebusd_21.3_374 sending PUBLISH (d0, q0, r1, m5, 'ebusd/global/signal', ... (4 bytes))
2022-01-27 14:32:26.942 [mqtt debug] log Client ebusd_21.3_374 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (2 bytes))
2022-01-27 14:32:26.948 [mqtt debug] log Client ebusd_21.3_374 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/signal', ... (4 bytes))
2022-01-27 14:32:42.965 [mqtt debug] publish ebusd/global/uptime 32
2022-01-27 14:32:42.965 [mqtt debug] log Client ebusd_21.3_374 sending PUBLISH (d0, q0, r0, m6, 'ebusd/global/uptime', ... (2 bytes))
2022-01-27 14:32:42.966 [mqtt debug] log Client ebusd_21.3_374 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (2 bytes))
2022-01-27 14:32:58.983 [mqtt debug] publish ebusd/global/uptime 48
2022-01-27 14:32:58.983 [mqtt debug] log Client ebusd_21.3_374 sending PUBLISH (d0, q0, r0, m7, 'ebusd/global/uptime', ... (2 bytes))
2022-01-27 14:32:58.984 [mqtt debug] log Client ebusd_21.3_374 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (2 bytes))
2022-01-27 14:33:00.714 [bus debug] arbitration lost
2022-01-27 14:33:00.714 [bus debug] ERR: arbitration lost during ready, retry
2022-01-27 14:33:00.746 [bus notice] new master 00, master count 3
2022-01-27 14:33:00.823 [update info] received MS cmd: 0008b509022802 / 0c020007043901004241493030
2022-01-27 14:33:00.823 [update notice] received unknown MS cmd: 0008b509022802 / 0c020007043901004241493030
2022-01-27 14:33:00.825 [bus debug] start request 31
2022-01-27 14:33:00.825 [bus debug] arbitration start with 31
2022-01-27 14:33:01.001 [bus debug] arbitration lost
2022-01-27 14:33:01.001 [bus debug] ERR: arbitration lost during ready, retry
2022-01-27 14:33:01.100 [update info] received MS cmd: 0008b509022802 / 0c020007043901004241493030
[... 281 lines cut ...]
2022-01-27 14:38:11.073 [bus debug] start request 31
2022-01-27 14:38:11.073 [bus debug] arbitration start with 31
2022-01-27 14:38:18.046 [mqtt debug] publish ebusd/global/uptime 368
2022-01-27 14:38:18.046 [mqtt debug] log Client ebusd_21.3_374 sending PUBLISH (d0, q0, r0, m34, 'ebusd/global/uptime', ... (3 bytes))
2022-01-27 14:38:18.049 [mqtt debug] log Client ebusd_21.3_374 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (3 bytes))

other one
bash-5.1# ebusctl i
version: ebusd 21.3.v3.15.0-4372-gd2c04b9343
device: /dev/ttyUSB0
access: *
signal: acquired
symbol rate: 68
max symbol rate: 68
reconnects: 0
masters: 3
messages: 2
conditional: 0
poll: 0
update: 0
address 00: master #1
address 03: master #11
address 08: slave #11
address 31: master #8, ebusd
address 36: slave #8, ebusd

john30, 2022-01-29:
no idea what you expect from me

john30, 2022-09-18:
closed due to inactivity

closed 2022-09-18 by john30