john30/ebusd #513: ESERA eBus ethernet potentiometer configuration

HSE83, 2022-02-05:
Hi,

I am currently trying to get the setup ebusd <-> ESERA eBus ethernet adapter <-> vaillant ecotec plus work and could not find any better place to ask (maybe you can also give me some hints where to better post such a question..). I have connected the ebus adapter using a CAT6 ethernet cable (2 twisted wires), the distance is around 2.5 meters. After some trying I managed to connect ebusd to the adapter via TCP and it is displaying the infamous 0xAA bytes which show it is detecting the bus synchronization characters.
It is not detecting any messages, though. This is part of the output:

`2022-02-04 23:32:58.363 [bus debug] ERR: SYN received during receive command, switching to ready
2022-02-04 23:32:58.475 [bus notice] <1076110115ff
2022-02-04 23:32:58.475 [bus debug] ERR: SYN received during receive command, switching to ready
2022-02-04 23:32:58.561 [bus notice] <bf
2022-02-04 23:32:58.561 [bus debug] ERR: SYN received during receive command, switching to ready
2022-02-04 23:32:58.674 [bus debug] ERR: CRC error during receive command CRC, switching to skip
2022-02-04 23:32:58.762 [bus notice] <10fe07040005ff
2022-02-04 23:32:58.922 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2022-02-04 23:32:58.963 [bus notice] <ff50070400ff
2022-02-04 23:32:58.964 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2022-02-04 23:32:59.205 [bus notice] <ea04754484ed
2022-02-04 23:32:59.205 [bus debug] ERR: SYN received during receive command, switching to ready
2022-02-04 23:32:59.448 [bus notice] <bf1050078454ff
2022-02-04 23:32:59.448 [bus debug] ERR: SYN received during receive command, switching to ready
2022-02-04 23:32:59.688 [bus notice] <fa834207c488ff
2022-02-04 23:32:59.688 [bus debug] ERR: SYN received during receive command, switching to ready
2022-02-04 23:32:59.930 [bus notice] <fe35227d9000d4
2022-02-04 23:32:59.930 [bus debug] ERR: SYN received during receive command, switching to ready
`

When I connect to the ebus adapter using a little test tool I wrote, I get roughly the same bytes as output from the adapter, so this might not be an ebusd issue alltogether. In the end the only tool I have to decode the ebus messages is ebusd, though. These are some byte sequences in between the 0xAA characters I have recorded:

`0xAA,0xAA,0xEA,0xFF,0xAA,0xAA
0xAA,0xAA,0xFA,0xFD,0xAA,0xAA
0xAA,0xAA,0xEA,0xFE,0x00,0x09,0x78,0x6A,0xF8,0xD0,0x77,0x01,0xFC,0x2F,0xFE,0xF5,0xAA,
0xAA,0xF8,0x01,0x01,0x9A,0xAA,0xAA,0xAA,0xFE,0xF5,0xAA,0xAA,0xAA,0xAA
`

Even with the background of roughly readig the ebus documents regarding message format etc I can not make sense of the received data. Maybe one of you recognizes the data format my vailant heater is outputting or can give any hints for me to solve the issue.

Best regards and thanks for your kind help,

Tobi

john30, 2022-02-13:
seems like the potentiometer is not adjusted well enough. some parts of the messages look good, but then crc errors and some weird bytes look bad

neutrino220, 2022-05-22:
I have the same issues but adjusting the poti brings no improvement

john30, 2022-05-29:
did you follow the poti adjustment advice in the wiki @HSE83 ?
it is really critical to have that adjusted as good as possible as any single bit being incorrect ruins the interpretation of course and will yield in invalid CRC messages as in your log extract.
the bridged clamps you mentioned are not connected to the ebus plug at all i hope?

HSE83, 2022-06-01:
Yes, I did the poti adjustment multiple times, trying settings at the lower and upper end of the scale and also something in right between. And no, the bridged clamps are not connected, thats just what I noticed on the heater PCB itself and was not sure it that is the right setting/interfering with anything else.
So yes, the issue is still open and I don't know how to resolve it.

john30, 2022-06-19:
usually the UDP mode is better due to less latency, maybe you want to try that

neutrino220, 2022-06-19:
> usually the UDP mode is better due to less latency, maybe you want to try that

UDP is not working! Absolutely no Data comes in and i use --initsend.

I've thought it was an issues from the Homeassistants eBusd add-on but i tried this also on my x86 Debian and there is completely the same behavior.

If i use lograwdata over TCP all Data comes in but if i switch the Esera to UDP mode and change the eBusd Parameter to udp: and add --initsend no Data comes in.

john30, 2022-06-19:
I know UDP is working better than TCP. do you have a special routing config, i.e. a router between two networks that does not forward UDP traffic?

neutrino220, 2022-06-19:
No, there is no Router between Esera and eBusd. Both are connected to the same switch

john30, 2022-06-19:
just tested it again and UDP works

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30