john30/ebusd #514: Number format of MQTT messages

mwildbolz, 2022-02-06:
#### Description
I recognized, that the number format of some fields sent out via MQTT differs from that when using `ebusctl read`.
Is there some sort of number formatting in place?

#### Actual behavior
For example the read command 
`ebusctl read -f -c 700 Hc1HeatCurve` leads to an output of: 0.15
If I look at the MQTT messages delivered (without `--mqttjson`) I receive: 0.2

If I set the value via MQTT set-command, i first get the right value (0.15) and 1 second later I get again 0.2

Some fileds deliver numbers with higher precisions (for example the outside temperature).

#### Expected behavior
The output of `ebusctl read` should be the same as sent out via MQTT

#### Environment
- OS: raspbian
- Architecture: armv71
- ebusd version: 21.3
- ebusd start arguments: 
`EBUSD_OPTS="--scanconfig --configpath=/etc/ebusd/ebusd-configuration/latest/de/ --pollinterval=5 --enablehex --httpport=8889 --mqttpass=xxxx --mqttport=1883 -d /dev/ttyUSB1`

- Integration:
  - [ x] TCP 
  - [ x] HTTP
  - [ x] MQTT
  - [ ] other

#### Logs
```txt
2022-02-06 11:12:44.894 [update notice] sent poll-read 700 Hc1HeatCurve QQ=31: 0.15
```

john30, 2022-02-12:
can you try with a build from latest source? It could be fixed with 4bda1a51404fc652cf3ad86c3a87ef8894dc554f already

mwildbolz, 2022-02-12:
Hi @john30!
Just tried the latest version 22.1 (which as I understood should have the mentioned fix included).
Same situation as described above!

mwildbolz, 2022-02-12:
I did the following:
1. I enabled raw logging
2. I issued `ebusctl read -f -c 700 Hc1HeatCurve`

What I got on commandline is: 
```
pi@raspberrypi2:~ $ ebusctl read -f -c 700 Hc1HeatCurve
0.15
```

In the logs I had
```
2022-02-12 20:05:07.097 [update notice] received write vr_71 Mc1FlowTempDesired QQ=10: on;25.0;on;15
2022-02-12 20:05:07.101 [bus notice] <1026b5230402000132210002010fb800
2022-02-12 20:05:07.342 [update notice] received write vr_71 Mc2FlowTempDesired QQ=10: off;0.0;on;-100
2022-02-12 20:05:07.346 [bus notice] <1026b52304020100009e0002019c2b00
2022-02-12 20:05:09.629 [update notice] sent poll-read 700 Hc1HeatCurve QQ=31: 0.15
2022-02-12 20:05:09.631 [bus notice] >3115b52406020002000f0064<000803020f009a99193e4d>00
2022-02-12 20:05:10.203 [update notice] sent poll-read 700 Hc1ActualFlowTempDesired QQ=31: 25.2685
2022-02-12 20:05:10.205 [bus notice] >3115b524060200020007007d<000801020700d125ca41b6>00
2022-02-12 20:05:14.973 [update notice] received read hmu Status01 QQ=10: 22.5;22.5;-3.062;-;36.5;off
2022-02-12 20:05:14.976 [bus notice] <1008b51101018900092d2df0fcff490000ff8500
2022-02-12 20:05:15.239 [update notice] received unknown MS cmd: 1076b5110101 / 09ff2df0fcff490000ff
2022-02-12 20:05:15.243 [bus notice] <1076b5110101160009ff2df0fcff490000ff9800
2022-02-12 20:05:15.506 [update notice] received unknown MS cmd: 1076b512030f0001 / 074c020000801402
2022-02-12 20:05:15.510 [bus notice] <1076b512030f0001fc00074c0200008014022500
```

On MQTT side I got on 20:05:11
```
{
     "0": {"name": "", "value": 0.2}}
```

mwildbolz, 2022-02-12:
And @john30, as additional information - maybe it is important:
It is not always, that there is an MQTT message with the right value (0.15) and shortly after with the wrong one (0.2).
Like in the example above, most of the time there is just one MQTT message sent out with the wrong value (0.2).

mwildbolz, 2022-02-12:
Another example (a little bit different behaviour with the new version):
```
pi@raspberrypi2:~ $ ebusctl read -f -c 700 Hc1ActualFlowTempDesired
25.7286
```

leads to MQTT message of:
```
{
     "tempv": {"value": 3e+01}}
```

closed 2022-02-13 by john30

mwildbolz, 2022-02-14:
> seems that the precision needs to be reset as well, can you check if it is solved with [f1cac3c](https://github.com/john30/ebusd/commit/f1cac3c48744931ff4ed1721f2917aacd4e5febf)

@john30 Thanks for your effort --> I can confirm, that the problem is solved!

commit 845bb1ab (2022-02-13): fix previous commit and fully reset stream before writing anything (should fix #514)