john30/ebusd #515: ebusd don't work with Vaillant auroMATIC 620

baggerfahrer441, 2022-02-07:
Hello, I need help for ebusd wirth Vaillant auroMATIC 620.
I get only the outsidetemp with ebusd + iobroker and can not find the failure.
Many houres I've read in WWW an here but no success.
I think Raspberry, eBus_Coupler and ebusd works - it a item with configuration.
At least I think ther is a problem with 
a) Timeout during scan
b) CSV-Files
I'm not a specialist with bus-technology, I only want to read some values from my heating.
Can sombody give me a little help?
Here some Info from my system:
root@raspberrypi:/home/pi# service ebusd status
● ebusd.service - ebusd, the daemon for communication with eBUS heating systems.
     Loaded: loaded (/lib/systemd/system/ebusd.service; disabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-02-07 19:09:31 CET; 6s ago
    Process: 8529 ExecStart=/usr/bin/ebusd $EBUSD_OPTS (code=exited, status=0/SUCCESS)
   Main PID: 8530 (ebusd)
      Tasks: 4 (limit: 1597)
        CPU: 88ms
     CGroup: /system.slice/ebusd.service
             └─8530 /usr/bin/ebusd --scanconfig=full -d /dev/ttyUSB0 -p 8888 --httpport=8889 --htmlpath=/var/ebusd/html--latency=10000 --receivetimeout=100000

Feb 07 19:09:31 raspberrypi systemd[1]: Starting ebusd, the daemon for communication with eBUS heating systems....
Feb 07 19:09:31 raspberrypi systemd[1]: Started ebusd, the daemon for communication with eBUS heating systems..
root@raspberrypi:/home/pi# 

root@raspberrypi:/home/pi# ebusctl info
version: ebusd 21.3.v21.3
device: /dev/ttyUSB0
signal: acquired
symbol rate: 33
max symbol rate: 118
min arbitration micros: 355
max arbitration micros: 638
min symbol latency: 3
max symbol latency: 6
reconnects: 0
masters: 3
messages: 18
conditional: 0
poll: 0
update: 4
address 10: master #2
address 23: slave
address 25: slave
address 26: slave
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 3f: master #23
address 50: slave
address ec: slave

root@raspberrypi:/home/pi# 

root@raspberrypi:/home/pi# ebusd --scanconfig=full --checkconfig FF08070400/0AB5454850303003277201
2022-02-07 19:24:47.564 [main notice] ebusd 21.3.v21.3 performing configuration check...
2022-02-07 19:24:47.694 [main notice] found messages: 11 (0 conditional on 0 conditions, 0 poll, 4 update)
2022-02-07 19:24:47.836 [main notice] read common config file vaillant/broadcast.csv
2022-02-07 19:24:47.870 [main notice] read common config file vaillant/scan.csv
2022-02-07 19:24:47.902 [main notice] read common config file vaillant/general.csv
2022-02-07 19:24:48.007 [main notice] read scan config file vaillant/08.ehp.csv for ID "ehp00", SW0327, HW7201
2022-02-07 19:24:48.102 [main notice] found messages: 340 (139 conditional on 21 conditions, 1 poll, 59 update)
2022-02-07 19:24:48.105 [main notice] ebusd stopped
root@raspberrypi:/home/pi# 

root@raspberrypi:/home/pi# ebusctl scan 23
ERR: read timeout

root@raspberrypi:/home/pi# 

root@raspberrypi:/home/pi# ebusctl scan 25
**ERR: read timeout**

root@raspberrypi:/home/pi# ebusctl scan 26
**ERR: read timeout**

root@raspberrypi:/home/pi# ebusctl scan 50
**ERR: read timeout**

root@raspberrypi:/home/pi# ebusctl scan ec
**ERR: ACK error**

root@raspberrypi:/home/pi# 

root@raspberrypi:/home/pi# ls /dev/serial/by-id/
usb-ESERA_GmbH_eBus_Coupler_12001_AL7DS696-if00-port0
root@raspberrypi:/home/pi# 

JSON:
{
 "broadcast": {
  "messages": {   "datetime": {
    "name": "datetime",
    "passive": true,
    "write": false,
    "lastup": 1644254467,
    "zz": 254,
    "fields": {
     "outsidetemp": {"value": 4.250},
     "time": {"value": "18:21:00"},
     "date": {"value": "07.02.2022"}
    }
   },
   "error": {
    "name": "error",
    "passive": true,
    "write": false,
    "lastup": 0
   },
   "id-u": {
    "name": "id",
    "passive": true,
    "write": false,
    "lastup": 0
   },
   "signoflife": {
    "name": "signoflife",
    "passive": true,
    "write": false,
    "lastup": 0
   }
  }
 },
 "scan.15": {
  "messages": {   "": {
    "name": "",
    "passive": false,
    "write": false,
    "lastup": 0
   }
  }
 },
 "scan.23": {
  "messages": {   "": {
    "name": "",
    "passive": false,
    "write": false,
    "lastup": 0
   }
  }
 },
 "scan.25": {
  "messages": {   "": {
    "name": "",
    "passive": false,
    "write": false,
    "lastup": 0
   }
  }
 },
 "scan.26": {
  "messages": {   "": {
    "name": "",
    "passive": false,
    "write": false,
    "lastup": 0
   }
  }
 },
 "scan.44": {
  "messages": {   "": {
    "name": "",
    "passive": false,
    "write": false,
    "lastup": 0
   }
  }
 },
 "scan.50": {
  "messages": {   "": {
    "name": "",
    "passive": false,
    "write": false,
    "lastup": 0
   }
  }
 },
 "scan.ec": {
  "messages": {   "": {
    "name": "",
    "passive": false,
    "write": false,
    "lastup": 0
   }
  }
 },
 "global": {
  "version": "21.3.v21.3",
  "updatecheck": "OK",
  "signal": true,
  "symbolrate": 21,
  "maxsymbolrate": 111,
  "minarbitrationmicros": 349,
  "maxarbitrationmicros": 652,
  "minsymbollatency": 5,
  "maxsymbollatency": 6,
  "qq": 49,
  "reconnects": 0,
  "masters": 3,
  "messages": 18,
  "lastup": 1644254467
 }
}

  GNU nano 5.4                                                                        /etc/default/ebusd                                                                                  
# /etc/default/ebusd:
# config file for ebusd service.

# Options to pass to ebusd (run "ebusd -?" for more info):

#EBUSD_OPTS="--configpath=http://ebusd.eu/config/ -d /dev/ttyUSB0 -p 8888  --latency=0 --receivetimeout=10000 -l /var/log/ebusd.log  --scanconfig=full  --httpport=8889 --htmlpath=/var/e>
#EBUSD_OPTS="--scanconfig=full  --configpath=/etc/ebusd/vaillant/ -d /dev/ttyUSB0 -p 8888  --latency=100 --receivetimeout=100 -l /var/log/ebusd.log  --scanconfig=full  --httpport=8889 ->

#EBUSD_OPTS="--scanconfig=full  --configpath=/etc/ebusd/ -d /dev/ttyUSB0 -p 8888  --httpport=8889  --htmlpath=/var/ebusd/html"
EBUSD_OPTS="--scanconfig=full -d /dev/ttyUSB0 -p 8888  --httpport=8889  --htmlpath=/var/ebusd/html"  --latency=10000  --receivetimeout=100000

# MULTIPLE EBUSD INSTANCES WITH SYSV
# In order to run multiple ebusd instances on a SysV enabled system, simply
# define several EBUSD_OPTS with a unique suffix for each. Recommended is to
# use a number as suffix for all EBUSD_OPTS settings. That number will then be
# taken as additional "instance" parameter to the init.d script in order to
# start/stop an individual ebusd instance instead of all instances.
# Example: (uncomment the EBUSD_OPTS above)
#EBUSD_OPTS1="--scanconfig -d /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0 -p 8888 -l /var/log/ebusd1.log"

baggerfahrer441, 2022-02-14:
Thanks for the reply.
Today I spoke with the manufacturer of the esera eBus Coupler USB. I received the note that ebusd may be too fast for the coupler now. In addition, the software always wants to write very fast to the ebus.
If so, does that mean that this coupler is no longer working with ebusd?
Is there a chance to continue with this coupler or do I have to switch to a self-built coupler ?
If Yes, then I have to finish the project because I cannot build and test such a board.
I only want to read some values from my heating (flow temperatures, solar yield, runtime solar pump).

john30, 2022-02-15:
hm, thats a weird answer. ebusd can't be too fast as it depends completely on how quick the data arrives. it is a matter of the eBUS protocol and the connected devices. if one participant does not follow the specs, it disturbs the others. luckily most of the devices are tolerant against that and that's why the "slow" adapters work at all in the first place (and don't damage the other devices).
you can use the adapter without any issue for listening to ebus traffic.
you remove the logging, right? so it's hard to tell anymore

baggerfahrer441, 2022-02-15:
The file ebusd.log is available:
[ebusd_20220215.log](https://github.com/john30/ebusd/files/8074591/ebusd_20220215.log)

baggerfahrer441, 2022-02-15:
Now I basically downgraded it.
I started my old Windows 2000 PC.
On it I installed "eBus-Connector" and the Windows 2000 drivers for the esera eBus Coupler USB.
And now it works, I see data from my Vaillant auroMATIC 620.
A first step towards success!
Now I have to sleep...

john30, 2022-02-18:
if you switch off the scan feature and pick the CSV files manually, you would get translated messages as well. but as long as sending does not work reliably, you cannot actively query the bus

ivanhoe-code, 2022-02-18:
I also have an issue using ebusd with a Vaillant ecoTEC plus and I do not know, if it might be related.
Most importantly, as hardware interface, I use a very simple setup based on [https://ebus-wiki.org/lib/exe/fetch.php/ebus/arduinoebus.pdf]
In my case, it is an Arduino Uno reading data from the EBUS using SoftSerial. The Arduino is connected via USB to a RPi Zero W 2, where ebusd is installed.
Of course, with this setup, I can only read and not write. I hoped, some useful data would still be broadcasted on the BUS, but so far I had no luck. 

The question is: can ebusd work at all without the ability to write on the BUS? And how would I test my setup?

What I got so far is:

```
> ebusctl info
version: ebusd 22.1.v22.1
update check: OK
device: /dev/ttyUSB0
signal: acquired
symbol rate: 116
max symbol rate: 125
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

and http://localhost:8889/data/ returns

```
{
 "broadcast": {
  "messages": {   "datetime": {
    "name": "datetime",
    "passive": true,
    "write": false,
    "lastup": 0
   },
   "error": {
    "name": "error",
    "passive": true,
    "write": false,
    "lastup": 0
   },
   "id-u": {
    "name": "id",
    "passive": true,
    "write": false,
    "lastup": 0
   },
   "signoflife": {
    "name": "signoflife",
    "passive": true,
    "write": false,
    "lastup": 0
   }
  }
 },
 "global": {
  "version": "22.1.v22.1",
  "signal": true,
  "symbolrate": 23,
  "maxsymbolrate": 80,
  "qq": 49,
  "reconnects": 0,
  "masters": 1,
  "messages": 11,
  "lastup": 0
 }
}
```

So there is no actual data about my Vaillant device in there.
How do I know, if the received data actually makes sense? Could it be that my Arduino setup just returns nonsense?
I had a look at the raw numbers from the serial interface and it returns a bunch of numbers I cannot make sense of. Maybe the data from the BUS is logically inverted or something?
Or would ebusd react differently in that case?

Any advice would be appreciated!

baggerfahrer441, 2022-02-18:
@tikismoke,
you asked after "eBus-Connector":
Internet page: https://www.ebus-wiki.org/doku.php/ebus/ebusconnector
download software (with registration): https://www.ebus-wiki.org/doku.php/ebus/download/ebusconnector
driver for my old PC for Virtual COM Port Drivers: https://ftdichip.com/drivers/vcp-drivers/
I've took the very old one for Windows 2000: https://www.ftdichip.com/Drivers/CDM/Win2k/CDM%202.06.00%20WHQL%20Certified.zip

After the installation I had to check first in the control panel of Windos 2000 which USB port is used by the driver.
this number must be entered in the file eBusConnector.ini:
_; ComNum legt fest, an welcher seriellen Schnittstelle der Konverter angeschlossen ist.
; die anderen Einstellungen sollten für eBus passen und sollten nicht geändert werden.
[eBusCom]
ComNum=**5**;

The program ran for a few hours (apparently a default) and generated some data.
At the weekend I will check how I can continue and whether I can use ebusd for myself.

mighty896, 2022-03-13:
Hi everybody, 

I seem to have a similar problem as @ivanhoe-code. Using an ebus adapter version 3 and connecting it to the bus, the board seems to run fine. yellow LED is on and green LED is flashing like indicating bus activity. Sometimes it shines a little brighter like something happens on the bus, but that's only my interpretation.

running ebusctl info it shows no other device than the ebus adapter:
`ebusctl info
version: ebusd 22.2.v22.2-7-g320782c
update check: revision v22.2 available
device: /dev/ttyUSB0
signal: acquired
symbol rate: 39
max symbol rate: 101
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
`
I still use the "standard" configuration as it was after the installation (meaning I kept the --scanconfig option in the ebusd option file) as I'm just at the beginning of the installation

Actually, there must be some traffic on the bus, because I also use a Vaillant Sensonet interface which works pretty ok and pushes data into the "small, but slow and low-tech" Vaillant cloud. 

Any ideas what I'm doing wrong? Restarted the heating twice in order to let it reset but it didn't help.

mighty896, 2022-03-13:
Resolved my issue on my own and with the help of other issue entries. 
It was a problem with the configuration file. 

Inserting -d enh:/dev/ttyUSB0 solved the issue.

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30