john30/ebusd #520: Docker certificate issue

andig, 2022-02-12:
### Description

Seeing this in my docker logs as of today 22:00:

```
2022-02-12 21:55:40.341 [network error] HTTP connect: 337047686=certificate verify failed
2022-02-12 21:55:40.342 [main error] invalid configPath URL
```

Haven't change anything for about 2 years now apart from updating the docker image.

### Actual behavior

certificate verify failed, no further information

### Expected behavior

no error 

### ebusd version

latest release

### ebusd arguments

--enabledefine --enablehex --accesslevel='*' --scanconfig -f -d       192.168.0.46:9999 --latency 20000 --mqtthost=nas.fritz.box --mqttport=1883

### Operating system

other

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

adapter 3 WiFi

### Logs

```
2022-02-12 21:55:11.323 [network error] HTTP connect: 337047686=certificate verify failed
2022-02-12 21:55:11.323 [main error] invalid configPath URL
2022-02-12 21:55:14.963 [network error] HTTP connect: 337047686=certificate verify failed
2022-02-12 21:55:14.963 [main error] invalid configPath URL
2022-02-12 21:55:17.400 [network error] HTTP connect: 337047686=certificate verify failed
2022-02-12 21:55:17.401 [main error] invalid configPath URL
2022-02-12 21:55:20.310 [network error] HTTP connect: 337047686=certificate verify failed
2022-02-12 21:55:20.310 [main error] invalid configPath URL
2022-02-12 21:55:23.358 [network error] HTTP connect: 337047686=certificate verify failed
2022-02-12 21:55:23.358 [main error] invalid configPath URL
2022-02-12 21:55:27.336 [network error] HTTP connect: 337047686=certificate verify failed
2022-02-12 21:55:27.337 [main error] invalid configPath URL
2022-02-12 21:55:32.337 [network error] HTTP connect: 337047686=certificate verify failed
2022-02-12 21:55:32.337 [main error] invalid configPath URL
2022-02-12 21:55:40.341 [network error] HTTP connect: 337047686=certificate verify failed
2022-02-12 21:55:40.342 [main error] invalid configPath URL
```

andig, 2022-02-12:
The adapter is alive as usual:

```
❯ telnet 192.168.0.46 9999
Trying 192.168.0.46...
Connected to ebus2.fritz.box.
Escape character is '^]'.
�����������������������������������������^C^C^C������������������������������������������������������	LE������R�#v����R
```

closed 2022-02-13 by john30

john30, 2022-02-13:
the config download was moved to HTTPS recently (e3dfe3235091d5c8e23e22418d66ada60600f232) but for the docker images the certs were forgotten

alphachris, 2022-04-14:
> the config download was moved to HTTPS recently ([e3dfe32](https://github.com/john30/ebusd/commit/e3dfe3235091d5c8e23e22418d66ada60600f232)) but for the docker images the certs were forgotten

fresh docker install on RPI4, tried all tags from Docker-Hub, issue still persists:
`
1931-02-12 20:15:13.038 [network error] HTTP connect: 337047686=certificate verify failed
1971-03-20 21:20:32.000 [main error] invalid configPath URL
`

john30, 2022-04-16:
not sure what to do about it. which image tag are you using?

alphachris, 2022-04-16:
> not sure what to do about it. which image tag are you using?

22.2 on a freshly installed RaspberryPi OS with freshly installed Docker using their install script

alphachris, 2022-04-16:
@john30 
following the advice from my first link on ServerFault, I was able to fix the issue: https://serverfault.com/a/1048351/497724
it basically boils down to this:
> It seems it is related to outdated libseccomp (the only dependency from Docker besides kernel itself) which explains why a privileged container was not affected (syscall filtering disabled).

probably it makes sense to update the README file of the docker image, I can imagine other users using a RPI4 experiencing the same issue.

commit 6be07b55 (2022-02-13): add missing ca-certs (fixes #520)