john30/ebusd #522: Simplify mqttjson dict format

wlcrs, 2022-02-13:
### Description

I'm using the new `mqqtjson` option to integrate ebusd in to Home Assistant. 

I'm however a bit surprised by how fields which contain multiple values are handled. I would propose to simplify how you return these fields, and remove those unnecessary `{"name": ,"value":}`.

In case of naming conflicts, you could suffix the index?

So instead of:
```
{
     "0": {"name": "temp1", "value": 53.0},
     "1": {"name": "temp1", "value": 51.0},
     "2": {"name": "temp2", "value": null},
     "3": {"name": "temp1", "value": null},
     "4": {"name": "temp1", "value": 48.0},
     "5": {"name": "pumpstate", "value": "off"}
}
```

Return this:

```
{
     "temp1_0": 53.0,
     "temp1_1": 51.0,
     "temp2": null,
     "temp1_3": null,
     "temp1_4:" null,
     "pumpstate": "off",
}
```

Nico184, 2022-02-25:
Hello,
I tried to add the field namens in configuration files, but it did not work as expected, as I described here [#255](https://github.com/john30/ebusd-configuration/issues/255#issue-1148570224) of the configuration rep. Did I something wrong?

john30, 2022-02-26:
with --mqttjson=short (commit 1798007f9d10470d045380b220df1813b479c12f) it is now possible to remove the extra object from the payload

john30, 2022-11-13:
completed

closed 2022-11-13 by john30