john30/ebusd #524: Unmatchable topic

andig, 2022-02-13:
### Description

After [XX](https://github.com/john30/ebusd/issues/520) I fail to collect the data I got before.

### Actual behavior

[image]

### Expected behavior

No error in log, continue data retrieval.

### ebusd version

latest release

### ebusd arguments

--enabledefine --enablehex --accesslevel='*' --scanconfig -f -d       192.168.0.46:9999 --latency 20000 --mqtthost=nas.fritz.box --mqttport=1883

### Operating system

other

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

adapter 3 WiFi

### Logs

Logfile shows:

```
2022-02-13 21:04:17.842 [update notice] received read bai Status01 QQ=10: 32.5;32.0;7.125;-;54.5;off
2022-02-13 21:04:18.136 [update notice] received read vr_70 SensorData QQ=10: -;-;-;-;32.75;29.44;c8 7d 00
2022-02-13 21:04:18.403 [update notice] received update-write bai SetMode QQ=10: auto;30.5;-;-;0;0;1;0;0;0
2022-02-13 21:04:18.661 [update notice] received write vr_70 SetActorState QQ=10: -;-;-;-;-;-;off
2022-02-13 21:04:18.911 [update notice] received write vr_70 Mc1FlowTempDesired QQ=10: on;30.5;on;-20
2022-02-13 21:04:19.158 [update notice] received write vr_70 Mc2FlowTempDesired QQ=10: on;30.5;on;5
2022-02-13 21:04:27.864 [update notice] received read bai Status01 QQ=10: 32.0;32.0;7.125;-;54.5;off
2022-02-13 21:04:28.160 [update notice] received read vr_70 SensorData QQ=10: -;-;-;-;32.62;29.38;c8 7d 00
2022-02-13 21:04:28.424 [update notice] received update-write bai SetMode QQ=10: auto;30.5;-;-;0;0;1;0;0;0
2022-02-13 21:04:28.680 [update notice] received read bai Status02 QQ=10: auto;60;55.0;70;70.0
2022-02-13 21:04:28.938 [update notice] received write vr_70 SetActorState QQ=10: -;-;-;-;-;-;off
2022-02-13 21:04:29.188 [update notice] received write vr_70 Mc1FlowTempDesired QQ=10: on;30.5;on;-15
2022-02-13 21:04:29.436 [update notice] received write vr_70 Mc2FlowTempDesired QQ=10: on;30.5;on;5
2022-02-13 21:04:37.929 [update notice] received read bai Status01 QQ=10: 32.0;32.0;7.125;-;54.5;off
2022-02-13 21:04:38.221 [update notice] received read vr_70 SensorData QQ=10: -;-;-;-;32.38;29.31;c8 7d 00
2022-02-13 21:04:38.490 [update notice] received update-write bai SetMode QQ=10: auto;30.5;-;-;0;0;1;0;0;0
2022-02-13 21:04:38.750 [update notice] received write vr_70 SetActorState QQ=10: -;-;-;-;-;-;off
2022-02-13 21:04:39.001 [update notice] received write vr_70 Mc1FlowTempDesired QQ=10: on;30.5;on;-15
2022-02-13 21:04:39.247 [update notice] received write vr_70 Mc2FlowTempDesired QQ=10: on;30.5;on;5
2022-02-13 21:04:40.987 [mqtt error] received unmatchable topic ebusd/700/SystemFlowTemp/get
2022-02-13 21:04:40.987 [mqtt error] received unmatchable topic ebusd/700/Hc1ActualFlowTempDesired/get
2022-02-13 21:04:40.987 [mqtt error] received unmatchable topic ebusd/700/Hc1FlowTemp/get
2022-02-13 21:04:40.988 [mqtt error] received unmatchable topic ebusd/700/Hc1MixerMovement/get
2022-02-13 21:04:40.988 [mqtt error] received unmatchable topic ebusd/700/Hc1PumpStatus/get
2022-02-13 21:04:40.988 [mqtt error] received unmatchable topic ebusd/700/Hc2ActualFlowTempDesired/get
2022-02-13 21:04:40.989 [mqtt error] received unmatchable topic ebusd/700/Hc2FlowTemp/get
2022-02-13 21:04:40.990 [mqtt error] received unmatchable topic ebusd/700/Hc2MixerMovement/get
2022-02-13 21:04:40.990 [mqtt error] received unmatchable topic ebusd/700/Hc2PumpStatus/get
2022-02-13 21:04:40.991 [mqtt error] received unmatchable topic ebusd/700/HwcStorageTemp/get
2022-02-13 21:04:40.992 [mqtt error] received unmatchable topic ebusd/700/HwcTempDesired/get
2022-02-13 21:04:40.992 [mqtt error] received unmatchable topic ebusd/bai/CirPump/get
2022-02-13 21:04:40.993 [mqtt error] received unmatchable topic ebusd/bai/FlowTempDesired/get
2022-02-13 21:04:40.994 [mqtt error] received unmatchable topic ebusd/bai/HcHours/get
2022-02-13 21:04:40.994 [mqtt error] received unmatchable topic ebusd/bai/HwcHours/get
2022-02-13 21:04:40.995 [mqtt error] received unmatchable topic ebusd/bai/ModulationTempDesired/get
2022-02-13 21:04:40.997 [mqtt error] received unmatchable topic ebusd/bai/PrimaryCircuitFlowrate/get
2022-02-13 21:04:40.997 [mqtt error] received unmatchable topic ebusd/bai/ReturnRegulation/get
2022-02-13 21:04:40.998 [mqtt error] received unmatchable topic ebusd/bai/ExtFlowTempDesiredMin/get
2022-02-13 21:04:40.999 [mqtt error] received unmatchable topic ebusd/bai/PumpPower/get
2022-02-13 21:04:41.000 [mqtt error] received unmatchable topic ebusd/bai/PumpPowerDesired/get
2022-02-13 21:04:41.000 [mqtt error] received unmatchable topic ebusd/vr_70/SensorData/get
```

andig, 2022-02-13:
Looks like https://github.com/john30/ebusd/issues/523 is related.

andig, 2022-02-13:
@john30 is there a last known good docker tag that I could use instead of `latest`?

john30, 2022-02-14:
you could stick v21.3 for now

john30, 2022-02-15:
latest should be fine now as well

andig, 2022-02-16:
The errors messages are gone, the mqtt setup is still imho broken, please see #535.

closed 2022-02-16 by andig