john30/ebusd #525: Write to ebus via mqtt

blanosko, 2022-02-13:
### Description

I cant write to ebus with mqtt publish messages. Maybe I am doing something wrong, but I cant find any information on wiki or anywhere how to properly publish messages on MQTT to write values on the bus.

### Actual behavior

When I publish a MQTT message:

`{
  "onoff": {
    "value": "off"
  }
}`

on topic "ebusd/e7c/HeatingEnable" nothing is written on the bus. In a while my message gets updated by value from controller. Tried even with /set suffix or publish simple 'off' message. Behaviour is the same.

When manualy writing with command ebusctl write, heating is properly switched off.

### Expected behavior

Publish message to the given topic and ebusd writes the value on the bus. Controller then sets heating off.

### ebusd version

latest release

### ebusd arguments

--scanconfig -d enh:192.168.0.117:9999 -l /var/log/ebusd.log --configpath=/root/ebusd-configuration/ebusd-2.1.x/en  --mqtthost=192.168.0.10 --mqttport=1883 --mqttuser=<user> --mqttpass=<pass> --mqttjson --accesslevel="*"

### Operating system

Debian 10 (Buster) / Raspbian 10 / Ubuntu 18-19

### CPU architecture

x64

### Hardware interface

adapter 3 WiFi

### Related integration

MQTT generic

### Logs

When publishing to the given topic with /set suffix, this appears in log. When publishing without /set, nothing is written to the log:

2022-02-13 21:47:50.548 [mqtt error] received unmatchable topic ebusd/e7c/HeatingEnable/set

wlcrs, 2022-02-14:
This looks exactly like my issue #523 which has been fixed yesterday evening :-)

blanosko, 2022-02-14:
> This looks exactly like my issue #523 which has been fixed yesterday evening :-)

So is it merged? Can I just download latest release and dpkg it? Or is it possible only when builded from source?

Thanks

blanosko, 2022-02-14:
I've just cloned repository and build it with make. So on version ebusd 22.1.v22.1-31-g4cdac87 its working and I can successfuly write to bus with MQTT publish message to ebusd/e7c/HeatingEnable/set.

2022-02-14 12:18:41.641 [mqtt notice] write e7c HeatingEnable: off

Thanks for help

closed 2022-02-14 by blanosko