john30/ebusd #529: Solar values and energy count missing from Home Assistant MQTT integration

nodomain, 2022-02-14:
### Description

I have been using ebusd for a long time - thank you for this impressive piece of software :) 

I recently changed the integration from the previous cumbersome REST integration to the newly added MQTT integration using the default example config.

However, I am missing some values.

This is the output from `ebusctl info`:

```
pi@rpi-ebusd:~ $ ebusctl info
version: ebusd 21.3.v21.3-194-g0a79098
update check: version 22.1 available
device: /dev/ttyUSB0
signal: acquired
symbol rate: 35
max symbol rate: 157
min arbitration micros: 318
max arbitration micros: 7747
min symbol latency: 0
max symbol latency: 24
reconnects: 0
masters: 3
messages: 750
conditional: 12
poll: 1
update: 12
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0604;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UI   ;SW=0508;HW=6201", loaded "vaillant/15.ui.csv"
address 23: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/23.solsy.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/25.solsy.hwc.csv"
address 26: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/26.solsy.hc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/50.solsy.mc.csv"
address ec: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/ec.solsy.sc.csv"
```

Previously I retrieved the solar temp using this command:

```yaml
- platform: tcp
  name: Solar temp
  host: automatix.fritz.box
  port: 8888
  timeout: 5
  payload: "r Coll1Sensor\n"
  value_template: "{{value.split (';')[0]}}"
  unit_of_measurement: '°C'

```

and

```yaml
- platform: tcp
  name: Energy Count HC1
  host: automatix.fritz.box
  port: 8888
  timeout: 5
  payload: "r PrEnergyCountHc1\n"

- platform: tcp
  name: Energy Sum HC1
  host: automatix.fritz.box
  port: 8888
  timeout: 5
  payload: "r PrEnergySumHc1\n"

- platform: tcp
  name: Energy Count HWC1
  host: automatix.fritz.box
  port: 8888
  timeout: 5
  payload: "r PrEnergyCountHwc1\n"

- platform: tcp
  name: Energy Sum HWC1
  host: automatix.fritz.box
  port: 8888
  timeout: 5
  payload: "r PrEnergySumHwc1\n"
```

### Actual behavior

These are the only ones available:

<img width="1485" alt="image" src="https://user-images.githubusercontent.com/104389/153809305-faecc32b-ecc5-4120-ae6e-18be3979b6c5.png">

Detail view 1:
<img width="699" alt="image" src="https://user-images.githubusercontent.com/104389/153809467-448bd51b-7d46-4b13-a3c4-31631b2a04ec.png">

Detail view 2:
<img width="687" alt="image" src="https://user-images.githubusercontent.com/104389/153809496-bbf77c82-f29c-4db5-a209-5c9c8a556d5e.png">

### Expected behavior

All values are available. Especially the energy count values are important as well because I want to derive the gas consumption from it.

If it is a config issue, I kindly ask to point me into the right direction.

### ebusd version

current source from git

### ebusd arguments

EBUSD_OPTS="--scanconfig --mqtthost 192.168.10.250 --mqttport 1883 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqttjson --loglevel=error"

### Operating system

Debian 10 (Buster) / Raspbian 10 / Ubuntu 18-19

### CPU architecture

armv7l

### Hardware interface

adapter 3 USB

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

See above

nodomain, 2022-02-14:
22.1 brought back the energy count and sum values, nevertheless no solar temp yet.

john30, 2022-02-15:
that''s because Coll1Sensor does not match the filter-name setting in [mqtt-hassio.cfg:106](https://github.com/john30/ebusd/blob/master/contrib/etc/ebusd/mqtt-hassio.cfg#L106) of "status|temp|yield|count|energy|power|runtime|hours|starts|mode|curve|^load$|^party$". If you want to include that message, change the filter string either directly in the cfg file or via `--mqttvar=...`, see docu

nodomain, 2022-02-18:
Awesome, thanks!

closed 2022-02-18 by nodomain