john30/ebusd #535: Mqtt messages format changed

andig, 2022-02-16:
### Description

Follow-up to https://github.com/john30/ebusd/issues/524, same config.

After upgrading to Docker `latest`, I still don't see data in my monitoring:

[image]

Looking into the topics it seems as if the message format has changed:

[image]

Note that the messages are no longer simple (and immediately-usable) numeric value but concatenated strings. Is this intentional?

john30, 2022-02-16:
seems you forgot to use "--mqttjson"

andig, 2022-02-16:
I've always run it like this:

```
  ebusd:
    container_name: ebusd
    image: john30/ebusd:v21.3
    ports:
      - 8888:8888
    command: --enabledefine --enablehex --accesslevel='*' --scanconfig -f -d
      192.168.0.46:9999 --latency 20000 --mqtthost=nas.fritz.box --mqttport=1883
    restart: always
```

I do not want to have json- which I need to parse- but just the plain numeric values for those topics that support it. That has been the behaviour sofar.

john30, 2022-02-17:
thanks for the feedback, found it.

closed 2022-02-17 by john30

andig, 2022-02-17:
Still not working for me in devel:

    ebusd: unrecognized option '--enabledefine'

is this intentional? I can re-check without the define if I have to remove it.

john30, 2022-02-17:
can't reproduce with image id f50e929baedf

andig, 2022-02-17:
I must have tried https://hub.docker.com/layers/john30/ebusd/devel/images/sha256-58ca588af52e7bddc31d614816d3e6847d9e9b6cedaf4b55ca4e15fba985a973?context=explore with devel tag. I can retry your hash tomorrow.

andig, 2022-02-18:
`devel` is working now ("simple" mqtt messages, no error), `latest` is still broken (no "simple" messages). These are the images I've tried:

```
admin@NAS:~$ sudo docker pull john30/ebusd:latest
latest: Pulling from john30/ebusd
Digest: sha256:e2562ab5a37fd693af57caf3136ca2b4d81084d1784d2c27ed66e59fac6574b5
Status: Image is up to date for john30/ebusd:latest
docker.io/john30/ebusd:latest

admin@NAS:~$ sudo docker pull john30/ebusd:devel
devel: Pulling from john30/ebusd
5eb5b503b376: Already exists
540218732afe: Pull complete
284a7d325952: Pull complete
928818473df5: Pull complete
44fe49cffeaf: Pull complete
Digest: sha256:fd8c2ef7c5f4bd22320df3b05679cac1abf9c949b475aea7bf472918ce044b80
Status: Downloaded newer image for john30/ebusd:devel
docker.io/john30/ebusd:devel
```

john30, 2022-02-19:
latest wasn't updated yet and will be done with next release

commit ea60ef91 (2022-02-17): fix non-json payload prefix (fixes #535)