john30/ebusd #540: Strange Value via mqtt

user1o1, 2022-02-19:
### Description

I have strange a strange value format i my mqtt messages with the new Version 22.1 of ebusd:

An example:
ebud.log:
[update notice] sent poll-read 700 HwcStorageTemp QQ=31: 50

is shown via mqtt: 

 mosquitto_sub -h localhost -t ebus/700/HwcStorageTemp
{
     "tempv": {"value": 5e+01}}

Have i miss something which have to configure in the new version?
After downgrade to 21.3 its fine.

### Actual behavior

 mosquitto_sub -h localhost -t ebus/700/HwcStorageTemp
{
     "tempv": {"value": 5e+01}}

### Expected behavior

 mosquitto_sub -h localhost -t ebus/700/HwcStorageTemp
{
     "tempv": {"value": 50}}

### ebusd version

latest release

### ebusd arguments

"--scanconfig --localhost --updatecheck=off --receivetimeout=100000 --sendretries=10 --configpath=/etc/ebusd --loglevel=error --mqtthost=x.x.x.x--mqttport=1883 --mqtttopic=ebus --mqttjson --mqttretain"

### Operating system

Debian 10 (Buster) / Raspbian 10 / Ubuntu 18-19

### CPU architecture

armv7l

### Hardware interface

adapter 2

### Logs

[update notice] sent poll-read 700 HwcStorageTemp QQ=31: 50

mwildbolz, 2022-02-19:
Hi!
That's the same that I already saw a couple of days ago - see #514 
Is solved with [f1cac3c](https://github.com/john30/ebusd/commit/f1cac3c48744931ff4ed1721f2917aacd4e5febf)

user1o1, 2022-02-20:
Okay thanks, i will close this for now. Seems be solved in the mentioned push.

closed 2022-02-20 by user1o1