john30/ebusd #542: State value too long

leeyuentuen, 2022-02-21:
### Description

When I use ebus on home assistant. It seems the state value is too long to let HA process them:

```
Exception in message_received when handling msg on 'ebusd/bai/SetMode': '{ "hcmode": {"value": "auto"}, "flowtempdesired": {"value": 28.0}, "hwctempdesired": {"value": 59.0}, "hwcflowtempdesired": {"value": null}, "disablehc": {"value": 0}, "disablehwctapping": {"value": 0}, "disablehwcload": {"value": 0}, "remoteControlHcPump": {"value": 0}, "releaseBackup": {"value": 0}, "releaseCooling": {"value": 0}}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 42, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 282, in message_received self.async_write_ha_state() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 539, in async_write_ha_state self._async_write_ha_state() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 663, in _async_write_ha_state self.hass.states.async_set( File "/usr/src/homeassistant/homeassistant/core.py", line 1357, in async_set state = State( File "/usr/src/homeassistant/homeassistant/core.py", line 1045, in __init__ raise InvalidStateError( homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.setmode. State max length is 255 characters.
> Exception in message_received when handling msg on 'ebusd/global/updatecheck': '"revision v22.1 available, broadcast.csv: different version available, memory.csv: different version available, vaillant/08.bai.csv: different version available, vaillant/bai.308523.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.csv: different version available, vaillant/hcmode.inc: different version available, vaillant/timerhc.inc: different version available"' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 42, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 282, in message_received self.async_write_ha_state() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 539, in async_write_ha_state self._async_write_ha_state() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 663, in _async_write_ha_state self.hass.states.async_set( File "/usr/src/homeassistant/homeassistant/core.py", line 1357, in async_set state = State( File "/usr/src/homeassistant/homeassistant/core.py", line 1045, in __init__ raise InvalidStateError( homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.ebusd_updatecheck_2. State max length is 255 characters.
```

### Actual behavior

-

### Expected behavior

-

### ebusd version

latest release

### ebusd arguments

/etc/ebusd/mqtt-hassio.cfg

### Operating system

other

### CPU architecture

x64

### Hardware interface

adapter 3 WiFi

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

-

john30, 2022-02-26:
that seems to be a limitation on HA side

Koky05, 2022-03-29:
I insert after HA restart new config for updatecheck:
`{
  "unique_id": "ebusd_global_updatecheck",
  "device": {
    "identifiers": "ebusd",
    "manufacturer": "ebusd.eu",
    "name": "ebusd",
    "sw_version": "22.2",
    "suggested_area": "Heating"
  },
  "value_template": "{{ (value_json).split(',')[0] }}",
  "state_topic": "ebusd/global/updatecheck",
  "name": "ebusd updatecheck",
  "state_class": "measurement"
}`

And it works perfectly for me, show in HA:
`Updatecheck
revision v22.2 available`

Koky05, 2022-03-29:
In MQTT Explorer I made publish into topic:
homeassistant/sensor/ebusd_global_updatecheck/config

commit 2fd1d9bd (2022-04-16): truncate updatecheck payload (fixes #542)
closed 2022-04-16 by john30