john30/ebusd #543: Read values get cached

PureDJ, 2022-02-24:
### Description

I have a problem wit reading some values.
I'm having a 360 thermostat and when I rad the topic "ComfTempDesired" and "RoomTemp" it gives the correct value, but when I read it again within a couple of seconds or a minute I get the same value and I do not see the read command in the log.
It takes several minutes until ebusd sends a new request on the bus.

I have MQTT running and it does send the value to the broker only when there was an actual read command from ebusd.

It looks like there is some overrun protection for read commands, but getting the actual temperature from the 360 once every 10 minutes is not fast enough for me.

In the logfile I see the first time this readcommand:
`[update notice] sent read 360 ComfTempDesired QQ=31: 20.00;20.00;20.00;19.00`
but no read command for the second read time.
When I restart ebusd I get the readcommand in the log the first time.

Nico184, 2022-02-24:
Hi, Did you do a read command with ebusctl read...? 
I think you should read the Wiki, chapter READ command. [https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#read]()

There is option -f and -m explained.

PureDJ, 2022-02-24:
I haven't seen that, this solved my problem, thanks!

closed 2022-02-24 by PureDJ