john30/ebusd #551: Adapter3 ETH signal lost

salvq, 2022-02-26:
### Description

While using Adapter 3 Ethernet version, signal got lost after working fine for some time, once is powered off and on, it works again.

### Actual behavior

Ebusd stop receiving data from ebus after few hours, it is required to power off and on in order to obtain signal again. It cycle after few hours again.

### Expected behavior

No power off and on requirement, signal will not get lost

### ebusd version

21.2

### ebusd arguments

    command: -f --scanconfig --configpath=/ebusd/ebusd-2.1.x/en -d enh:192.168.6.98:9999 -l /var/log/ebusd.log --logareas=all --loglevel=info --httpport=8080 --accesslevel=* --mqtthost=192.168.1.106 --mqttport=1883 --mqttuser=xzy --mqttpass=xzy --mqttjson --mqtttopic=ebusd/%circuit/%name

### Operating system

other

### CPU architecture

x64

### Dockerized

other

### Hardware interface

adapter 3 Ethernet

### Related integration

MQTT generic

### Logs

Log when signbal got lost
```
2022-02-26 13:41:10.010 [bus info] poll cmd: 3108b509030d4400
2022-02-26 13:41:10.148 [update info] sent MS cmd: 3108b509030d4400 / 0101
2022-02-26 13:41:10.149 [update notice] sent poll-read bai WP QQ=31: on
2022-02-26 13:41:12.625 [update info] received MS cmd: 1008b5100900005a78ffff00ff00 / 0101
2022-02-26 13:41:12.625 [update notice] received update-write bai SetMode QQ=10: auto;45.0;60.0;-;0;0;0;0;0;0
2022-02-26 13:41:16.036 [bus info] poll cmd: 3108b509030d3f00
2022-02-26 13:41:16.175 [update info] sent MS cmd: 3108b509030d3f00 / 0101
2022-02-26 13:41:16.176 [update notice] sent poll-read bai extWP QQ=31: on
2022-02-26 13:41:16.620 [update info] received MS cmd: 1008b5110101 / 0959500080ff710100ff
2022-02-26 13:41:16.620 [update notice] received read bai Status01 QQ=10: 44.5;40.0;-;-;56.5;on
2022-02-26 13:41:18.599 [update info] received MS cmd: 1008b5110102 / 06033c78468c8c
2022-02-26 13:41:18.600 [update notice] received read bai Status02 QQ=10: auto;60;60.0;70;70.0
2022-02-26 13:41:22.011 [bus info] poll cmd: 3108b509030d9e00
2022-02-26 13:41:22.149 [update info] sent MS cmd: 3108b509030d9e00 / 0100
2022-02-26 13:41:22.150 [update notice] sent poll-read bai Storageloadpump QQ=31: 0
2022-02-26 13:41:22.597 [update info] received MS cmd: 1008b5100900005a78ffff00ff00 / 0101
2022-02-26 13:41:22.598 [update notice] received update-write bai SetMode QQ=10: auto;45.0;60.0;-;0;0;0;0;0;0
2022-02-26 13:41:27.073 [bus error] signal lost
2022-02-26 13:41:32.604 [bus notice] device invalid
2022-02-26 13:41:40.669 [bus error] unable to open 192.168.6.98:9999: ERR: generic I/O error
2022-02-26 13:41:40.669 [bus notice] device invalid
2022-02-26 13:41:48.732 [bus error] unable to open 192.168.6.98:9999: ERR: generic I/O error
2022-02-26 13:41:48.732 [bus notice] device invalid
2022-02-26 13:41:56.797 [bus error] unable to open 192.168.6.98:9999: ERR: generic I/O error
2022-02-26 13:41:56.797 [bus notice] device invalid
2022-02-26 13:42:00.679 [mqtt info] received set topic for f47 Hc1HeatCurve
[... 119 lines cut ...]
version: ebusd 21.2.v21.2
update check: version 22.1 available, broadcast.csv: different version available, memory.csv: different version available, vaillant/08.bai.csv: newer version available, vaillant/15.f47.csv: different version available, vaillant/bai.308523.inc: newer version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.csv: different version available, vaillant/hcmode.inc: newer version available, vaillant/timercc.inc: different version available, vaillant/timerhc.inc: different version available, vaillant/timerhwc.inc: different version available, vaillant/yield3d43.inc: different version available
access: *
signal: acquired
symbol rate: 23
max symbol rate: 137
min arbitration micros: 3
max arbitration micros: 4
min symbol latency: 6
max symbol latency: 10
reconnects: 1
masters: 3
messages: 437
conditional: 3
poll: 47
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0604;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=F4700;SW=0124;HW=6102", loaded "vaillant/15.f47.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd

#
```

john30, 2022-03-01:
can you reproduce with the current firmware 20220220?

pepsonEL, 2022-03-01:
Where is current fw 20220220 to download?

salvq, 2022-03-02:
Will try and update....

@pepsonEL, see the [firmware](https://adapter.ebusd.eu/firmware/ChangeLog) link where you can also download

pepsonEL, 2022-03-02:
But it is FW for module ETH LAN not for ESP ?

salvq, 2022-03-02:
This is PIC firmware i.e. firmware for main eBUS Adapter 3

pepsonEL, 2022-03-02:
Ok thanks. I try flash it by USB

Koky05, 2022-03-02:
OOT: When FW will be available as OTA?
[image]

pepsonEL, 2022-03-02:
I try flash my adapter by Bootloader Host Application from Microchip... I setup J11 pin3 and pin 4 by jumper and try. But no succesfully. This is screen,. Setup serial USB to speed 115200
[image]

john30, 2022-03-02:
@pepsonEL stop posting the [same thing](https://github.com/john30/ebusd/discussions/566) over and over again! you did it in the past and if you do it again, I'll have to block you. this is super-annoying and not more than wasting other peoples time!

john30, 2022-03-02:
[ebusd-esp](https://github.com/john30/ebusd-esp) is the firmware for the ESP (e.g. Wemos) only, nothing more, nothing less.
[this here](https://adapter.ebusd.eu/picfirmware.en.html#versions) is the firmware for the PIC soldered on the adapter.
For Wifi feature, you need both and firmware versions are independant.

pepsonEL, 2022-03-03:
Hi
I have still problem with my adapter.... afetr fight i flash firmware PIC to new latest 20220220 and after that when power on all works ok. Then i login to wemos adapter and click reboot without save and after that again NO SIGNAL. 

```
Build: 20211226
ebusd device string: enh:192.168.4.39:9999
WiFi signal: 70% (-65dBm)
[Status](http://192.168.4.39/)
[Configuration](http://192.168.4.39/config)
[PINs](http://192.168.4.39/pins)
[Firmware update](http://192.168.4.39/update)
Chip ID: 001a54f8, [d1mini](http://192.168.4.39/?verbose)
Hostname: ebus-1a54f8
Up time: 60
Free heap: 29928
ebusd connected: yes (inactive)
eBUS signal: no signal
```

Then i must power off and again power on adapter and all works again ok.

john30, 2022-03-19:
different topic since it is Wifi and not Ethernet, but anyway: what does the LEDs on the adapter look like in this case?

salvq, 2022-03-19:
There are no more connection issues with ethernet adapter, it has not been disconnected for about the week or so.

The issue is closed for me.

```
pi@rpi4-tv:~ $ sudo docker logs 7b9b9e0547c0 | grep -e '192.168' -e 'bus error' -e 'ERR'
2022-03-11 20:31:50.132 [bus error] poll bai TargetFanSpeed failed: ERR: wrong symbol received
2022-03-11 21:35:34.008 [bus error] device status: unexpected available enhanced following byte 1
2022-03-11 22:52:08.109 [bus error] poll f47 Hc1NightTemp failed: ERR: SYN received
2022-03-12 00:27:32.252 [bus error] device status: eBUS comm error: framing
2022-03-12 02:55:44.149 [bus error] poll bai Flame failed: ERR: SYN received
2022-03-12 04:39:17.746 [bus error] device status: unexpected available enhanced byte 2
2022-03-12 06:04:49.652 [bus error] device status: unexpected available enhanced following byte 1
2022-03-12 06:46:36.270 [bus error] device status: unexpected available enhanced byte 2
2022-03-12 15:16:57.965 [bus error] device status: eBUS comm error: framing
2022-03-12 18:33:11.259 [bus error] device status: eBUS comm error: framing
2022-03-12 18:35:57.341 [bus error] device status: unexpected available enhanced following byte 1
2022-03-12 19:34:50.155 [bus error] poll bai TargetFanSpeed failed: ERR: read timeout
2022-03-12 20:20:26.144 [bus error] poll f47 Hc1NightTemp failed: ERR: read timeout
2022-03-12 22:09:02.124 [bus error] poll bai HcStarts failed: ERR: SYN received
2022-03-13 02:22:38.419 [bus error] device status: unexpected available enhanced byte 2
2022-03-13 04:41:11.371 [bus error] device status: eBUS comm error: framing
2022-03-13 10:06:29.273 [bus error] device status: eBUS comm error: framing
2022-03-13 11:00:44.362 [bus error] poll bai currenterror failed: ERR: CRC error
2022-03-13 13:04:25.996 [bus error] device status: unexpected available enhanced byte 2
2022-03-14 01:03:52.571 [bus error] device status: unexpected available enhanced byte 2
2022-03-14 04:50:05.549 [bus error] device status: eBUS comm error: framing
2022-03-14 12:12:37.302 [bus error] device status: unexpected available enhanced byte 2
2022-03-14 13:46:08.175 [bus error] poll f47 ActualRoomTempDesiredHc1 failed: ERR: read timeout
2022-03-14 17:59:54.970 [bus error] device status: unexpected available enhanced following byte 1
2022-03-14 20:26:43.270 [bus error] device status: eBUS comm error: framing
2022-03-15 01:54:32.648 [bus error] device status: unexpected available enhanced byte 2
2022-03-15 03:23:00.584 [bus error] device status: eBUS comm error: framing
2022-03-15 04:43:02.250 [bus error] device status: eBUS comm error: framing
2022-03-15 09:42:26.156 [bus error] device status: eBUS comm error: framing
2022-03-15 09:42:26.156 [bus error] poll bai HwcWaterflow failed: ERR: CRC error
2022-03-15 10:17:56.157 [bus error] device status: eBUS comm error: framing
2022-03-15 10:17:56.157 [bus error] poll bai TargetFanSpeed failed: ERR: CRC error
2022-03-15 16:41:45.799 [bus error] device status: eBUS comm error: framing
2022-03-15 17:03:47.627 [bus error] device status: unexpected available enhanced following byte 1
2022-03-15 20:16:46.987 [bus error] device status: unexpected available enhanced byte 2
2022-03-15 20:28:50.133 [bus error] device status: eBUS comm error: framing
2022-03-15 20:56:26.215 [bus error] poll bai TargetFanSpeed failed: ERR: read timeout
2022-03-15 22:03:38.204 [bus error] poll f47 Hc1HolidayStartPeriod failed: ERR: read timeout
2022-03-15 22:16:51.182 [bus error] device status: eBUS comm error: framing
2022-03-16 01:20:26.155 [bus error] device status: eBUS comm error: framing
2022-03-16 01:20:26.155 [bus error] poll bai IonisationVoltageLevel failed: ERR: CRC error
2022-03-16 01:31:20.118 [bus error] poll f47 Hc1SFMode failed: ERR: wrong symbol received
2022-03-16 05:46:50.265 [bus error] poll f47 Hc1HolidayStartPeriod failed: ERR: CRC error
2022-03-16 13:21:29.808 [bus error] device status: unexpected available enhanced byte 2
2022-03-16 20:09:32.201 [bus error] poll bai PositionValveSet failed: ERR: read timeout
2022-03-17 00:48:41.800 [bus error] device status: eBUS comm error: framing
2022-03-17 02:28:54.141 [bus error] device status: eBUS comm error: framing
2022-03-17 03:11:10.913 [bus error] device status: unexpected available enhanced byte 2
2022-03-17 05:25:14.163 [bus error] device status: eBUS comm error: framing
2022-03-17 05:25:14.166 [bus error] poll bai WP failed: ERR: CRC error
2022-03-17 10:03:59.514 [bus error] device status: unexpected available enhanced following byte 1
2022-03-17 10:38:34.330 [bus error] device status: eBUS comm error: framing
2022-03-17 11:20:35.270 [bus error] device status: unexpected available enhanced byte 2
2022-03-18 03:43:06.274 [bus error] device status: eBUS comm error: framing
2022-03-18 14:26:49.635 [bus error] device status: eBUS comm error: framing
2022-03-18 22:23:37.803 [bus error] device status: unexpected available enhanced byte 2
2022-03-19 00:30:31.209 [bus error] device status: eBUS comm error: framing
2022-03-19 01:22:44.583 [bus error] device status: unexpected available enhanced following byte 1
2022-03-19 02:33:21.960 [bus error] device status: unexpected available enhanced following byte 1
2022-03-19 06:38:41.276 [bus error] device status: eBUS comm error: framing
2022-03-19 09:54:06.521 [bus error] device status: unexpected available enhanced following byte 1
2022-03-19 12:20:20.140 [bus error] poll f47 Hc1DayTemp failed: ERR: SYN received
2022-03-19 12:46:14.164 [bus error] device status: unexpected available enhanced following byte 1
2022-03-19 12:46:14.239 [bus error] poll bai FlowTempDesired failed: ERR: read timeout
2022-03-19 12:57:23.361 [bus error] device status: eBUS comm error: framing

```

john30, 2022-03-20:
thanks for checking

closed 2022-03-20 by john30