john30/ebusd #552: Info over the bus, ebusd does not pic up

PureDJ, 2022-02-27:
### Description

I'm wondering how to get all the information on the bus.
For example, my ColorMatic does know when their is hot water (shower) request, but all the messages I get from ebusd do not change.
I think there are other messages ebusd does not pick up, or the messages does not encode complete?

Now I sent some request messages on the bus and I get result, but the info must be there already, because the colormatic is showing the "tap" symbol.

Is there a way to figure this out?
I did allready a  ebusctl grab result all

and this was the result:
`3108070400 / 0ab5424149303005187401 = 1: scan.08
3108b5090127 / 09000000000000000000 = 8: scan.08 id
7f15070400 / 0ab5333630303001397301 = 51682: scan.15
3115b5090127 / 094e39ffffffffffffff = 3: scan.15 id
3184070400 / 0ab54f5453303002059101 = 1: scan.84
3108b510090000555affff00ff00 / 0101 = 51400: bai SetMode
1008b512020064 / 00 = 443: bai StatusCirPump
10feb5160800550420ffffffff = 4383: broadcast vdatetime
1008b5110101 / 09505200804e520000ff = 1: bai Status01
10feb5160800152417ffffffff = 1: broadcast vdatetime
3115b5090124 / 09003231303834363030 = 1: scan.15 id
1008b5040100 / 0a00211900ffffffff0080 = 8761: bai DateTime
1008b5110101 / 09524e00804a500100ff = 26298: bai Status01
1008b5110102 / 05033c8c4670 = 8787: bai Status02
1008b513020507 / 00 = 435
3108b509030d0200 / 03b70700 = 1639: bai WaterPressure
3108b509030d0500 / 01f0 = 9800: bai Flame
3108b509030d1600 / 03730200 = 1: bai HwcTemp
3108b509030d2004 / 02c20b = 30: bai HoursTillService
3108b509030d5800 / 0100 = 7: bai HwcDemand
3115b509030d0f00 / 03530100 = 9803: 360 RoomTemp
3115b509030d1000 / 02ecff = 3: 360 RoomTempOffsetSelfWarming
3115b509030d1100 / 020000 = 4: 360 RoomTempOffset
3115b509030d3900 / 02c003 = 1: 360 HwcTempDesired
3115b509030d3b00 / 084001400140014001 = 1684: 360 ComfTempDesired
3115b509050e1000ecff / 00 = 2: 360 RoomTempOffsetSelfWarming
3115b509050e11000000 / 00 = 3: 360 RoomTempOffset
3115b509040e150000 / 00 = 41: 360 OperatingMode
3115b5090b0e3b004001400140014001 / 00 = 70: 360 ComfTempDesired
1008b510090000146effff01ff00 / 0101 = 1: bai SetMode
`

Some messages are ther because I requested them (for example HwcTemp is requested by me, it was not there before)
Only SetMode, Status01,Status02 and DateTime are sent on a regular interval.

john30, 2022-03-08:
what hardware interface do you have?
there are some (actually many) that might miss some bytes or misinterpret the voltage value seen on the bus.
you could use 'log raw bytes' to see every byte on the bus and maybe even switch on debug logging where you could see things like CRC errors

john30, 2022-09-17:
closed due to inactivity

closed 2022-09-17 by john30