john30/ebusd #555: recent ebusctl acting differently when started from cron

a-dekker, 2022-02-28:
### Description

After upgrading from ebusd-21.3 to ebusd-22.2 I noticed my crontab scripts no longer worked.
It turned out to be related to the ebusctl output, which does not no longer show up when running from cron. It's also not send to stderr now as I assumed. I tried to put as much environment settings in the cron as I could find in my terminal session, but to no avail. The script still works fine from the commandline. Any idea what has changed between 21.3 and 22.2 that could be related?
After reverting back to 21.3 I get my expected output back.

A simple testcase:
`/usr/bin/ebusctl -p 8887 state > /tmp/ebusd_state.txt`
started from cron gives me output when using ebusd-21.3, but it is empty when started with ebusd-22.2 installed.

### Actual behavior

All output is gone using cron, so grep commands in scripts started from cron fail.

### Expected behavior

Get regular ebusctl output to show up just like in 21.3.

### ebusd version

latest release

### ebusd arguments

EBUSD_OPTS="--scanconfig -c /etc/ebusd --port=8887 -d /dev/serial/by-id/usb-ESERA-Automation_eBus_Coupler_Iso_12001_AL188X0D-if00-port0 --mqttport=1883 --httpport=8081 --accesslevel=* --mqtttopic=ebusd/%circuit/%name"

### Operating system

Debian 11 (Bullseye) / Raspbian 11 / Ubuntu 20-21

### CPU architecture

x64

### Hardware interface

Esera USB

### Related integration

MQTT generic

### Logs

ebusd itself works fine, no error messages.

john30, 2022-03-01:
does the same happen if you use `echo "s"|nc -N localhost 8887 >  /tmp/ebusd_state.txt` as cron entry instead?

a-dekker, 2022-03-01:
Thanks for your reply. Just checked, using netcat works fine on ebusd-22.2.

a-dekker, 2022-03-02:
I also have a python script in the background (started via cron, at reboot) that calls ebusctl via "subprocess.run" and fails to provide output.
As a workaround I replaced all ebusctl calls with 'nc' or socket connections. Works just as well, and from the python point of view it's a cleaner solution than calling OS commands.

closed 2022-03-12 by john30

weinerg, 2022-05-23:
I encounter the same effect.
No output at cron by ebusctl.
Hope for a soon solution. 
Thx.

a-dekker, 2022-05-23:
It should be fixed and released in v22.3

weinerg, 2022-05-23:
Sorry, but I don't think it's fixed in V22.3, because I already use this version.

a-dekker, 2022-05-23:
Although I switch to using netcat, I can confirm this issue that I created is indeed solved.
Crontab now delivers output again using ebusctl in my case (using 22.3). So I am not sure what the problem is in your case.

weinerg, 2022-05-23:
I don't use the version with MQTT support. Perhaps this is the reason?

weinerg, 2022-05-23:
I use ebusd at 32 Raspbian 11 and it does't work with V22.3.

weinerg, 2022-05-23:
I just now checked the issue with old version ebusd-21.3_armv7-bullseye_mqtt1.deb. 
There it runs under cron without fault. So I get output.
In the newest version ebusd-22.3_armv7-bullseye_mqtt1.deb i get no output at cron jobs.
So the error is still existing in armv7 distribution.
I would be glad if I could get an information, when this is also solved in armv7.
Many Thx.

a-dekker, 2022-05-23:
Here is another confirmation that it is fixed, also by someone using armv7: https://github.com/john30/ebusd/issues/578#issuecomment-1072992301
Meanwhile you could consider using netcat (nc) as an alternative like I did.

cakysk, 2022-05-23:
well.. I'm having the same issue, testing every new version, however the last one that is working is ebusd-21.3_armv7-bullseye

john30, 2022-06-19:
just tested it with 22.3 armv7 and works. you might want to check if you really use the ebusctl binary from the package and not some older version

civic9, 2022-10-29:
Hi. I have the same problem with v.22.4. armv6l. pi zero. It happens only for a few of my readings (randomly) when I am reading many (about 50) values in one script (via crontab). And it looks like the problem is much more bigger (more empty readings via crontab) when I run command like:
``while true ; do ebusctl read  -c bai -f OutdoorstempSensor  ; done``
concurrently in an interactive shell. But still the problem is only with readings under crontab.

cakysk, 2022-10-29:
.. yep, I am facing same behavior, with similar cron scrip.. I am still using version 21.3

commit 320782c2 (2022-03-12): allow one more read from socket after input is closed (fix #555)