john30/ebusd #558: MQTT Autodiscovery in Home Assistant

estegewr, 2022-03-01:
### Description

I use the MQTT Auto Discovery function to register the ebusd sensors in Home Assistant. This works well when ebusd is started. However, if I restart the Home Assistant server, I have to restart ebusd as well. Otherwise the sensors are displayed as "unavailable". Even the reception of corresponding payloads does not change anything. Only a restart of ebusd revives the sensors.

### Actual behavior

See description.

### Expected behavior

Same behavior as other Home Assistant integrations which register its sensors via Auto Discovery (in my configuration), e. g. owntracks, zigbee2mqtt, RPI monitor. Ebusd is the only one, which currenty shows this behavior.

### ebusd version

latest release

### ebusd arguments

-c /etc/ebusd.min -d enh:/dev/ttyUSB_ebus3 --mqtthost=localhost --mqttport=1883 --mqttuser=xxx --mqttpass=xxx  --mqttjson --mqtttopic=ebusd/%circuit/%name --mqttint=/etc/ebusd/mqtt-hassio.cfg --loglevel=error --logareas=main --logareas=network --ac cesslevel=* --latency=20000 --scanconfig

### Operating system

Debian 10 (Buster) / Raspbian 10 / Ubuntu 18-19

### CPU architecture

armv7l

### Hardware interface

adapter 3 RPi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

n/a

Koky05, 2022-03-01:
Use retain flag "--mqttretain" to keep data in MQTT, so after restart of HA you will have all sensors with last received data:
`-c /etc/ebusd.min -d enh:/dev/ttyUSB_ebus3 --mqtthost=localhost --mqttport=1883 --mqttuser=xxx --mqttpass=xxx --mqttjson --mqttretain --mqtttopic=ebusd/%circuit/%name --mqttint=/etc/ebusd/mqtt-hassio.cfg --loglevel=error --logareas=main --logareas=network --ac cesslevel=* --latency=20000 --scanconfig`

estegewr, 2022-03-01:
Confirmed: It's working as intended!

closed 2022-03-01 by estegewr

john30, 2022-03-01:
retention is not the ideal solution in my opinion, but a valid option of course.
the integration is set up to automatically re-push the definitions once home assistant is back online due to https://github.com/john30/ebusd/blob/master/contrib/etc/ebusd/mqtt-hassio.cfg#L331-L334

estegewr, 2022-03-01:
I used the default _mqtt-hassio.cfg_ provided with the release, but ebusd did not republish the configuration.

Anyhow, thanks to the flexible options you implemented, @john30, we got freedom of choice depending on the individual use case. I am migrating to Home Assistant and must restart server frequently to update configuration. In this case I am satisfied with the retention mode.

john30, 2022-03-02:
you could check if HA sends a topic "homeassistant/status" with payload "online" once it is back. this is taken as the signal to republish the definitions

estegewr, 2022-03-02:
Bullseye!
I use Mosquitto's ACL to limit permission & visibility of MQTT devices to follow a need-to-know principle. I was not aware that ebusd also reads on topic  _homeassistant_. So this is the root cause why ebusd did not refresh the configuration as you described.
Does ebusd rely on more messages on topic _homeassistant_ ?

Personally I still prefer the retention mode: Configured once and resistant even if MQTT broker restarts. But good to know how I can easily refresh config by sending a dedicated MQTT message.

john30, 2022-03-08:
no, that's the only thing ebusd listens on and is defined in the last lines of the cfg file