john30/ebusd #559: Different --mqtttopic handling

DeadWalkingDeath, 2022-03-01:
### Description

After upgrading from v21.3 to current v22.2 docker image I've noticed different handling of parameter --mqtttopic.

I've set this to ebusd/%name, as I don't need the circuit name.

In v21.3 the mqtt topic used was for example ebusd/HK1Raumtemp

### Actual behavior

In v22.2 the mqtt topic used is now ebusd/HK1RaumtempR16/
The circuit name is appended to the message name and there is an additional empty topic after the message name.

### Expected behavior

ebusd/HK1Raumtemp should be used when setting --mqtttopic=ebusd/%name, as it was before.

### ebusd version

latest release

### ebusd arguments

--pollinterval=10 --mqtthost=192.168.0.210 --mqttport=1883 --mqtttopic=ebusd/%name

### Operating system

Debian 11 (Bullseye) / Raspbian 11 / Ubuntu 20-21

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

adapter 3 USB

### Logs

none

john30, 2022-03-08:
good spot! the topic definition had to be reworked for the HA integration and so was the default. need to correct that for your (rare) use case

john30, 2022-03-12:
due to the new MQTT integration support, redoing the old behaviour would prevent capabilities of that feature.
thus you need to use the new option to append an '#' to the topic to tell ebusd to not add any part of the default, i.e. use these ebusd arguments instead:
`--pollinterval=10 --mqtthost=192.168.0.210 --mqttport=1883 --mqtttopic=ebusd/%name/#`

closed 2022-03-12 by john30

JanHBade, 2023-06-18:
setting the topic to --mqtttopic=ebusd/lueftung results in strange behaviour

the version with the # is not the right for me....

[image]