john30/ebusd #568: High CPU usage in x86 docker

salvq, 2022-03-04:
### Description

I am facing high CPU usage in QNAP NAS x64 in docker

CPU usage is very high constantly constantly ~ 25% (100%/1core from 4)

### Actual behavior

No error, working fine but high CPU usage

### Expected behavior

Less CPU usage ~ few percent

### ebusd version

latest release

### ebusd arguments

-f --scanconfig --configpath=/ebusd/ebusd-2.1.x/en -d enh:192.168.6.98:9999 -l /var/log/ebusd.log --logareas=all --loglevel=info --httpport=8080 --accesslevel=* --mqtthost=192.168.1.106 --mqttport=1883 --mqttuser=xyz --mqttpass=xyz --mqttjson --mqtttopic=ebusd/%circuit/%name

### Operating system

other

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

adapter 3 Ethernet

### Related integration

MQTT generic

### Logs

Standard log (info) doesn't show anything unusual

**linux top log in host**
```
Mem: 7509332K used, 469772K free, 336744K shrd, 1000620K buff, 2274252K cached
CPU:  4.9% usr 24.1% sys  4.7% nic 63.8% idle  2.2% io  0.0% irq  0.0% sirq
Load average: 2.78 2.52 2.52 4/1269 17568
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
22073 22045 admin    S N   261m  3.3   0 25.3 ebusd -f --scanconfig --configpath=/ebusd/ebusd-2.1.x/en -d enh:192.168.6.98:9999 -l /var/log/ebusd.log --logareas=all --loglevel=info --httpport=8080 --accesslevel=* --mqtthost=
21992 20722 admin    S N   216m  2.7   0  0.8 container-station/python2.7 docker-monitor.pyc
 3114     1 admin    S    2012m 25.7   0  0.7 /sbin/hal_daemon -f
20070 24665 admin    S N  3217m 41.1   3  0.4 python3 -m homeassistant --config /config

[~] # docker exec -it d1b75f2d2746 uname -a
Linux NASEFC472 5.10.60-qnap #1 SMP Sat Jan 29 09:16:22 CST 2022 x86_64 GNU/Linux
[~] #

```

**linux top log in docker**
```
[~] # docker exec -it 1791afda2d2e top
top - 14:37:42 up 6 days,  3:01,  0 users,  load average: 2.43, 2.47, 2.50
Tasks:   3 total,   1 running,   2 sleeping,   0 stopped,   0 zombie
%Cpu(s):  1.2 us, 23.6 sy,  4.9 ni, 70.2 id,  0.0 wa,  0.0 hi,  0.1 si,  0.0 st
KiB Mem :  7979104 total,   460128 free,  3829084 used,  3689892 buff/cache
KiB Swap: 32826160 total, 32395096 free,   431064 used.  3297376 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
    1 root      25   5  267264   1484      0 S 100.3  0.0   2531:35 ebusd
   38 root      25   5   41192    836    380 S   0.0  0.0   0:29.65 top
   97 root      25   5   41020   3148   2692 R   0.0  0.0   0:00.01 top
```

I tried version v21.2 and also v22.2 and same result with ~100% CPU utilization / 1 core:
```
[~] # docker exec -it 1791afda2d2e ebusctl info
version: ebusd 21.2.v21.2
```

salvq, 2022-03-07:
I made a test in raspberry pi and CPU usage in docker is close to 0%

```
pi@rpi4-tv:~/ebusd $ uname -a
Linux rpi4-tv 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux
pi@rpi4-tv:~/ebusd $ docker exec -it d052541dc9ca top
top - 11:21:53 up 8 days, 22:50,  0 users,  load average: 0.15, 0.04, 0.01
Tasks:   2 total,   1 running,   1 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.0 us,  0.0 sy,  0.0 ni,100.0 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :   1867.2 total,   1083.5 free,    173.0 used,    610.7 buff/cache
MiB Swap:    100.0 total,     99.7 free,      0.2 used.   1600.0 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
    1 root      20   0   52780   4364   3328 S   0.0   0.2   0:03.75 ebusd
  369 root      20   0    5228   2172   1788 R   0.0   0.1   0:00.02 top

```

Does someone observer similar behaviour in x86 docker i.e. high CPU usage ?

john30, 2022-03-08:
can you check the load per thread? ps command usually supports printing info about each thread

salvq, 2022-03-08:
Hmmm, this is network related...however I have number of other dockers also database related but no such overload

Anything more I can troubleshoot for ?

**Executed within docker**

```
# top -H -p 1
top - 23:06:02 up 1 day,  3:12,  0 users,  load average: 4.03, 3.23, 2.37
Threads:   5 total,   1 running,   4 sleeping,   0 stopped,   0 zombie
%Cpu(s):  6.9 us, 25.3 sy,  6.4 ni, 60.8 id,  0.6 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :   7792.1 total,   1092.6 free,   2800.6 used,   3898.8 buff/cache
MiB Swap:  32056.8 total,  32052.7 free,      4.1 used.   4170.2 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
    8 root      25   5  239992   8604   6836 R  99.9   0.1   4:59.79 network
    7 root      25   5  239992   8604   6836 S   0.3   0.1   0:00.47 bushandler
    1 root      25   5  239992   8604   6836 S   0.0   0.1   0:00.07 ebusd
    9 root      25   5  239992   8604   6836 S   0.0   0.1   0:00.02 mainloop
   10 root      25   5  239992   8604   6836 S   0.0   0.1   0:00.02 MQTT

```

john30, 2022-03-19:
is it the same with the latest devel image?

salvq, 2022-03-19:
Yes, see below

**devel version**

```
# ebusctl info
version: ebusd 22.2.p20220319
device: 192.168.6.98:9999, enhanced
access: *
signal: acquired
symbol rate: 24
max symbol rate: 132
min arbitration micros: 2
max arbitration micros: 5
min symbol latency: 7
max symbol latency: 9
reconnects: 0
masters: 3
messages: 437
conditional: 3
poll: 47
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0604;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=F4700;SW=0124;HW=6102", loaded "vaillant/15.f47.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd

#

```
**Network still overload the cpu**

```
# top -H -p 1
top - 18:28:12 up 11 days, 22:34,  0 users,  load average: 4.13, 3.67, 2.67
Threads:   5 total,   1 running,   4 sleeping,   0 stopped,   0 zombie
%Cpu(s):  7.0 us, 24.2 sy,  0.8 ni, 67.5 id,  0.5 wa,  0.0 hi,  0.1 si,  0.0 st
MiB Mem :   7792.1 total,    316.3 free,   3331.4 used,   4144.3 buff/cache
MiB Swap:  32056.8 total,  31218.8 free,    837.9 used.   3768.6 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
    8 root      20   0  379264   8560   6784 R  99.9   0.1   2:47.32 network
    7 root      20   0  379264   8560   6784 S   0.3   0.1   0:00.26 bushandler
    1 root      20   0  379264   8560   6784 S   0.0   0.1   0:00.07 ebusd
    9 root      20   0  379264   8560   6784 S   0.0   0.1   0:00.02 mainloop
   10 root      20   0  379264   8560   6784 S   0.0   0.1   0:00.00 MQTT

```

salvq, 2022-05-26:
Just tested with latest devel and still CPU is close 100% in docker in qnap

```
root@NASEFC472:/# ebusctl info
version: ebusd 22.3.p20220513
device: 192.168.6.98:9999, enhanced
access: *
signal: acquired
symbol rate: 40
max symbol rate: 133
min arbitration micros: 1
max arbitration micros: 31
min symbol latency: 6
max symbol latency: 11
reconnects: 0
masters: 3
messages: 437
conditional: 3
poll: 47
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0604;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=F4700;SW=0124;HW=6102", loaded "vaillant/15.f47.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd

```

```
root@NASEFC472:/# top -H -p 1
top - 18:52:19 up 5 days, 19:45,  0 users,  load average: 2.62, 2.62, 2.05
Threads:   5 total,   1 running,   4 sleeping,   0 stopped,   0 zombie
%Cpu(s):  9.4 us, 24.7 sy,  0.0 ni, 64.4 id,  1.5 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :   7792.1 total,    851.1 free,   3622.2 used,   3318.8 buff/cache
MiB Swap:  32056.8 total,  31305.0 free,    751.8 used.   3680.1 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
    8 root      20   0  379304   8556   6800 R  98.7   0.1   6:37.24 network
    7 root      20   0  379304   8556   6800 S   0.3   0.1   0:00.61 bushandler
    1 root      20   0  379304   8556   6800 S   0.0   0.1   0:00.07 ebusd
    9 root      20   0  379304   8556   6800 S   0.0   0.1   0:00.03 mainloop
   10 root      20   0  379304   8556   6800 S   0.0   0.1   0:00.02 MQTT
```

john30, 2022-06-19:
just tried it on a new debian hyperv install and it does not make any difference if run from docker or directly on the host.

anyway, just read that you're using a "QNAP NAS x64", so why do you run x86 docker in the first place?

salvq, 2022-06-25:
Well I am using x86_64 bit version all the time, i.e., all the above screen from QNAP are related to x86_64

```
[pristup@NASEFC472 ~]$ docker exec -it be0288582b8a top -H -p 1
top - 19:43:23 up 18 days,  2:01,  0 users,  load average: 2.83, 2.72, 2.17
Threads:   5 total,   1 running,   4 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.0 us, 26.9 sy,  3.8 ni, 46.2 id, 23.1 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :   7792.1 total,    640.6 free,   3749.1 used,   3402.5 buff/cache
MiB Swap:  32056.8 total,  31036.0 free,   1020.8 used.   3494.9 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
    9 root      25   5  240036   8580   6840 R  99.9   0.1   7:21.24 network
    1 root      25   5  240036   8580   6840 S   0.0   0.1   0:00.07 ebusd
    8 root      25   5  240036   8580   6840 S   0.0   0.1   0:00.63 bushandler
   10 root      25   5  240036   8580   6840 S   0.0   0.1   0:00.03 mainloop
   11 root      25   5  240036   8580   6840 S   0.0   0.1   0:00.02 MQTT

[pristup@NASEFC472 ~]$ docker inspect be0288582b8a | grep ARCH
                "EBUSD_ARCH=amd64",
[pristup@NASEFC472 ~]$ docker inspect be0288582b8a | grep version
                "com.docker.compose.version": "1.29.2",
                "version": "22.3-amd64-devel"
[pristup@NASEFC472 ~]$
```

salvq, 2022-06-25:
I have been finally able to resolve the issue, it is related to HTTP port setup. As I have not been using HTTP client, I changed the port to 8085 and it worked. I did have colision with 8080 however I didn't troubleshoot deeper why it was working with increased utilization etc.

I just took out HTTP port setup from configuration, removed `--httpport=8080` and close to 0% utilization

```
[pristup@NASEFC472 ~]$ docker exec -it 73e91db2fd91 top -H -p 1
top - 20:02:30 up 18 days,  2:21,  0 users,  load average: 2.23, 2.64, 2.49
Threads:   5 total,   0 running,   5 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.0 us,  4.7 sy,  2.3 ni, 69.8 id, 23.3 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :   7792.1 total,    672.9 free,   3734.5 used,   3384.7 buff/cache
MiB Swap:  32056.8 total,  31036.5 free,   1020.3 used.   3508.5 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
    1 root      25   5  239820   7744   6284 S   0.0   0.1   0:00.07 ebusd
    7 root      25   5  239820   7744   6284 S   0.0   0.1   0:00.17 bushandler
    8 root      25   5  239820   7744   6284 S   0.0   0.1   0:00.00 network
    9 root      25   5  239820   7744   6284 S   0.0   0.1   0:00.02 mainloop
   10 root      25   5  239820   7744   6284 S   0.0   0.1   0:00.00 MQTT

[pristup@NASEFC472 ~]$ docker logs 73e91db2fd91
2022-06-25 20:00:41.908 [main notice] ebusd 22.3.p20220513 started with auto scan on enhanced device 192.168.6.98:9999
2022-06-25 20:00:41.908 [main info] loading configuration files from /ebusd/ebusd-2.1.x/en/
2022-06-25 20:00:41.908 [main info] reading templates /
2022-06-25 20:00:41.908 [main info] read templates in /
2022-06-25 20:00:41.908 [main info] reading file broadcast.csv
2022-06-25 20:00:41.909 [main info] successfully read file broadcast.csv
2022-06-25 20:00:41.909 [main info] reading file memory.csv
2022-06-25 20:00:41.909 [main info] successfully read file memory.csv
2022-06-25 20:00:41.909 [main info] read config files, got 11 messages
2022-06-25 20:00:41.936 [bus notice] device status: resetting
2022-06-25 20:00:41.936 [bus notice] bus started with own address 31/36
2022-06-25 20:00:41.936 [main info] registering data handlers
2022-06-25 20:00:41.936 [mqtt info] mosquitto version 2.0.11 (compiled with 2.0.11)
2022-06-25 20:00:41.937 [main info] registered data handlers
2022-06-25 20:00:41.937 [mqtt notice] connection established
2022-06-25 20:00:41.962 [bus notice] signal acquired
2022-06-25 20:00:42.381 [bus notice] new master 10, master count 2
2022-06-25 20:00:42.445 [bus notice] new master 03, master count 3
2022-06-25 20:00:42.445 [update info] received MS cmd: 1008b5110101 / 095a5b0080ff780000ff
2022-06-25 20:00:42.445 [update notice] received unknown MS cmd: 1008b5110101 / 095a5b0080ff780000ff
2022-06-25 20:00:42.868 [bus notice] device status: reset, supports info
2022-06-25 20:00:44.392 [update info] received BC cmd: 10feb51603019017
2022-06-25 20:00:44.392 [update notice] received unknown BC cmd: 10feb51603019017
2022-06-25 20:00:48.451 [update info] received MS cmd: 1008b5100900000078ffff01ff00 / 0101
2022-06-25 20:00:48.451 [update notice] received unknown MS cmd: 1008b5100900000078ffff01ff00 / 0101
2022-06-25 20:00:51.937 [bus info] scan 08 cmd: 3108070400
2022-06-25 20:00:51.996 [bus info] arbitration delay 3 - 3 micros
2022-06-25 20:00:52.003 [bus info] send/receive symbol latency 7 - 7 ms
2022-06-25 20:00:52.102 [update info] sent MS cmd: 3108070400 / 0ab5424149303006045502
2022-06-25 20:00:52.103 [bus notice] scan 08: ;Vaillant;BAI00;0604;5502
2022-06-25 20:00:52.103 [update notice] store 08 ident: done
2022-06-25 20:00:52.103 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0604;5502
2022-06-25 20:00:52.103 [bus notice] scan 08: ;Vaillant;BAI00;0604;5502
2022-06-25 20:00:52.103 [main info] reading templates vaillant
2022-06-25 20:00:52.104 [main info] read templates in vaillant
2022-06-25 20:00:52.105 [main notice] read common config file vaillant/broadcast.csv
2022-06-25 20:00:52.105 [main notice] read common config file vaillant/general.csv
2022-06-25 20:00:52.105 [main notice] read common config file vaillant/scan.csv
2022-06-25 20:00:52.105 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0604, HW5502
2022-06-25 20:00:52.106 [bus info] send message: 3108b5090124
2022-06-25 20:00:52.158 [bus info] arbitration delay 3 - 4 micros
2022-06-25 20:00:52.266 [update info] sent MS cmd: 3108b5090124 / 09003231313433333030
2022-06-25 20:00:52.266 [update notice] sent scan-read scan.08 id QQ=31:
2022-06-25 20:00:52.266 [bus info] send message: 3108b5090125
2022-06-25 20:00:52.429 [update info] sent MS cmd: 3108b5090125 / 09313030313137333830
2022-06-25 20:00:52.429 [update notice] sent scan-read scan.08 id QQ=31:
2022-06-25 20:00:52.429 [bus info] send message: 3108b5090126
2022-06-25 20:00:52.652 [update info] sent MS cmd: 3108b5090126 / 09303031303035333938
2022-06-25 20:00:52.652 [update notice] sent scan-read scan.08 id QQ=31:
2022-06-25 20:00:52.652 [bus info] send message: 3108b5090127
[... 8 lines cut ...]

services:
  ebusd:
    image: john30/ebusd:devel
    container_name: ebusd
    restart: unless-stopped
    privileged: true
    network_mode: host
    environment:
      - EBUSD_SCANCONFIG=
      - EBUSD_CONFIGPATH=/ebusd/ebusd-2.1.x/en
      - EBUSD_LOGAREAS=all
      - EBUSD_LOGLEVEL=info
      - EBUSD_ACCESSLEVEL=*
      - EBUSD_MQTTHOST=192.168.1.106
      - EBUSD_MQTTPORT=1883
      - EBUSD_MQTTUSER=XXXX
      - EBUSD_MQTTPASS=YYYY
      - EBUSD_MQTTTOPIC=ebusd/%circuit/%name
      - EBUSD_MQTTJSON=
    volumes:
      - /share/Container/ebusd/ebusd-configuration:/ebusd
      - /etc/localtime:/etc/localtime:ro
    command: -d enh:192.168.6.98:9999 -l /var/log/ebusd.log
```

closed 2022-06-28 by john30