john30/ebusd #569: WaterPresure lost

tall-r, 2022-03-05:
### Description

Last night ebusd stopped to collect information about WaterPresure from Protherm 9 KE/14

### Actual behavior

I use HA integrated with ebusd and today I found error about ebusd integration something like "WaterPresure is not in 'bai'". I checked ebusd' logs and found that WaterPresure now doesn't collect from heater. It looks like something changed in configuration files but I use cfg.ebusd.eu for configs and can't prove it.

### Expected behavior

nothing to add to "actual behavior"

### ebusd version

21.3

### ebusd arguments

 -f --scanconfig -d enh:192.168.2.107:9999 --latency=20 --log bus:debug --log main:debug --log other:debug

### Operating system

other

### CPU architecture

x64

### Dockerized

same as ebusd version

### Hardware interface

adapter 3 WiFi

### Related integration

TCP (cmdline client like ebusctl or netcat), HTTP

### Logs

ebus_1           | 2022-03-05 14:37:37.625 [main notice] ebusd 21.3.v21.3 started with auto scan on enhanced device 192.168.2.107:9999
ebus_1           | 2022-03-05 14:37:37.647 [main info] loading configuration files from http://ebusd.eu/config/
ebus_1           | 2022-03-05 14:37:37.774 [main info] reading templates /
ebus_1           | 2022-03-05 14:37:37.904 [main info] read templates in /
ebus_1           | 2022-03-05 14:37:37.904 [main info] reading file memory.csv
ebus_1           | 2022-03-05 14:37:38.035 [main info] successfully read file memory.csv
ebus_1           | 2022-03-05 14:37:38.035 [main info] reading file broadcast.csv
ebus_1           | 2022-03-05 14:37:38.172 [main info] successfully read file broadcast.csv
ebus_1           | 2022-03-05 14:37:38.172 [main info] read config files
ebus_1           | 2022-03-05 14:37:38.239 [bus notice] device status: resetting
ebus_1           | 2022-03-05 14:37:38.239 [bus notice] bus started with own address 31/36
ebus_1           | 2022-03-05 14:37:38.239 [main info] registering data handlers
ebus_1           | 2022-03-05 14:37:38.239 [main info] registered data handlers
ebus_1           | 2022-03-05 14:37:38.275 [bus debug] ERR: SYN received during no signal, switching to ready
ebus_1           | 2022-03-05 14:37:38.275 [bus notice] signal acquired
ebus_1           | 2022-03-05 14:37:39.150 [bus notice] device status: reset
ebus_1           | 2022-03-05 14:37:48.240 [main debug] performing regular tasks
ebus_1           | 2022-03-05 14:37:51.533 [bus notice] new master 00, master count 2
ebus_1           | 2022-03-05 14:37:51.618 [bus debug] ERR: read timeout during receive command ACK, switching to skip
ebus_1           | 2022-03-05 14:37:52.376 [bus debug] ERR: read timeout during receive command ACK, switching to skip
ebus_1           | 2022-03-05 14:37:58.240 [main debug] performing regular tasks
ebus_1           | 2022-03-05 14:37:58.240 [bus info] scan 05 cmd: 3105070400
ebus_1           | 2022-03-05 14:37:58.276 [bus debug] start request 31
ebus_1           | 2022-03-05 14:37:58.277 [bus debug] arbitration start with 31
ebus_1           | 2022-03-05 14:37:58.324 [bus debug] arbitration won
ebus_1           | 2022-03-05 14:37:58.325 [bus debug] arbitration delay 4 micros
ebus_1           | 2022-03-05 14:37:58.325 [bus info] arbitration delay 4 - 4 micros
ebus_1           | 2022-03-05 14:37:58.325 [bus debug] switching from ready to send command
ebus_1           | 2022-03-05 14:37:58.334 [bus debug] send/receive symbol latency 9 ms
ebus_1           | 2022-03-05 14:37:58.334 [bus info] send/receive symbol latency 9 - 9 ms
ebus_1           | 2022-03-05 14:37:58.343 [bus debug] send/receive symbol latency 8 ms
ebus_1           | 2022-03-05 14:37:58.343 [bus info] send/receive symbol latency 8 - 9 ms
ebus_1           | 2022-03-05 14:37:58.352 [bus debug] send/receive symbol latency 9 ms
ebus_1           | 2022-03-05 14:37:58.362 [bus debug] send/receive symbol latency 9 ms
ebus_1           | 2022-03-05 14:37:58.362 [bus debug] switching from send command to send command CRC
ebus_1           | 2022-03-05 14:37:58.371 [bus debug] send/receive symbol latency 9 ms
ebus_1           | 2022-03-05 14:37:58.371 [bus debug] switching from send command CRC to receive command ACK
ebus_1           | 2022-03-05 14:37:58.456 [bus debug] notify request: ERR: read timeout
ebus_1           | 2022-03-05 14:37:58.456 [bus debug] ERR: read timeout during receive command ACK, switching to skip
ebus_1           | 2022-03-05 14:37:58.456 [main error] scan config 05: ERR: read timeout
ebus_1           | 2022-03-05 14:38:00.456 [main debug] performing regular tasks
ebus_1           | 2022-03-05 14:38:07.140 [main debug] performing regular tasks
ebus_1           | 2022-03-05 14:38:07.140 [main debug] >>> read -m 900 -c bai HwcTemp
ebus_1           | 2022-03-05 14:38:07.140 [main debug] <<< ERR: element not found
ebus_1           | 2022-03-05 14:38:07.143 [main debug] >>> read -m 900 -c bai StorageTemp
ebus_1           | 2022-03-05 14:38:07.143 [main debug] <<< ERR: element not found
ebus_1           | 2022-03-05 14:38:07.145 [main debug] >>> read -m 900 -c bai StorageTempDesired
ebus_1           | 2022-03-05 14:38:07.145 [main debug] <<< ERR: element not found
ebus_1           | 2022-03-05 14:38:07.148 [main debug] >>> read -m 900 -c bai averageIgnitiontime
ebus_1           | 2022-03-05 14:38:07.148 [main debug] <<< ERR: element not found
ebus_1           | 2022-03-05 14:38:07.149 [main debug] >>> read -m 900 -c bai maxIgnitiontime
ebus_1           | 2022-03-05 14:38:07.149 [main debug] <<< ERR: element not found
ebus_1           | 2022-03-05 14:38:07.151 [main debug] >>> read -m 900 -c bai ReturnTemp
ebus_1           | 2022-03-05 14:38:07.151 [main debug] <<< ERR: element not found
ebus_1           | 2022-03-05 14:38:07.153 [main debug] >>> read -m 900 -c bai FlowTempDesired
ebus_1           | 2022-03-05 14:38:07.153 [main debug] <<< ERR: element not found
ebus_1           | 2022-03-05 14:38:07.154 [main debug] >>> read -m 900 -c bai FlowTemp
ebus_1           | 2022-03-05 14:38:07.154 [main debug] <<< ERR: element not found
ebus_1           | 2022-03-05 14:38:07.158 [main debug] >>> read -m 900 -c bai StateNumber
ebus_1           | 2022-03-05 14:38:07.158 [main debug] <<< ERR: element not found
ebus_1           | 2022-03-05 14:38:07.159 [main debug] >>> read -m 900 -c bai HeatingSwitch
ebus_1           | 2022-03-05 14:38:07.159 [main debug] <<< ERR: element not found
ebus_1           | 2022-03-05 14:38:17.159 [main debug] performing regular tasks
ebus_1           | 2022-03-05 14:38:27.160 [main debug] performing regular tasks
ebus_1           | 2022-03-05 14:38:37.141 [main debug] performing regular tasks
ebus_1           | 2022-03-05 14:38:37.141 [main debug] >>> read -m 900 -c bai HwcTemp
ebus_1           | 2022-03-05 14:38:37.141 [main debug] <<< ERR: element not found

john30, 2022-03-08:
you can check which configs are available by using e.g. `ebusctl find`
it is always a good idea to check the message definitions before looking at higher level integrations like MQTT

Mozzarella7, 2022-03-12:
Hi, this is probably according to bug mentioned on HA blog

> ebusd
A typo was fixed in the ebusd integration for monitored condition WaterPressure. It is incorrectly named “WaterPreasure” in the code.
> 
> The [documentation](https://www.home-assistant.io/integrations/ebusd/#waterpressure) represents the correct value.
> ([@nklebedev](https://github.com/nklebedev) - [#66355](https://github.com/home-assistant/core/pull/66355)) ([documentation](https://www.home-assistant.io/integrations/ebusd))

john30, 2022-03-19:
not an ebusd issue then

closed 2022-03-19 by john30