john30/ebusd #571: I'm not able to connect to Vaillant VRC 700

brozikcz, 2022-03-10:
### Description

Hello,

I would like to ask if someone help me with troubleshooting my ebus setup.
I'm using my own ESP adapter with the latest [ebus-v3_esp01.bin](https://github.com/john30/ebusd-esp/blob/master/dist/ebus-v3_esp01.bin) firmware.

The wifi signal is about ~70% and when I'm testing the ESP adapter via https://github.com/john30/ebusd/wiki/6.-Hardware#adjusting-the-potentiometer so I don't get any `signal lost` and just I see many `<aa` chars. So I assume the connection is stable.

My problem is I can't recognize the VRC700 module which is connected to the boiler. 

ebusctl info:
```
version: ebusd 22.2.p20220227
update check: revision v22.2 available
device: 192.168.100.223:9999, enhanced
access: *
signal: acquired
symbol rate: 22
max symbol rate: 95
min arbitration micros: 6
max arbitration micros: 32
min symbol latency: 5
max symbol latency: 41
reconnects: 0
masters: 3
messages: 225
conditional: 3
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0105;HW=7803", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

### Actual behavior

I can't recognize the VRC700 module which is connected to the boiler

### Expected behavior

I want to recognize the VRC700 module which is connected to the boiler

### ebusd version

latest release

### ebusd arguments

--scanconfig -d enh:192.168.100.223:9999 --latency=20 --loglevel=info --accesslevel="*"

### Operating system

Debian 11 (Bullseye) / Raspbian 11 / Ubuntu 20-21

### CPU architecture

arm64

### Hardware interface

adapter 3 WiFi

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

2022-03-10 12:37:52.947 [main notice] ebusd 22.2.p20220227 started with auto scan on enhanced device 192.168.100.223:9999
2022-03-10 12:37:52.947 [main info] loading configuration files from https://cfg.ebusd.eu/
2022-03-10 12:37:53.228 [main info] reading templates /
2022-03-10 12:37:53.501 [main info] read templates in /
2022-03-10 12:37:53.501 [main info] reading file memory.csv
2022-03-10 12:37:53.727 [main info] successfully read file memory.csv
2022-03-10 12:37:53.728 [main info] reading file broadcast.csv
2022-03-10 12:37:53.957 [main info] successfully read file broadcast.csv
2022-03-10 12:37:53.957 [main info] read config files, got 11 messages
2022-03-10 12:37:54.005 [bus notice] device status: resetting
2022-03-10 12:37:54.006 [bus notice] bus started with own address 31/36
2022-03-10 12:37:54.007 [bus notice] device status: reset
2022-03-10 12:37:54.007 [main info] registering data handlers
2022-03-10 12:37:54.008 [main info] registered data handlers
2022-03-10 12:37:54.026 [bus notice] signal acquired
2022-03-10 12:37:57.374 [bus notice] new master 10, master count 2
2022-03-10 12:38:04.010 [bus info] scan 15 cmd: 3115070400
2022-03-10 12:38:04.062 [bus info] arbitration delay 8 - 8 micros
2022-03-10 12:38:04.069 [bus info] send/receive symbol latency 6 - 6 ms
2022-03-10 12:38:04.255 [main error] scan config 15: ERR: read timeout
2022-03-10 12:38:29.174 [update info] received BC cmd: 10feb51603016008
2022-03-10 12:38:29.174 [update notice] received unknown BC cmd: 10feb51603016008
2022-03-10 12:38:29.901 [bus notice] new master 03, master count 3
2022-03-10 12:38:29.901 [update info] received MS cmd: 1008b5120204ff / 0101
2022-03-10 12:38:29.902 [update notice] received unknown MS cmd: 1008b5120204ff / 0101
2022-03-10 12:38:30.163 [update info] received MS cmd: 1008b513020508 / 00
2022-03-10 12:38:30.164 [update notice] received unknown MS cmd: 1008b513020508 / 00
2022-03-10 12:38:30.360 [update info] received BC cmd: 10feb510020601
2022-03-10 12:38:30.360 [update notice] received unknown BC cmd: 10feb510020601
2022-03-10 12:38:36.258 [bus info] scan 08 cmd: 3108070400
2022-03-10 12:38:36.306 [bus info] arbitration delay 8 - 32 micros
2022-03-10 12:38:36.315 [bus info] send/receive symbol latency 6 - 8 ms
2022-03-10 12:38:36.341 [bus info] send/receive symbol latency 6 - 10 ms
2022-03-10 12:38:36.426 [update info] sent MS cmd: 3108070400 / 0ab5424149303001057803
2022-03-10 12:38:36.426 [bus notice] scan 08: ;Vaillant;BAI00;0105;7803
2022-03-10 12:38:36.426 [update notice] store 08 ident: done
2022-03-10 12:38:36.426 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0105;7803
2022-03-10 12:38:36.426 [bus notice] scan 08: ;Vaillant;BAI00;0105;7803
2022-03-10 12:38:36.438 [bus info] send/receive symbol latency 5 - 10 ms
2022-03-10 12:38:36.648 [main info] reading templates vaillant
2022-03-10 12:38:36.878 [main info] read templates in vaillant
2022-03-10 12:38:37.330 [main notice] read common config file vaillant/broadcast.csv
2022-03-10 12:38:37.573 [main notice] read common config file vaillant/scan.csv
2022-03-10 12:38:37.818 [main notice] read common config file vaillant/general.csv
2022-03-10 12:38:38.061 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0105, HW7803
2022-03-10 12:38:38.062 [bus info] send message: 3108b5090124
2022-03-10 12:38:38.129 [bus info] arbitration delay 6 - 32 micros
2022-03-10 12:38:38.141 [bus info] send/receive symbol latency 5 - 12 ms
2022-03-10 12:38:38.186 [bus info] send/receive symbol latency 5 - 16 ms
2022-03-10 12:38:38.261 [update info] sent MS cmd: 3108b5090124 / 09003231323132363030
2022-03-10 12:38:38.262 [update notice] sent scan-read scan.08 id QQ=31: 
2022-03-10 12:38:38.262 [bus info] send message: 3108b5090125
2022-03-10 12:38:38.384 [update info] sent MS cmd: 3108b5090125 / 09313030323138373530
2022-03-10 12:38:38.385 [update notice] sent scan-read scan.08 id QQ=31: 
2022-03-10 12:38:38.386 [bus info] send message: 3108b5090126
2022-03-10 12:38:38.587 [bus info] send/receive symbol latency 5 - 17 ms
2022-03-10 12:38:38.587 [update info] sent MS cmd: 3108b5090126 / 09303031303035323936
2022-03-10 12:38:38.588 [update notice] sent scan-read scan.08 id QQ=31: 
2022-03-10 12:38:38.589 [bus info] send message: 3108b5090127
2022-03-10 12:38:38.629 [bus info] send/receive symbol latency 5 - 41 ms
2022-03-10 12:38:38.757 [update info] sent MS cmd: 3108b5090127 / 094e353c3c3c3c3c3c3c
2022-03-10 12:38:38.758 [update notice] sent scan-read scan.08 id QQ=31: 21;21;26;0010021875;0001;005296;N5
2022-03-10 12:38:39.517 [main notice] found messages: 225 (3 conditional on 26 conditions, 0 poll, 10 update)
2022-03-10 12:38:39.518 [bus info] scan 08 cmd: 3108b5090124
2022-03-10 12:38:39.713 [update info] sent MS cmd: 3108b5090124 / 09003231323132363030
2022-03-10 12:38:39.713 [update notice] sent scan-read scan.08 id QQ=31: 21;21;26;0010021875;0001;005296;N5
2022-03-10 12:38:39.714 [bus info] scan 08 cmd: 3108b5090125
2022-03-10 12:38:39.894 [update info] sent MS cmd: 3108b5090125 / 09313030323138373530
2022-03-10 12:38:39.895 [update notice] sent scan-read scan.08 id QQ=31: 21;21;26;0010021875;0001;005296;N5
2022-03-10 12:38:39.895 [bus info] scan 08 cmd: 3108b5090126
2022-03-10 12:38:40.061 [update info] sent MS cmd: 3108b5090126 / 09303031303035323936
2022-03-10 12:38:40.062 [update notice] sent scan-read scan.08 id QQ=31: 21;21;26;0010021875;0001;005296;N5
2022-03-10 12:38:40.062 [bus info] scan 08 cmd: 3108b5090127
2022-03-10 12:38:40.228 [update info] sent MS cmd: 3108b5090127 / 094e353c3c3c3c3c3c3c
2022-03-10 12:38:40.229 [update notice] sent scan-read scan.08 id QQ=31: 21;21;26;0010021875;0001;005296;N5
2022-03-10 12:38:40.230 [bus notice] scan 08: ;21;21;26;0010021875;0001;005296;N5
2022-03-10 12:38:40.230 [main info] scan config 08 message received
2022-03-10 12:39:29.887 [update info] received BC cmd: 10feb51603016008
2022-03-10 12:39:29.889 [update notice] received update-read broadcast outsidetemp QQ=10: 8.375
2022-03-10 12:39:58.693 [update info] received MS cmd: 1008b5110102 / 06033c5a468c6a
2022-03-10 12:39:58.694 [update notice] received read bai Status02 QQ=10: auto;60;45.0;70;70.0
2022-03-10 12:40:29.857 [update info] received BC cmd: 10feb51603016008
2022-03-10 12:40:29.858 [update notice] received update-read broadcast outsidetemp QQ=10: 8.375
2022-03-10 12:40:42.517 [main notice] update check: revision v22.2 available

brozikcz, 2022-03-10:
It seems to be working fine after I did "reconfiguration" the potentiometer on my board.

john30, 2022-03-11:
that's why we got rid of the potentiometer in the first place starting with [adapter v2 and continued with v3](https://adapter.ebusd.eu/)

brozikcz, 2022-03-11:
I see but the adapter is currently unavailable that's why I have a "temporary solution" for now.