john30/ebusd #572: raspberry bullseye ipv6 problem makes ebusd fail

Aid1295, 2022-03-10:
### Description

During startup it throws "[main error] invalid configPath URL". If make it use ipv4 with /etc/hosts it works.

It looks like raspberry out-of-box ipv6 config doesn't get the routes right - though it tries to use ipv6. Other hosts, like mac work fine with same ipv6 router.

### Actual behavior

If feasible fall back to ipv4.

### Expected behavior

maybe a note at covenient place 

### ebusd version

latest release

### ebusd arguments

package defaults after installation

### Operating system

Debian 11 (Bullseye) / Raspbian 11 / Ubuntu 20-21

### CPU architecture

arm64

### Hardware interface

other

### Logs

never gets to ebusd.logh - throws invalid configpath

john30, 2022-03-11:
are you sure to use the latest release 22.2? please post the ebusd startup settings found in /etc/default/ebusd

Aid1295, 2022-03-11:
Here we go. The problem is with ipv6  - even curl cannot fetch anything using -6 from this bullseye raspberry 3b. It might be the Huawei 5G box, it's less than perfect, but ipv6 has been working with older 32bit Raspbian versions - I have been using ipv6 because Huawei ipv4 DHCP won't give same address when a computer asks for renewal. Maybe i'll need to give fixed ipv4 addresses.. just been lazy - ipv6 addresses are fixed by nature. I'll debug the ipv6 problems at sometime.

This is the very first start:

he job identifier is 967.
Mar 10 20:37:42 raspi3-1 ebusd[1521]: 2022-03-10 20:37:42.032 [network error] HTTP connect: invalid result -1
Mar 10 20:37:42 raspi3-1 ebusd[1521]: 2022-03-10 20:37:42.033 [main error] invalid configPath URL
Mar 10 20:37:42 raspi3-1 systemd[1]: ebusd.service: Control process exited, code=exited, status=22/n/a

/etc/default# ebusd -V
ebusd 22.2.p20220227

cat ebusd 
# /etc/default/ebusd:
# config file for ebusd service.

# Options to pass to ebusd (run "ebusd -?" for more info):
EBUSD_OPTS="--scanconfig" 

# MULTIPLE EBUSD INSTANCES WITH SYSV
# In order to run multiple ebusd instances on a SysV enabled system, simply
# define several EBUSD_OPTS with a unique suffix for each. Recommended is to
# use a number as suffix for all EBUSD_OPTS settings. That number will then be
# taken as additional "instance" parameter to the init.d script in order to
# start/stop an individual ebusd instance instead of all instances.
# Example: (uncomment the EBUSD_OPTS above)
#EBUSD_OPTS1="--scanconfig -d /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0 -p 8888 -l /var/log/ebusd1.log"
#EBUSD_OPTS2="--scanconfig -d /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A900acTF-if00-port0 -p 8889 -l /var/log/ebusd2.log"
#EBUSD_OPTS3="--scanconfig -d /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A900beCG-if00-port0 -p 8890 -l /var/log/ebusd3.log"

# MULTIPLE EBUSD INSTANCES WITH SYSTEMD
# In order to run muiltiple ebusd instances on a systemd enabled system, just
# copy the /lib/systemd/system/ebusd.service file to /etc/systemd/system/
# with a different name (e.g. ebusd-2.service), remove the line starting with
# 'EnvironmentFile=', and replace the '$EBUSD_OPTS' with the options for that
# particular ebusd instance.

john30, 2022-03-12:
still the question, which ebusd version?

Aid1295, 2022-03-12:
/etc/default# ebusd -V
ebusd 22.2.p20220227 - was in the middle of previous post.

But I managed to find the root cause.. though it needed tcpdump with various filters. Just in the middle of my raspi upgrade round one of my wlan boxes - 10 years old D-link has gone crazy, messed up the configs. So it according to nmap it doesn't listen any tcp port anymore - unmanageable - but has started to send ipv6 route advertisements and ipv4 this & that from net 192.168.0.0/24. So l'll close this.

closed 2022-03-12 by Aid1295