john30/ebusd #578: Since ebusd update, crontab script doesn't receive values

Kornelius777, 2022-03-17:
### Description

my /etc/default/ebusd:

`EBUSD_OPTS="--pidfile /var/run/ebusd.pid --scanconfig --readonly --configlang=de --configpath=/etc/ebusd --device=/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A9WFNXT5-if00-port0 --port=8888"`

script runs every minute:

`#!/bin/bash                                                                                                                                                                                                        
                                                                                                                                                                                                                   
Temp=$(/usr/bin/ebusctl r -c mischermodul vorlauftemperatur)                                                                                                                                                       
echo $Temp                                                                                                                                                                                                         
if [[ $Temp != *"ERR"* && $Temp != *"error"* ]]; then                                                                                                                                                              
        echo -en "password\nsetreading Heizung temp_mischer_vorlauf $Temp\nquit\n" |  /bin/nc -w5 fhem.lan.suhle.de 7072                                                                                       
fi`

### Actual behavior

Variable $Temp receives no value

### Expected behavior

Get correct value

### ebusd version

22.2

### ebusd arguments

--pidfile /var/run/ebusd.pid --scanconfig --readonly --configlang=de --configpath=/etc/ebusd --device=/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A9WFNXT5-if00-port0 --port=8888

### Operating system

Debian 11 (Bullseye) / Raspbian 11 / Ubuntu 20-21

### CPU architecture

armv7l

### Hardware interface

other

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

nothing relevant discovered

Commifreak, 2022-03-17:
(May I ask which software generates such graphs?)

john30, 2022-03-19:
seems to be a duplicate of https://github.com/john30/ebusd/issues/555
please check if behaviour is the same with latest source

Kornelius777, 2022-03-19:
With the latest source, ebusctl appears to be working, again.
Thank you for the fix!

john30, 2022-03-19:
great

closed 2022-03-19 by john30